mu-editor/mu #1170: Running Pygame in Mu on MacOS

cturland, 2020-12-09:
If you are reporting a bug, we would like to know:

* What you were trying to do,
Run very basic pygame project on MacOS

* What steps you took to make this happen,
Installed Mu and started a new Python 3 project

* What you expected to happen,
A window should have opened with a red square in the top corner

* What actually happened,
A blank window opened

* Why this difference is problematic (it may not be a bug!),
The code runs differently on Windows machines, which displays the output correctly

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
1.0.3 MU
Multiple versions of MacOS (problem was identified whilst using it in a classroom with a bring your own device environment). All Mac users had the same issue

Log and screenshot of code we were trying to run

Log is from my Windows version which is working (sadly didn't get a log from a student machine before they left)
[image]

2020-12-09 08:50:13,124 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.3
2020-12-09 08:50:13,126 - root:113(run) INFO: uname_result(system='Windows', node='R90Y7SX4-CARTUR', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 12, GenuineIntel')
2020-12-09 08:50:13,127 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\cturland\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-12-09 08:50:13,127 - root:115(run) INFO: Language code: en_GB
2020-12-09 08:50:13,312 - mu.logic:543(__init__) INFO: Setting up editor.
2020-12-09 08:50:13,312 - mu.logic:562(__init__) INFO: Settings path: C:\Users\cturland\AppData\Local\python\mu\settings.json
2020-12-09 08:50:13,313 - mu.logic:563(__init__) INFO: Session path: C:\Users\cturland\AppData\Local\python\mu\session.json
2020-12-09 08:50:13,313 - mu.logic:564(__init__) INFO: Log directory: C:\Users\cturland\AppData\Local\python\mu\Logs
2020-12-09 08:50:13,313 - mu.logic:565(__init__) INFO: Data directory: C:\Users\cturland\AppData\Local\python\mu
2020-12-09 08:50:13,339 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2020-12-09 08:50:13,934 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\cturland\mu_code
2020-12-09 08:50:13,934 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\cturland\AppData\Local\python\mu\session.json
2020-12-09 08:50:13,934 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['C:\\Users\\cturland\\mu_code\\test.py', 'C:\\Users\\cturland\\mu_code\\resize.py', 'C:\\Users\\cturland\\mu_code\\bg.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2020-12-09 08:50:13,934 - mu.logic:715(_load) INFO: Loading script from: C:\Users\cturland\mu_code\resize.py
2020-12-09 08:50:13,935 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-12-09 08:50:13,935 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-12-09 08:50:13,936 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2020-12-09 08:50:13,936 - mu.logic:789(_load) DEBUG: #This is needed to give us access to the resize method
import pygame

#You have to keep an original copy to resize
alien_original = Actor('alien')

#This will be used as the resized actor
alien = Actor('alien')

#Set the starting size variables
alien_height = 500
alien_width = 500

#Doesn't work as we are using 1.2
alien.opacity = 1.0

def draw():
    screen.fill((128, 0, 128))
    alien.draw()

def update():
    #Get access to variables set outside the function
[... 142 lines cut ...]

screen.fill(white)

pygame.draw.rect(screen, red, (0, 0, 100, 100), 0)

pygame.display.update()
2020-12-09 09:43:29,295 - mu.logic:1178(autosave) INFO: Autosave detected and saved changes in C:/Users/cturland/mu_code/pygameTest.py.
2020-12-09 09:43:37,083 - mu.modes.python3:202(stop_script) DEBUG: Stopping script.
2020-12-09 09:43:37,544 - mu.interface.panes:651(start_process) INFO: Running script: c:\users\cturland\mu_code\pygametest.py
2020-12-09 09:43:37,544 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2020-12-09 09:43:37,545 - mu.interface.panes:656(start_process) INFO: Command args: []
2020-12-09 09:43:37,545 - mu.interface.panes:683(start_process) INFO: Python paths set via C:\Users\cturland\AppData\Roaming\Python\Python36\site-packages\mu.pth
2020-12-09 09:43:37,546 - mu.interface.panes:714(start_process) INFO: Working directory: C:/Users/cturland/mu_code
2020-12-09 09:43:37,546 - mu.interface.panes:719(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\cturland\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-12-09 10:24:09,775 - mu.modes.python3:202(stop_script) DEBUG: Stopping script.
2020-12-09 10:24:10,196 - mu.interface.panes:651(start_process) INFO: Running script: c:\users\cturland\mu_code\pygametest.py
2020-12-09 10:24:10,196 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2020-12-09 10:24:10,196 - mu.interface.panes:656(start_process) INFO: Command args: []
2020-12-09 10:24:10,196 - mu.interface.panes:683(start_process) INFO: Python paths set via C:\Users\cturland\AppData\Roaming\Python\Python36\site-packages\mu.pth
2020-12-09 10:24:10,197 - mu.interface.panes:714(start_process) INFO: Working directory: C:/Users/cturland/mu_code
2020-12-09 10:24:10,197 - mu.interface.panes:719(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\cturland\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-12-09 10:24:13,030 - mu.modes.python3:202(stop_script) DEBUG: Stopping script.
2020-12-09 10:29:30,207 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\cturland\AppData\Local\python\mu\Logs\mu.log

Thank you for contributing to Mu! :-)

carlosperate, 2020-12-09:
Hi @cturland ,
Does the bottom pane show any errors when the play button is clicked?

cturland, 2020-12-09:
> Hi @cturland ,
> Does the bottom pane show any errors when the play button is clicked?

No, no error at all.

devdanzin, 2020-12-10:
Looks like you're in Python mode. Any chance they were in pygame mode and somehow didn't get the `AttributeError: 'pygame.Surface' object has no attribute 'surface'` error I get in Windows  on that mode? If I rename screen to something else in that mode the result is a blank screen.

cturland, 2020-12-10:
> Looks like you're in Python mode. Any chance they were in pygame mode and somehow didn't get the `AttributeError: 'pygame.Surface' object has no attribute 'surface'` error I get in Windows on that mode? If I rename screen to something else in that mode the result is a blank screen.

No definitely in normal Python 3 mode, with no error output. Had another class today and tried again. All students with MacBooks couldn't get it to work. I managed to get the output from their MacBook this time

2020-12-10 09:13:07,746 - root:113(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2

2020-12-10 09:13:07,747 - root:114(run) INFO: uname_result(system='Darwin', node='Divyas-MBP.isl.ch', release='19.6.0', version='Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64', machine='x86_64', processor='i386')

2020-12-10 09:13:07,747 - root:115(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Users/divyavenkatsridhar/Library/Application Support/mu/site-packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']

2020-12-10 09:13:07,747 - root:116(run) INFO: Language code: en_CH

2020-12-10 09:13:08,058 - mu.logic:587(__init__) INFO: Setting up editor.

2020-12-10 09:13:08,058 - mu.logic:609(__init__) INFO: Settings path: /Users/divyavenkatsridhar/Library/Application Support/mu/settings.json

2020-12-10 09:13:08,058 - mu.logic:610(__init__) INFO: Session path: /Users/divyavenkatsridhar/Library/Application Support/mu/session.json

2020-12-10 09:13:08,058 - mu.logic:611(__init__) INFO: Log directory: /Users/divyavenkatsridhar/Library/Logs/mu

2020-12-10 09:13:08,059 - mu.logic:612(__init__) INFO: Data directory: /Users/divyavenkatsridhar/Library/Application Support/mu

2020-12-10 09:13:08,067 - mu.logic:625(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero

2020-12-10 09:13:08,169 - mu.logic:1266(change_mode) INFO: Workspace directory: /Users/divyavenkatsridhar/mu_code

2020-12-10 09:13:08,170 - mu.logic:684(restore_session) INFO: Restoring session from: /Users/divyavenkatsridhar/Library/Application Support/mu/session.json

2020-12-10 09:13:08,170 - mu.logic:685(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': 0, 'y': 23, 'w': 1440, 'h': 775}}

2020-12-10 09:13:08,170 - mu.logic:706(restore_session) INFO: Loaded files.

2020-12-10 09:13:08,170 - mu.logic:710(restore_session) INFO: User defined environment variables: []

2020-12-10 09:13:08,170 - mu.logic:714(restore_session) INFO: Minify scripts on micro:bit? False

2020-12-10 09:13:08,218 - mu.logic:1266(change_mode) INFO: Workspace directory: /Users/divyavenkatsridhar/mu_code

2020-12-10 09:13:08,294 - mu.logic:742(restore_session) INFO: Starting with blank file.

2020-12-10 09:18:06,547 - mu.modes.microbit:154(flash) INFO: Preparing to flash script.

2020-12-10 09:18:06,547 - mu.modes.microbit:164(flash) DEBUG: Python script:

2020-12-10 09:18:06,547 - mu.modes.microbit:165(flash) DEBUG: b'# Write your code here :-)\n\nimport pygame\n\npygame.init()\n\nscreen = pygame.display.set_mode((640, 480))\nprint("MY NAME IS DIVYA")\n'

2020-12-10 09:18:06,584 - mu.modes.microbit:212(flash) INFO: Path to micro:bit: None

2020-12-10 09:18:06,584 - mu.modes.base:265(find_device) WARNING: Could not find device.

2020-12-10 09:18:06,584 - mu.modes.base:266(find_device) DEBUG: Available ports:

2020-12-10 09:18:06,585 - mu.modes.base:270(find_device) DEBUG: ['PID:0x0000 VID:0x0000 PORT:cu.Bluetooth-Incoming-Port', 'PID:0x0000 VID:0x0000 PORT:tty.Bluetooth-Incoming-Port']

2020-12-10 09:18:06,585 - mu.modes.microbit:217(flash) INFO: Serial port: None

2020-12-10 09:18:06,585 - mu.modes.microbit:218(flash) INFO: Device serial number: None

2020-12-10 09:18:10,734 - mu.interface.main:302(get_microbit_path) DEBUG: Getting micro:bit path: /Users/divyavenkatsridhar

2020-12-10 09:18:10,734 - mu.modes.microbit:229(flash) DEBUG: User defined path to micro:bit: /Users/divyavenkatsridhar

2020-12-10 09:18:10,734 - mu.modes.microbit:243(flash) INFO: Checking target device.

2020-12-10 09:18:10,736 - mu.modes.microbit:279(flash) WARNING: Could not detect version of MicroPython.

[... 52 lines cut ...]
2020-12-10 09:24:19,833 - mu.interface.panes:789(start_process) INFO: Runner: /Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3

2020-12-10 09:24:19,833 - mu.interface.panes:790(start_process) INFO: Args: ['-i', '/Users/divyavenkatsridhar/Desktop/testXmas.py']

2020-12-10 09:24:29,868 - mu.modes.python3:207(stop_script) DEBUG: Stopping script.

2020-12-10 09:24:30,080 - mu.interface.panes:691(start_process) INFO: Running script: /Users/divyavenkatsridhar/Desktop/testXmas.py

2020-12-10 09:24:30,080 - mu.interface.panes:693(start_process) INFO: Running with interactive mode.

2020-12-10 09:24:30,080 - mu.interface.panes:696(start_process) INFO: Command args: []

[... cut ...]
2020-12-10 09:24:30,081 - mu.interface.panes:789(start_process) INFO: Runner: /Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3

2020-12-10 09:24:30,081 - mu.interface.panes:790(start_process) INFO: Args: ['-i', '/Users/divyavenkatsridhar/Desktop/testXmas.py']

2020-12-10 09:24:39,696 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/divyavenkatsridhar/Library/Logs/mu/mu.log

2020-12-10 09:24:39,698 - mu.logic:163(installed_packages) INFO: Packages found: []

carlosperate, 2021-03-03:
I can replicate this in v1.1.0-alpha.2 in macOS Big Sur (x64).
Running Mu from source code seems to work fine though. We should re-test this once we have the next macOS release installer ready.

tmontes, 2021-03-15:
One data point:

* Just installed 1.1.0b2 on my macOS 10.14.6 (Mojave) system.
* Sample code above fine in Python mode.
* For "completeness" I added a "sleep forever" loop after 12.

What's not so nice:

* The PyGame window is being shown behined Mu's own window, so I had to go looking for it.

carlosperate, 2021-03-15:
> The PyGame window is being shown behined Mu's own window, so I had to go looking for it.

Yes, I had this as well the last time I tested it, but something made me think this was "fixed", but testing now t definitely still happens from source and the packaged version.
On the bright side pgzero windows open on top, so should be fixable.