mu-editor/mu #1180: I'm unable to flash any code from Mu under MacOS Catalina

andreagrandi, 2020-12-12:
Hello folks,

I'm trying to flash a simple script to my Microbit.

I'm using MacOS and I've followed the instructions here: https://codewith.mu/en/howto/1.0/install_macos

When I click the Flash button I get this error:

[image]

If I check the logs I see these errors:

```
2020-12-12 16:33:28,622 - mu.logic:649(restore_session) INFO: Loaded files.
2020-12-12 16:33:28,622 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2020-12-12 16:33:28,622 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2020-12-12 16:33:28,632 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/andrea/mu_code
2020-12-12 16:33:32,887 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2020-12-12 16:33:32,887 - mu.modes.microbit:249(flash) DEBUG: Python script:
2020-12-12 16:33:32,887 - mu.modes.microbit:250(flash) DEBUG: b'import microbit as m\n\nSENSITIVITY = 250\n\nwhile (True):\n    x = m.accelerometer.get_x()\n\n    if abs(x) < SENSITIVITY:\n        m.display.show(m.Image.HAPPY)\n    else:\n        m.display.show(m.Image.SAD)\n\n    m.sleep(100)'
2020-12-12 16:33:32,907 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: /Volumes/MICROBIT
2020-12-12 16:33:32,907 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem14102
2020-12-12 16:33:32,907 - mu.modes.base:228(find_device) INFO: Serial number: 9900023431864e45002710050000006300000000cc4d28bd
2020-12-12 16:33:32,907 - mu.modes.microbit:302(flash) INFO: Serial port: /dev/cu.usbmodem14102
2020-12-12 16:33:32,907 - mu.modes.microbit:303(flash) INFO: Device serial number: 9900023431864e45002710050000006300000000cc4d28bd
2020-12-12 16:33:32,908 - mu.modes.microbit:328(flash) INFO: Checking target device.
2020-12-12 16:33:33,478 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0', 'version': 'v1.7-9-gbe020eb on 2016-04-18', 'machine': 'micro:bit with nRF51822'}
2020-12-12 16:33:33,478 - mu.modes.microbit:351(flash) INFO: Board MicroPython: 0.0.1
2020-12-12 16:33:33,479 - mu.modes.microbit:353(flash) INFO: Mu MicroPython: 1.0.1
2020-12-12 16:33:33,479 - mu.modes.microbit:381(flash) INFO: Flashing new MicroPython runtime onto device
2020-12-12 16:33:33,495 - mu.modes.microbit:75(run) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2020-12-12 16:33:33,495 - mu.modes.microbit:532(flash_failed) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2020-12-12 16:33:33,507 - mu.interface.main:723(show_message) DEBUG: There was a problem flashing the micro:bit.
2020-12-12 16:33:33,507 - mu.interface.main:724(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2020-12-12 16:33:40,005 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/andrea/Library/Logs/mu/mu.log
```

The Mu editor has full disk access already:

[image]

What else am I missing? Thanks
 
## Versions

- MacOS: 10.15.6
- Mu: 1.0.3

ron333, 2020-12-16:
Take a look at: [Using Mu with micro:bit V2](https://support.microbit.org/support/solutions/articles/19000121431-using-mu-with-micro-bit-v2)
This is a temporary work around.  It worked for me.

andreagrandi, 2020-12-16:
Hey @ron333 thank you so much for your answer! Even if I ended up doing something different, your advice pointed me to the root of the problem. My Microbit still had a very very old version of the firmware (I think I got it early days in 2015 and never updated it!). I've followed the instructions here https://microbit.org/get-started/user-guide/firmware/ and updated it to the latest firmware. Now I'm able to flash the Python scripts to the microbit, even without running Mu with sudo!
Thanks 🙌

andreagrandi, 2020-12-16:
I spoke too quickly 😓
It worked a couple of times and then I'm still getting the same error.
There must be something else.

carlosperate, 2020-12-16:
Right, this seems related to Catalina and having rights to access the USB drive.

If you manually copy a micropython hex file into the micro:bit drive, and then in Mu click flash, does it work?
Every once in a while it might still encounter some internal errors and try to reflash Micropython, at which point it will fail again. But at least that way we know that transferring files via serial does work.

Hex can be found here: https://github.com/bbcmicrobit/micropython/releases/tag/v1.0.1

andreagrandi, 2020-12-16:
@carlosperate hi, if I copy that file you linked to me, the Microbit is being ejected at the end of the copy, then it's being mounted again. Even if I click Flash when the Microbit is again mounted, nothing happens and all I can see new in the logs is:
```
2020-12-16 20:50:42,304 - mu.logic:649(restore_session) INFO: Loaded files.
2020-12-16 20:50:42,304 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2020-12-16 20:50:42,304 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2020-12-16 20:50:42,316 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/andrea/mu_code
2020-12-16 20:52:08,272 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2020-12-16 20:52:13,994 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/andrea/Library/Logs/mu/mu.log
```
I'm a bit confused 🤔

andreagrandi, 2020-12-16:
p.s: not sure if it's related to the .hex you told me to flash, but now if I connect the USB cable to the Microbit, the device isn't being mounted anymore (I don't see it in Finder) 😓

carlosperate, 2020-12-16:
After flashing the MicroPython hex you would normally see something like this in the log:

```
2020-12-16 19:59:21,324 - mu.modes.microbit:243(flash) INFO: Checking target device.
2020-12-16 19:59:21,841 - mu.modes.microbit:247(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.1', 'version': 'micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2020-12-16 19:59:21,842 - mu.modes.microbit:266(flash) INFO: Board MicroPython: 1.0.1
2020-12-16 19:59:21,842 - mu.modes.microbit:268(flash) INFO: Mu MicroPython: 1.0.1
2020-12-16 19:59:21,842 - mu.modes.microbit:420(copy_main) INFO: Copying main.py onto device
```

But maybe it fails at the point it tries to access the USB drive, even if it doesn't use it, and that stops the step that would just send the file via serial.

carlosperate, 2020-12-16:
> p.s: not sure if it's related to the .hex you told me to flash, but now if I connect the USB cable to the Microbit, the device isn't being mounted anymore (I don't see it in Finder) 😓

Oh that's odd, are you using the same USB cable? Some cables are power only (no data).
If you press the reset button while plugging the USB cable into the micro:bit does a MAINTENANCE drive appear instead of the MICROBIT drive?

andreagrandi, 2020-12-16:
Yes I'm using the same cable I've been using all day. I'm aware of maintenance mode because it's the one I used today to upgrade the firmware (I followed these instructions https://microbit.org/get-started/user-guide/firmware/ ), but the problem is that if I now keep Reset button pressed and plug the cable, the Microbit is not being mounted at all. Nor in maintenance or normal mode. The led doesn't even flash. If I plug the cable without keeping Reset pressed, at least the led turns on but that's it.

carlosperate, 2020-12-16:
Right, can you see the serial port when the USB cable is plugged? Does the orange led blink at all when plugged? Does it stay on afterwards?

Could you also try these things? In some cases it has resolved some issues: restarting your computer, trying a different USB port, and trying a different computer.

andreagrandi, 2020-12-16:
I will try it tomorrow morning, I’m already “far” from the laptop and in bed, but thanks in the mean time, I will keep you posted

Scarica Outlook per iOS<https://aka.ms/o0ukef>

andreagrandi, 2020-12-17:
I've rebooted my Mac and now the Microbit is correctly mounted when I plug it!
This is what I see with `lsusb`:
```
Bus 020 Device 007: ID 0d28:0204 0d28 BBC micro:bit CMSIS-DAP  Serial: 9900000031864e4500271005000000630000000097969901
```
Now, I've tried to drag&drop the .hex file to the microbit. The file is being copied, the led flashes and in the end the Microbit is being unmounted and mounted again.

Is this the expected behaviour?

After all these operations, I've been able to use Mu to flash this code example to the Microbit:

```
from microbit import *

while True:
    display.scroll('Hello, World!')
    display.show(Image.HEART)
    sleep(2000)
```
and it worked.

I don't know if now it will keep working or not (yesterday it worked a couple of times, then stopped working, that's why I reopened the issue).

carlosperate, 2020-12-17:
Glad it's working again!

> Now, I've tried to drag&drop the .hex file to the microbit. The file is being copied, the led flashes and in the end the Microbit is being unmounted and mounted again.
> 
> Is this the expected behaviour?

Yep, when the file is finished transferring the USB drive has to unmount and remount so that OS doesn't show cached data (it's faking a MSD, so it needs to play some tricks to work correctly).

> I don't know if now it will keep working or not (yesterday it worked a couple of times, then stopped working, that's why I reopened the issue).

It's not uncommon to hit issues with the serial file transfer, at which point it will try the old method of flashing which causes issues in Catalina, so eventually it will show the same error.

Of course this is just a temporary workaround until we can release the next alpha.

Alternatively, if you are happy to run Mu from source code, it probably won't hit the permission issues: https://mu.readthedocs.io/en/latest/setup.html

carlosperate, 2021-03-04:
If it's okay I'll close this issue as there aren't any action we need to do on the Mu project, the next release should work in Catalina/Big Sur without issues.
However, please feel free to reopen if you are still having issues.

closed 2021-03-04 by carlosperate