mu-editor/mu #1181: If a micro:bit takes longer than 10s to flash it can fail to transfer file via serial

carlosperate, 2020-12-12:
Basically the write operation is fully blocking in Windows, but in macOS and Linux it returns before the operating system has been able to transfer the file completely.
In my test environment (macbook running Mojave) a file write/copy returns in a couple of seconds, while the hex transfer takes between 8 to 15 seconds, depending on many factors.

So due to the early write return in unix-based systems we have this in the micro:bit flash() method:
https://github.com/mu-editor/mu/blob/6aada27394cfb2bb44cd8bb15d5e3aec8b354979/mu/modes/microbit.py#L375-L381

Where it waits for 10 seconds and then tries to transfer the main.py via serial.

If the file takes longer than 10 seconds, then the serial connection will be attempted before the flashed MicroPython has a chance to start running.

I'll look into uFlash to see if I can make the hex copy blocking until the transfer is fully finished.

carlosperate, 2020-12-12:
Possibly related issues: https://github.com/mu-editor/mu/issues/1016

carlosperate, 2020-12-12:
I've opened this PR in uFlash: https://github.com/ntoll/uflash/pull/69

aivarannamaa, 2020-12-18:
Have you tried using `os.fsync` to synchronize with writing?  Seems to work well in Thonny: https://github.com/thonny/thonny/blob/6d40f22bcc6e51a80c0748c83be4275d1690180e/thonny/plugins/micropython/uf2dialog.py#L368

carlosperate, 2020-12-18:
No I didn't, thanks for the suggestion! I somehow didn't encountered this when looking into how to flush the file write, seems to work perfectly 👍 
I'll give it a few more tests with other DAPLink vernsions and operating system before updating the PR.

commit e959c74b (2020-12-16): microbit: Remove the timer for unix flashes, now uFlash always blocks.

Parent commit makes uFlash blocking in all platforms, so we can
remove the 10s timer used in Linux and macOS.
This was required because before the uFlash update it would
return before the micro:bit file transfer was complete.

This also fixes issues that happen when the file transfer takes
longer than 10s and MicroPython is not ready to enter raw REPL
when Mu tries to send main.py via serial.

Fixes #1181
commit a1a48ae1 (2021-01-07): microbit: Remove the timer for unix flashes, now uFlash always blocks.

Parent commit makes uFlash blocking in all platforms, so we can
remove the 10s timer used in Linux and macOS.
This was required because before the uFlash update it would
return before the micro:bit file transfer was complete.

This also fixes issues that happen when the file transfer takes
longer than 10s and MicroPython is not ready to enter raw REPL
when Mu tries to send main.py via serial.

Fixes #1181
closed 2021-01-29 by ntoll