mu-editor/mu #1182: Null ine in csv file

JLFra, 2020-12-15:
I have openned the csv file in data_capture after using the plotter. But there is a Null line inbetween each data line? A don"t understand this null line?

tjguk, 2020-12-15:
Looks like looks like `remove_plotter` in `modes/base.py` opens its output file without `newline=''`

ntoll, 2020-12-15:
Hmm... should a CSV file have `newline=""` set..? I assume this is a Windows newline related thing..? (Sorry about that.)

tjguk, 2020-12-15:
Don't think it's Windows-specific. (Although it might be)

https://docs.python.org/3/library/csv.html#id3

JLFra, 2020-12-19:
So did you found a solution? When we creat a csv file in micropython, it works perfectly? So why do you add a newline when you create the csv file? thanks to help

commit 42418443 (2020-12-19): Refactor slightly to support better testing of CSV output (#1199)

Fixes #1182
closed 2020-12-19 by tjguk