mu-editor/mu #1193: virtual environment process runner doesn't handle error exit

tjguk, 2020-12-17:
Within the `virtual_environment` module we've implemented a `Process` class which uses Qt's `QProcess` mechanism to interact with ui elements.

At present, while it does collect output from stdout & stderr, it doesn't detect failed completion and silently succeeds

tjguk, 2021-03-17:
This has been at least partly addressed. The `virtual_environment.Process:wait` function which acts for blocking runs does detect a failure condition on exit and raises a `VirtualEnvironmentError`. In addition, the process itself is set up to combine stderr/stdout which can then be read by the readData signal.

What I'm not 100% sure about is what happens when the process is run non-blocking and finishes with an failure condition. At present we have no code in `Process._finished` to take any action in those circumstances

carlosperate, 2021-04-19:
Does this ticket still need something to be implemented @tjguk?

tjguk, 2021-04-19:
Strictly, it still does. What I've done is to drop back to `subprocess.run` for the elements of setup of the venv -- and added logging from those. I think we're still due to add some more logging for the `QProcess`-based processes. Not so urgent, though.

carlosperate, 2021-10-06:
@tjguk is this still applicable?