mu-editor/mu #1197: New release/alpha?

gohai, 2020-12-18:
Hello mu-editor community,

I'll be teaching a physical computing course to beginners, starting in a few weeks, and I wanted to use MicroPython, since there are many other classes being offered to the students that build on the same language. Since we went with ESP32 boards from M5Stack we'll need an editor for REPL and to save their sketches to the ESP's flash.

Mu seems to be the most beginner-friendly, cross-platform editor in this space - however, I was surprised that there are currently no pre-built releases available that allow connecting an M5Stack ESP32 device out of the box. When I am trying with `1.1.0.alpha.2`, I am receiving the message that no attached device could be found.

There is support for the M5Stack's USB ID since https://github.com/mu-editor/mu/commit/3716f227f6a7f352e8c0847ccd901a9affd6921a from June of this year, but so far not even an Alpha release that incorporates this change. Is there any estimate when the next alpha release can be expected? Even I am struggling with getting mu to run from sources here on macOS (currently hitting an unmet dependency on libSDL) - this is nothing that one can expect beginner students could manage, which is unfortunate because this seems like it could be such a valuable resource to enable them to write code in (Micro-)Python.

Thank you,
gohai

tjguk, 2020-12-19:
@gohai -- yes, we're acutely aware of the lack of a new release. Apart from many other fixes / improvements, there have been two big stumbling blocks which have used up a lot of scarce volunteer time: the switch to a real runtime virtual environment; and a solution to packaging -- especially on MacOS where Apple have a tendency to move the goalposts.

The first of these has just landed in PR #1072. And the second is bearing fruit in a [sub-project](https://github.com/mu-editor/pup) by @tmontes. He's made great progress and can now package a minimal app on Windows & MacOS. But the big obstacle there is signing, without which users are confronted by scary warnings and/or difficult workarounds.

So, be assured that we are very definitely working as fast as our time allows towards a releasable version. I'll leave the other developers to comment on the device-specific issues of your post: they're not my area of expertise.

re the libSDL deps, I suspect you've been bitten by the fact that PyGame Zero, which we depend on, is pinned to a version of pygame < 2.0 (essentially: 1.9.6), and there are no suitable MacOS wheels for that version. So pip tries to build pygame from scratch -- which is notoriously a nightmare. I can see that the current Github version of pygame zero is now pinning pygame < 2.0 for Python < 3.8; and >= 2.0 for Python >= 3.8. I'm happy to talk you through getting that working with a dev version of Mu if that would take you anywhere.

Our gitter channel (also available via Matrix) is `mu-editor/general`. Feel free to drop in there for slightly more timely support.

gohai, 2020-12-20:
@tjguk Thank you very much for your time to reply to my report. You guys are doing amazing work - I am glad that this is very much on your radar, and I wish the project all the best in working towards the next release.

Let me go ahead and close this bug report, since it really doesn't add anything useful. I finally got the current tree to run here locally, and will perhaps see into modifying the USB identifiers that are hardcoded in a previously released version to work with the particular board the students will be using. (Most of them will be on Windows, so that should hopefully be less of an issue with notarization.) Thank you for your offer for help, I so appreciate this.

closed 2020-12-20 by gohai

dybber, 2020-12-20:
Hi @gohai, I'm in the same situation actually. I've been ramping up on using M5Stack's ESP32 based devices for teaching for the last half year, and we're about to test some new teaching material in the first weeks of february.

What we've found is that one of the development builds actually does work for most Mac OS X installation we've encountered (Mojave, Catalina, Big Sur). It apparantly was made at a time after the M5Stack support and before we upgraded PyQt to a version that doesn't support Big Sur.

It's this version from October 8th 2020:
https://s3-eu-west-2.amazonaws.com/mu-builds/osx/mu-editor_2020-10-08_22_06_master_e6adf68.zip

What's important to know is that it's not notarized/signed correctly, as we're still waiting for a new packaging system to come in place. This means that for users to open it they will have to do a right-click -> Open, or CTRL+click -> Open, to get it to launch. Otherwise they will get a "'mu-editor' cannot be opened because the developer cannot be verified." error.

If they still can't open it, your students can try running `sudo spctl --master-disable` in the terminal to disable the verification step, however I do recommend that they turn that back on again afterwards with `sudo spctl --master-enable`

Best,
Martin

gohai, 2020-12-25:
Thank you, @dybber. I'll give this version a try.