mu-editor/mu #1201: win_installer.PYNSIST_CFG_TEMPLATE hardcodes 3.6.8, but builds on 3.7 without warning

devdanzin, 2020-12-22:
While I couldn't actually identify any ill effects from building in 3.7 (it actually seems to download the right wheels), maybe we should preemptively check for the mismatch. Another option would be setting version to the actual Python running the build.

I can provide a fix if it's desirable, once we figure out what to do.

Found this while tinkering with the wheels we ship, had one of them unzipped and got:
`nsist.copymodules.ExtensionModuleMismatch: Found an extension module that will not be usable on Python 3.6.8`

tjguk, 2020-12-22:
@devdanzin I'll let @tmontes chime in on this as our Packaging Lead -- but I suspect/hope that the NSIS-based installer will very soon be overtaken by the new Pup-based work which @tmontes is trialling over there --> https://github.com/mu-editor/pup

tmontes, 2021-01-27:
@devdanzin,

Thanks for raising the issue. I have been working on `pup` as an external tool to package Mu (and eventually other Python GUI applications) which should simplify the whole process, hopefully making it both more robust and more flexible -- in particular supporting 3.7, 3.8, and 3.9 on both macOS and Windows, at least.

On big benefit of that will be the removal of (nearly) everything packaging related from the repository (we still want icons, for example) -- who doesn't want a tidier repo? :)

I'm in the process of merging in the `pup`-packaing bits under #1234. Windows will follow soon! :)

carlosperate, 2022-05-17:
We can close this issue as now we are using pup to build the Windows installers.