mu-editor/mu #1203: Format of settings files

tjguk, 2020-12-22:
At present our settings files are stored as JSON. With work going on via #1200 to standardise the use of settings files, some discussion has taken place as to whether a different format might be more useful, especially as it needs to be user-editable.

Obvious candidates suggested so far include JSON (no change); .ini; YAML; TOML

dybber, 2020-12-22:
Here's some thoughts that was shared on the Gitter/mu-editor/general chat today:

**.json**: Can encode whatever information we need. But not easy to edit by hand.

**.ini**: Sysadmins will know this, simple to edit. Not as good support for lists, maps, tuples and other composite values. Mostly a key/value store with subsections.

**.yaml**: might be too complex still for users (e.g. indentation is meaningful), but is still usereditable *and* supports more complex values like lists, tuples, maps

**.toml**: similar to .yaml, but perhaps getting traction in Python community when moving towards pyproject.toml

carlosperate, 2020-12-22:
I pretty much agree with everything summarised in @dybber's response.

-----

>.ini: Sysadmins will know this

Before this point I would have suggested TOML hands down, but I do wonder if .ini files might be better for sysadmins.

As far as user "editability" is concerned, I think TOML is so close to ini (while better define as a format) that I think it would be fine to use it. However, I do wonder if the extension does somehow matter for deployment or anything related to using it by a sys admin.

dybber, 2020-12-22:
I'm also leaning towards TOML. If we document it well enough, I don't think it will be a problem, as it's so similar to .ini for the most part. I don't think the file ending will be a problem other than school sysadmins might be less likely to open it at first, as they don't recognize the file type.

dybber, 2020-12-22:
What would the use case for a sysadmin look like? It would probably be to be able to install the exact same version of Mu on all machines of a school, as well as putting in a configuration file on all the machines as well. Perhaps also resetting configuration/session files every time the machine is booted, could be something they would want to do.