mu-editor/mu #1206: ver. 1.03 Serial Does Not Work with Seeedstudio XIAO

jake653, 2020-12-24:
Announced on the Seeedstudio forum  Jan 16, 2020
 https://forum.seeedstudio.com/t/circuitpython-on-the-seeeduino-xiao-working/7507

"Note that the recommended Mu editor won’t recognize the XIAO for serial communication yet unless you add the USB VID (0x2886) and PID (0x002f) to the Mu code (see circuitpython.py or adafruit.py in the Mu app), but works fine for writing/saving programs to it. The most recent (alpha) version of Mu may work if you used the SAMD21 Mini uf2, but you may need to add the VID/PID to the current stable version of Mu."

I poked around in the OSX app package contents, I could not locate what they were referring to.
  (I open Arduino simultaneously as a serial display)

steamfoundry, 2020-12-31:
I have this issue too.  I haven't tested a fix yet but the file they are referring to should be mu/modes/circuitpython.py.  
It has the table of device IDs to match against.

steamfoundry, 2020-12-31:
Tested that this fix works. It looks as simple as adding it to the table of devices.  So I edited my current installation (which lists it as adafruit.py) and upon loading Mu it recognized the device and Serial works.  
If someone else can put the fix in that would be great... otherwise I will once I can refresh myself on git.

steamfoundry, 2021-01-13:
Pull #1231  is ready for review please.

Also, if someone with the permission could tag this with Modes that would be appreciated.

tjguk, 2021-01-13:
I've added the tag, but it'll need to wait until one of the more device-oriented of the dev team is available to review.

carlosperate, 2021-01-23:
PR #1231 has been merged, thanks everyone!

closed 2021-01-23 by carlosperate