mu-editor/mu #1209: Tab switching during autosave causes incorrect tab name

laikas123, 2020-12-26:
If you are reporting a bug, we would like to know:

* What you were trying to do,
Press CTRL+Tab to switch to the next file open

* What steps you took to make this happen,
I made a change to one file, pressed CTRL+TAB to switch to the next file (without saving the first file) autosave occurs and if you press CTRL+TAB too fast the next tab of the file you are switching to gains the name of the previous file that runs its auto save

* What you expected to happen,
Switch to new file with its tab name intact
* What actually happened,
Switched to correct file but the tab name takes on that of the previous file
* Why this difference is problematic (it may not be a bug!),
Not a huge deal, but it did confuse me and make me wonder if my other file was overwritten
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Mu 1.0.2. 
Two python files open 
Python 3
Raspberry Pi 4 raspbian buster 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Here are the logs from before and after the error 

2020-12-26 10:22:31,594 - mu.logic:869(save_tab_to_file) INFO: Saving script to: /home/pi/mu_code/firestore_update_temp.py
2020-12-26 10:22:31,594 - mu.logic:870(save_tab_to_file) DEBUG: from google.cloud import storage
2020-12-26 10:22:31,603 - mu.logic:1176(autosave) INFO: Autosave detected and saved changes in /home/pi/mu_code/firestore_update_temp.py.
2020-12-26 10:22:38,377 - mu.logic:869(save_tab_to_file) INFO: Saving script to: /home/pi/mu_code/firestore_update_temp.py
2020-12-26 10:22:38,378 - mu.logic:870(save_tab_to_file) DEBUG: from google.cloud import storage

To be clear what happens is that basically if you are in one file and make a change but don't press save, and then you press CTRL+TAB and switch to the next tab, it seems the autosave function causes the tab of the file you are switching to, to gain the name of the previous tab that initiates its auto save.

Thanks for the awesome text editor :) and please let me know if I can further help.

carlosperate, 2020-12-26:
I think this is the same issue as https://github.com/mu-editor/mu/issues/595, that has been fixed in https://github.com/mu-editor/mu/pull/1063.

I assume you've installed Mu from the Raspbian install tool or via `apt-get install`/`apt install`? 
If you are comfortable installing and running Python programmes you could test if the problem is still present in the latest version from this repository.
https://mu.readthedocs.io/en/latest/setup.html

laikas123, 2020-12-26:
Thanks for the reply.

I followed the steps in the link you sent me specifically the section: "Raspberry Pi" 

This let me run version 1.1.0.alpha.3 and the issue does not appear in that version.

For what it's worth I had initially installed the software from the recommended software section on the raspberry pi:

[image]

Thanks for the quick reply!

carlosperate, 2020-12-26:
Thanks @laikas123, yes we haven't realised an updated version of Mu in a while, so the version in the raspbian repositories is a bit old. This fix should be included in the next release.

I'll close this issue as you have been able to confirm the fix is already in master, thanks!

closed 2020-12-26 by carlosperate