mu-editor/mu #1212: Easy to find settings.json file

carlosperate, 2020-12-31:
From the refactor in https://github.com/mu-editor/mu/pull/1200#issuecomment-752983477 it was found that we have a bit of code from the old PyInstaller era, to try to locate a settings.json file in an easy-to-find location for the user. In that case it was in the same folder as the app, as back then it was downloaded as a single file self-executable.

The reason this was originally useful was because navigating to the app_data location (path is printed in the log) can be a bit difficult for non-experienced users. Apart from having to find the path in the Mu settings, the way to "get there" changes depending on the OS version, and not everybody is familiar with this process.

We could either remove this feature or update it to a different location. If we chose a different location the `mu_code` folder is a good candidate.

If we do update this I would also suggest to make this a "read only" file, so we still save the settings/sessions to the original file, but Mu can read this file to replace the default. This also means the user would keep only the relevant settings they want to overwrite without having the noise of all the other data thee.

tjguk, 2020-12-31:
re the read-only idea -- the #1200 logic will only save back to the "user's" file settings which are loaded from it, or which are set programatically. (This is a slight simplification but near enough...). 

At present the only options set programatically are the "session" settings (theme, zoom, last loaded file etc.). And these are currently stored in a separate session.json file. There's room for discussion about merging those but the #1200 PR keeps them separate.

tjguk, 2021-01-03:
This issue seems relevant, albeit for historical purposes:

https://github.com/mu-editor/mu/issues/358