mu-editor/mu #1213: ESP32 file connection impossible after boot.py modification

street-grease-coder, 2020-12-31:
Everything worked beautifully with the mu editor (version 1.1.0.alpha2) and my esp32. REPL worked, file transfer worked. Then I modified the boot.py (silly code below), this 'killed' the file transfer (the code worked as expected though). Is this a known issue and I should have RTFM more intensely? I fixed this by removing boot.py from REPL.

Fix:
`
import uos
uos.remove('boot.py')
`

boot.py file that produced the issue:
```
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
#import webrepl
#webrepl.start()

import time
from machine import TouchPad, Pin
from machine import UART
from machine import PWM
from machine import ADC
from machine import DAC

uart = UART(1, 9600,tx=12,rx=14)                         # init with given baudrate
uart.init(9600, bits=8, parity=None, stop=1) # init with given parameters

while True:
    time.sleep_ms(10)
```

Log: first I tested the REPL: this worked. Then the file system toggle: did not work.

`2020-12-31 18:44:56,047 - mu.modes.base:290(toggle_repl) INFO: Toggle REPL off.
2020-12-31 18:44:56,533 - mu.modes.base:261(find_device) INFO: Found device on port: COM10
2020-12-31 18:44:56,534 - mu.modes.base:262(find_device) INFO: Serial number: 0001
2020-12-31 18:44:56,541 - mu.modes.esp:181(toggle_files) INFO: Toggle filesystem on.
2020-12-31 18:44:58,699 - mu.modes.base:435(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
  File "C:\Program Files\Mu\pkgs\mu\modes\base.py", line 432, in ls
    result = tuple(microfs.ls(self.serial))
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 188, in ls
    ], serial)
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 139, in execute
    raw_on(serial)
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 100, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 72, in flush_to_msg
    raise IOError('Could not enter raw REPL.')
OSError: Could not enter raw REPL.
2020-12-31 18:44:58,705 - mu.interface.main:763(show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.
2020-12-31 18:44:58,705 - mu.interface.main:764(show_message) DEBUG: None
2020-12-31 18:45:00,618 - mu.logic:1161(show_admin) INFO: Showing admin with logs from C:\Users\Felix Widmer\AppData\Local\python\mu\Logs\mu.log
2020-12-31 18:45:00,619 - mu.logic:163(installed_packages) INFO: Packages found: []
2020-12-31 18:46:36,350 - mu.logic:1192(show_admin) INFO: No admin settings changed.
2020-12-31 18:46:37,263 - mu.modes.esp:185(toggle_files) INFO: Toggle filesystem off.
2020-12-31 18:46:45,760 - mu.modes.base:261(find_device) INFO: Found device on port: COM10
2020-12-31 18:46:45,760 - mu.modes.base:262(find_device) INFO: Serial number: 0001
2020-12-31 18:46:45,768 - mu.modes.esp:181(toggle_files) INFO: Toggle filesystem on.
2020-12-31 18:46:47,903 - mu.modes.base:435(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
  File "C:\Program Files\Mu\pkgs\mu\modes\base.py", line 432, in ls
    result = tuple(microfs.ls(self.serial))
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 188, in ls
    ], serial)
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 139, in execute
    raw_on(serial)
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 100, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "C:\Program Files\Mu\pkgs\mu\contrib\microfs.py", line 72, in flush_to_msg
    raise IOError('Could not enter raw REPL.')
OSError: Could not enter raw REPL.`

devdanzin, 2020-12-31:
Thanks for reporting this!

That `while True:` block would cause the script to never terminate in most Python variants, not sure about the ESP. Can you try the same code without that block and then just that block and nothing else?

street-grease-coder, 2020-12-31:
same thing without the while loop. I would not expect it to be blocking since I have REPL access and can press CTRL+C. It's the ESP32 Devkit version I should add

dybber, 2021-01-02:
How did you get to run the 
```import uos
uos.remove('boot.py')
```
Was that through picocom or similar?

Currently, my thought is that there's nothing we can do different in Mu to fix this problem, but if you somehow can abort that program from e.g. picocom, please let us know how. My understanding is that can't actually abort `boot.py` before it's finished, I haven't tested recently, but I think I bricked an ESP32 doing something similar. 

In general you shouldn't add an infinite loop to the boot.py, your main application logic should go in to `main.py`. The `boot.py` file should mainly be used to set up various small configuration options that you need across all applications you might want to run.

dhalbert, 2021-01-02:
> My understanding is that can't actually abort `boot.py` before it's finished, I haven't tested recently, but I think I bricked an ESP32 doing something similar.

I'm not sure if you considered it permanently bricked, but you can always get it into DFU mode and erase all of flash with `esptool.py` or similar. That should get it back to a usable state.

street-grease-coder, 2021-01-02:
@bricket: definitely not! It works as usual. It's also highly reproducible on my dev1 kit. I imported uos using the REPL in Mu, the code editor that was suggested somewhere (the latest alpha version). I'm guessing uos was included

dybber, 2021-01-02:
@dhalbert: Yes, what I meant was that I had to flash it with new firmware.

@street-grease-coder: Okay, so you were able to open the REPL, it was only the Files dialog that didn't work? Perhaps we need an extra abort somewhere. However, that would then always abort boot.py from executing when opening "Files".

street-grease-coder, 2021-01-03:
@dybber That's right. I think a boot.py abort would be ideal (especially since you can just re-run boot.py by just restarting the device), a manual confirmation dialogue would be a cherry on top

dybber, 2021-02-06:
I have just confirmed that this is an issue, and I guess the issue must be somewhere in the `microfs` library. 

If the boot.py script is running in the REPL, then a CTRL-C can abort it, but for some reason this doesn't happen when listing files in the Files pane. Here the `microfs` library is used, and `microfs` does send appropriate keyboard interrupt signals, so I'm not really sure what is the problem. I guess this will require a deeper dive some day.

Here's the execute function from `microfs`: https://github.com/ntoll/microfs/blob/96fe2575fb55c1a4235cd099e8f6ec25a210f68b/microfs.py#L122

dybber, 2021-02-06:
Oh wait, it's probably because microfs sends a soft-reboot after the keyboard interrupt, thus restarting the execution of the boot.py script:
https://github.com/ntoll/microfs/blob/master/microfs.py#L147

I guess I should rewrite that execute function using what I learned from writing the ESP-mode.

ntoll, 2021-02-14:
Happy to have a PR and will speedily test / merge. :-)

dybber, 2021-02-17:
What is the history behind the following lines of code?
https://github.com/ntoll/microfs/blob/master/microfs.py#L92-L100

It doesn't seem necessary to do a soft-reboot to go into raw-mode.

The soft-reboot triggers the boot.py file on the device to run, and for certain devices e.g. M5Stack devices, EduBlocks (see https://github.com/mu-editor/mu/issues/985#issuecomment-605528011), the boot.py runs for a very long time (e.g. connecting to WiFi)

Did we have any problems without these lines?

ntoll, 2021-02-18:
My memory is fuzzy, but I believe the soft-reboot always ensures the device is in a known good state. But this was written before we had `boot.py` style startup scripts. Perhaps it's due for removal..?

dybber, 2021-02-18:
Alternatively, it should be: soft-reboot -> pause -> send additional interrupts to stop the boot scripts from running?

dybber, 2021-02-18:
That's actually what we do when we run a script, so perhaps we just need some additional keyboard interrupts after the soft-reboot:
https://github.com/mu-editor/mu/blob/52cb622575f60290318892c42ebc5a6a8e0665b9/mu/modes/base.py#L173-L180

carlosperate, 2021-02-18:
The soft boot also ensures there is enough memory available to run the script to transfer files. Otherwise if the user code is at the memory limits it will fail.

dybber, 2021-02-18:
Okay, I just tried adding two extra `KEYBOARD_INTERRUPT` to `raw_on` in microfs. Line 101 in this commit: https://github.com/dybber/microfs/commit/fb229d366a8bb2dbd3bb549b0a51ff54edf75425

This makes file listing work, by effectively stopping the soft-reboot before the boot.py scripts are finished running (as expected).

However, this is NOT nice, as the boot.py scripts might be very necessary for correct functionality on devices such as M5Stack, where the boot.py adds to the Python path:
```
import sys
sys.path.append('flowlib/lib')
```
if these lines are not executed in the M5Stack boot.py script, the built-in modules for the M5 sensors etc. will not be found.

So we can either:
 - don't do the reboot, and perhaps run into the problem carlos just mentioned (not enough memory)
 - do the reboot and cancel the boot.py scripts, and run into the problem that MicroPython modules might not be found, or other boot code os is not executed

A potential ugly middle way, is to try and wait a bit on the boot.py script, to see if it will finish, and then send the `KEYBOARD_INTERRUPT`'s after the timeout