mu-editor/mu #1214: Image is not loading from Images

Mauliklm10, 2021-01-02:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

Mauliklm10, 2021-01-02:
2021-01-02 10:50:51,375 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.3
2021-01-02 10:50:51,375 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-JFCQFB1', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-01-02 10:50:51,375 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\askma\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2021-01-02 10:50:51,375 - root:115(run) INFO: Language code: en_US
2021-01-02 10:50:51,576 - mu.logic:543(__init__) INFO: Setting up editor.
2021-01-02 10:50:51,576 - mu.logic:562(__init__) INFO: Settings path: C:\Users\askma\AppData\Local\python\mu\settings.json
2021-01-02 10:50:51,576 - mu.logic:563(__init__) INFO: Session path: C:\Users\askma\AppData\Local\python\mu\session.json
2021-01-02 10:50:51,576 - mu.logic:564(__init__) INFO: Log directory: C:\Users\askma\AppData\Local\python\mu\Logs
2021-01-02 10:50:51,576 - mu.logic:565(__init__) INFO: Data directory: C:\Users\askma\AppData\Local\python\mu
2021-01-02 10:50:51,607 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2021-01-02 10:50:52,804 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\askma\mu_code
2021-01-02 10:50:52,804 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\askma\AppData\Local\python\mu\session.json
2021-01-02 10:50:52,804 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'pygamezero', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2021-01-02 10:50:52,804 - mu.logic:649(restore_session) INFO: Loaded files.
2021-01-02 10:50:52,804 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2021-01-02 10:50:52,804 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2021-01-02 10:50:52,853 - mu.logic:680(restore_session) INFO: Starting with blank file.
2021-01-02 10:50:52,884 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\askma\mu_code
2021-01-02 10:51:16,556 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: C:\Users\askma\mu_code
2021-01-02 10:51:18,502 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: 
2021-01-02 10:52:37,692 - mu.logic:702(new) INFO: Added a new tab.
2021-01-02 10:52:39,973 - mu.logic:1107(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-01-02 10:52:42,077 - mu.logic:1112(select_mode) INFO: New mode selected: python
2021-01-02 10:52:42,099 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\askma\mu_code
2021-01-02 10:52:47,118 - mu.logic:1107(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-01-02 10:54:26,136 - mu.logic:702(new) INFO: Added a new tab.
2021-01-02 10:56:22,816 - mu.modes.python3:225(toggle_repl) INFO: Toggle REPL on.
2021-01-02 10:56:22,832 - mu.modes.python3:57(start_kernel) INFO: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\askma\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2021-01-02 10:56:22,832 - mu.modes.python3:62(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2021-01-02 10:56:22,994 - traitlets:256(start_kernel) DEBUG: Starting kernel: ['C:\\Program Files\\Mu\\Python\\pythonw.exe', '-m', 'ipykernel_launcher', '-f', 'C:\\Users\\askma\\AppData\\Local\\Temp\\tmpndqyvizy.json']
2021-01-02 10:56:23,010 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:50177
2021-01-02 10:56:23,471 - traitlets:140(shell_channel) DEBUG: connecting shell channel to tcp://127.0.0.1:50174
2021-01-02 10:56:23,471 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:50174
2021-01-02 10:56:23,474 - traitlets:152(iopub_channel) DEBUG: connecting iopub channel to tcp://127.0.0.1:50175
2021-01-02 10:56:23,474 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:50175
2021-01-02 10:56:23,476 - traitlets:164(stdin_channel) DEBUG: connecting stdin channel to tcp://127.0.0.1:50176
2021-01-02 10:56:23,477 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:50176
2021-01-02 10:56:23,478 - traitlets:176(hb_channel) DEBUG: connecting heartbeat channel to tcp://127.0.0.1:50178
2021-01-02 10:56:24,055 - traitlets:212(_handle_history_reply) DEBUG: Received history reply with 621 entries
2021-01-02 10:56:28,956 - mu.modes.python3:229(toggle_repl) INFO: Toggle REPL off.
2021-01-02 10:56:34,813 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: C:\Users\askma\mu_code
2021-01-02 10:56:44,914 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: C:/Users/askma/OneDrive/Desktop/Tutoring/Python/Space_attack.py
2021-01-02 10:56:44,914 - mu.logic:715(_load) INFO: Loading script from: C:/Users/askma/OneDrive/Desktop/Tutoring/Python/Space_attack.py
2021-01-02 10:56:44,914 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2021-01-02 10:56:44,914 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
[... cut ...]
2021-01-02 14:04:53,576 - mu.interface.panes:719(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\askma\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2021-01-02 14:04:55,917 - mu.modes.pygamezero:143(stop_game) DEBUG: Stopping script.
2021-01-02 14:04:57,603 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\askma\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2021-01-03:
Thanks for the report @Mauliklm10. Can you also post the error you are getting?

Can you show us the c:\users\askma\onedrive\desktop\tutoring\python\ folder? There should be a images folder there?
If you click the "Images" buton in Mu, what is the full path to the folder it opens? (screenshot would be good)

Mauliklm10, 2021-01-03:
[image]
This is the screenshot

Mauliklm10, 2021-01-03:
[image]

carlosperate, 2021-01-03:
right so the images folder should be inside c:\users\askma\onedrive\desktop\tutoring\python\ , but for some reason Mu hasn't managed  that correctly.

Could you try the latest alpha to see if that works? https://codewith.mu/en/download

Mauliklm10, 2021-01-03:
Since that is a newer version of mu will have to launch another app? or is that just a software update? I downloaded the alpha version but when I open mu it is still the same app i.e. will the alpha install a new mu app?

carlosperate, 2021-01-03:
If both versions were installed in the default location the latest should have replaced the previous.
The version is in the window title, does it still say 1.0.3?

Mauliklm10, 2021-01-03:
Yes it still says 1.0.3

Mauliklm10, 2021-01-03:
Ok so now I got 1.1 to work but the images is now an issue.
[image]

carlosperate, 2021-01-03:
If you place the alien image there it should work now.

Mauliklm10, 2021-01-03:
Alright got it to work. Thank you so much!!!

closed 2021-01-03 by Mauliklm10