mu-editor/mu #1215: Hello, I have an m5stack device, are you going to support this type of device?

fraramon, 2021-01-03:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

dybber, 2021-01-26:
Hi @fraramon, yes, it's actually already added to Mu a year ago, but we haven't been able to release a new version since then due to packaging issues, that we had to get resolved. We hope to do a beta release soon with all the new additions. I'm closing this.

closed 2021-01-26 by dybber

fraramon, 2021-01-26:
Hello @dybber,
Ok, thank you very much...