mu-editor/mu #1219: F3 is already used to "Flash your code onto the micro:bit" in that mode

tjguk, 2021-01-05:
ref https://github.com/mu-editor/mu/pull/1164

The (commonly-used) choice of F3 used in that PR conflicts with the use of F3 in micro:bit mode to re-flash

dybber, 2021-02-20:
@devdanzin @tjguk Could we use CTRL+SHIFT+F for find again?

tjguk, 2021-02-20:
I'd be perfectly happy with that myself.

dybber, 2021-02-20:
I guess it's a bit hard to type keyboard shortcut for small fingers though.

ntoll, 2021-02-20:
This is, alas, a minefield and we're likely to get all sorts of clashes depending on operating system and activity being shortcut-ed.

When I added shortcuts (mostly for accessibility reasons), I went for the most widely used shortcut for a given existing action. If such an action didn't exist (for example, re-flash the micro:bit) then I made an educated-guess for the best "currently unused" key.

FWIW - full list of shortcuts are here: https://codewith.mu/en/tutorials/1.0/shortcuts (and mode specific ones could be revised if they cause a clash).

silverr0b, 2021-08-13:
Hi,

I've been testing using MU editor for a blind student, when trying to Flash a file to a microbit using F3 nothing happens.

csworcs, 2022-11-14:
Hi,
I appreciate that this is closed but I have version 1.1.1 of Mu and f3 still doesn't work.