mu-editor/mu #1224: Ability to remove essential packages (pip, setuptools, wheel)

dybber, 2021-01-07:
Hi all,

I just played around with the new master branch, which contains the new virtualenv based backend for managing installed packages. However, I found a way to remove pip, which makes Mu crash when you try to launch it again.

Here are the steps:
 - Install some third party package (e.g. packagedummy)
 - Close mu
 - Open mu again
 - Open the Third Party Packages, now pip, setuptools and wheel are added to the list:

[image]

 - Remove pip from the list and click Ok:

[image]

 - Try opening the admin dialog again

Mu crashes with the following error:

```
Traceback (most recent call last):
  File "/Users/dpr964/Development/micropython/mu/mu/virtual_environment.py", line 261, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/dpr964/Development/micropython/mu/mu/logic.py", line 1394, in show_admin
    baseline_packages, user_packages = venv.installed_packages()
  File "/Users/dpr964/Development/micropython/mu/mu/virtual_environment.py", line 519, in installed_packages
    for package, version in self.pip.installed():
  File "/Users/dpr964/Development/micropython/mu/mu/virtual_environment.py", line 267, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages
```

tjguk, 2021-01-07:
Thanks, @dybber  -- that's just the kind of thing I knew we'd have to addressing once the branch had landed. One probably simple workaround here is to include pip, setuptools etc. in our list of baseline packages so they will be excluded from the user's view.

dybber, 2021-01-07:
If you install a package that depends on PyQt5, you can also remove PyQt5. However, I can still launch it afterwards. But maybe that's because I have PyQt5 in the development environment? I'm not sure exactly what happens, but it is somehow removed, but still there:

[image]

Perhaps the real issue here is that it installed that dependency instead of just using PyQt5 version already available with mu?

(This was after installing the package `AgentsPy` which were developing at the moment for agent-based simulation like Netlogo - possibly to be renamed to `netturtle` to make it closer to `turtle`)

dybber, 2021-01-07:
I think all packages installed as dependencies shouldn't be shown in the dialog, but should be automatically removed when no longer required by any packages. It should only list packages the user has installed, or perhaps a greyed out list of all packages installed, so the user cannot change it

tjguk, 2021-02-03:
I was starting to look at the first element of this -- excluding pip, setuptools etc. And I couldn't reproduce it. In fact, the baseline packages held in venv.json include pip, setuptools & wheel. So there's no need to exclude them specifically.

I agree that it would be good to track "topline" packages separately from their dependencies. I'll try to put a PR together

carlosperate, 2021-08-04:
I believe Tim has already resolved this by storing the "base" dependencing in the venv.json file and not displaying them to the user, so they cannot be removed:

[image]

I'll close this as resolved, but feel free to reopen if this is still an issue 👍 

Thanks everyone!

closed 2021-08-04 by carlosperate