mu-editor/mu #1230: REPL does not work after using "Files" (esp32, Windows 8.1)

ttttio007, 2021-01-08:
PC OS: Windows 8.1
Device: NodeMCU ESP32 (CP210x) https://ae01.alicdn.com/kf/HLB1NvxzQMHqK1RjSZJnq6zNLpXav/ESP-12E-CH340-NodeMcu-v3-ESP32-D1MINI-Lua.jpg_Q90.jpg_.webp

It seems the device is in boot-load mode after using "Files".
If I do physical reconnect my device, REPL works again.
So I think it is because serial connection should be close after using "Files", or before using REPL.

But I'd like to note, this problem does not touch other my esp32-devices (I use esp32 in different form-factors).

Logfile does not containt any useful information when the problem occurs.

I tried several Mu versions, including mu_2020-12-13_17_19_master_3a80d0f_32bit.exe

[image]

dybber, 2021-02-01:
Woops, I thought to assign myself, as I usually take care of ESP32 related issues with Mu.

However, I don't own a Windows computer, so I wouldn't be able to debug this unfortunately. This is a Windows issue, I've seen it before on my students computers, but I don't have a good idea for a solution, and as mentioned I can't really debug it without a Windows PC.

dybber, 2021-02-01:
Might be a hardware issue because a so called strapping pin is pulled low on reboot. In this case MTDO (GPIO15) and/or GPIO5. But I guess that's not the case?

Or it might be something to do with DTR (Data Terminal Ready) and RTS (Request To Send) signals, which apparently can be used to trigger Download mode: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#automatic-bootloader. I'm not sure why DTR/RTS they would be different in this situation on Windows.

MatAtBread, 2022-03-25:
I have this issue. RTS is asserted by default in the Windows USB serial driver I have nasty workarounds (using RealTerm, or a TCP redirector and accessing the device from WSL), but it'be really nice to have an option to disable RTS (it's clearly possible via the Windows serial API, tho I don't know the details).