mu-editor/mu #1232: Cannot run under macOS Big Sur

mattpackwood, 2021-01-09:
Error when trying to run in Big Sur v11.1

Error is: `“mu-editor” can’t be opened because Apple cannot check it for malicious software.`

[image]

ajs256, 2021-01-16:
Right-click ➡️ Open ➡️ Open Anyway (or similar) ➡️ Enter your password

It's because the developers are a small team and can't afford Apple's $99/yr dev licenses.

filipejsbrandao, 2021-01-16:
If you click Ok it will close the message and Mu won't open. Then if you go to System Preferences, under Security & Privacy > General tab you can open the app.

mattpackwood, 2021-01-16:
I tried "Command - Click" to get the "Open anyway", selected that, the app appears in the Task Bar, but I have to Force Close it as it does not do anything...

ajs256, 2021-01-17:
Then you're probably running into #1147. Take a look at my instructions in https://github.com/mu-editor/mu/issues/1147#issuecomment-731609561 to build from source.

ajs256, 2021-01-17:
Actually, try [the alpha version](https://github.com/mu-editor/mu/releases/tag/1.1.0-alpha.2) first.

filipejsbrandao, 2021-01-17:
It worked! @ajs256 THANKS!!

dybber, 2021-02-01:
This is actually a duplicate of #940, as it's an old issue we also had with Catalina. It will be fixed when #940 is fixed, so I'm closing this. The fix would be part of the next beta release.

closed 2021-02-01 by dybber