mu-editor/mu #1235: Unable to stop the code from running

intarpit, 2021-01-11:
Using the alpha version of Mu.
Running micro-python on ESP32.
Using window 10.

I was able to start my code using the "play button" from the menu bar of the ide but the play button doesn't switch to stop.
It still shows the play button and when clicked again it will re-run the code. 
There is no way to stop the code from running.

steamfoundry, 2021-01-31:
My own experience is limited with micro-python still. Someone else please chime in if I'm wrong. 

Did the older versions of Mu show a stop button?  (If it did, ignore my next part please )

I thought that the behaviour for Micro-python is that the "play" button loads your code onto the device and starts the execution.  It would be up to the code or the ESP32 to decide when to stop executing. 

What are you trying to accomplish with the stop button?   You may just need to use serial monitor to break out of the current execution and access REPL.

dybber, 2021-02-01:
We can't detect when a program on a microcontroller is finished, we could perhaps add a separate Abort/Stop-button, which would be visible all the time in the main toolbar, or as part of the REPL-pane.

To abort a program, you send CTRL-C keypress to the REPL, so that's what such a button should do, and what you do right now to stop a program. Move focus to the REPL, and push CTRL-C.

steamfoundry, 2021-02-01:
Thanks @dybber .

One other caveat for @intarpit is that some things won't stop when the execution does.  Things such as LEDs (especially Neopixel type LEDs) may hold the last state until power is removed or they are cleared by the code. 

Speaking as someone who forget this, then spent 2 days troubleshooting code before realizing it was an intermittent connection on the data in for an LED.

dybber, 2021-02-01:
If you want to completely reset the device, you can do a Soft Reboot by pushing CTRL-D in the REPL. (Hard reset is usually done with a hardware button)