mu-editor/mu #1236: In Linux we may exit Mu with a KeyboardInterrupt if a QProcess fails to launch and we kill PID 0

devdanzin, 2021-01-11:
In some places, like Web mode, we launch a `QProcess` and later get its PID so we can send it a `SIGINT`:

```python
    def stop_server(self):
        if self.runner:
            try:
                pid = self.runner.process.processId()
                os.kill(pid, signal.SIGINT)
            except Exception as ex:
```

However, if somehow that `QProcess` isn't running (e.g. `flask` isn't available/failed to launch) it returns 0 as PID and that kills Mu on Linux with a `KeyboardInterrupt` (because we guard against `Exception`, but only `BaseException` would catch `SIGINT`).

Same code on Windows doesn't result in Mu closing, instead we get `OSError: [WinError 87] The parameter is incorrect` in the log.

The code below can be used to check what happens in other OSes:
```python
import os, signal
try:
    os.kill(0, signal.SIGINT)
except Exception as e:
    print("All is well, we caught:", e)
```

This isn't something likely to happen, but if it ever does we never get the chance to log an error. I'd like to add a check that `pid` isn't zero nor the same as the current process (`os.getpid()`) so we don't quit Mu and maybe raise an informative exception which would get logged. Thoughts?

dybber, 2021-02-21:
I guess this is linked to #1236, I think it would be better to use QProcess.kill or QProcess.terminate to terminate process. Perhaps first attempt with QProcess.terminate (sends SIGTERM), and if the process doesn't close gracefully we force it with QProcess.kill (sends SIGKILL)

dybber, 2021-02-21:
We could add a function like this to the PythonProcessPane:

```python
def stop_process(self):
    if self.process:
        self.process.terminate()
        terminated = self.process.waitForFinished(10)
        if not terminated:
            self.process.kill()
            self.process.waitForFinished()
        self.running = False
```

and then use that in both Python-mode, Web-mode and Pygame-mode? 

Then it will try to terminate, wait up to 10 seconds, if it doesn't close within 10 seconds, it will kill the process instead.

dybber, 2021-02-21:
Unless there are any specific reasons why we use:
```python
pid = self.runner.process.processId()
os.kill(pid, signal.SIGINT)
```
for the Web-mode?

ntoll, 2021-02-21:
Memory is hazy, but yes, there is/was a reason I did it via `os.kill`. I'll have to revisit the code to remember why.

dybber, 2021-02-22:
> Memory is hazy, but yes, there is/was a reason I did it via `os.kill`. I'll have to revisit the code to remember why.

Okay, now I just attempted to do it without `os.kill` just to see what happens, and I'm still not sure why it wouldn't work with the `QProcess.kill`, but perhaps my PR will help you remember?

ntoll, 2021-07-11:
Fixed now #1319 is merged.

closed 2021-07-11 by ntoll