mu-editor/mu #1237: cant seem to start in circuitpython environment

mkarikom, 2021-01-11:
trying to open mu editor in circuitpython causes crash on ubuntu 20.04, mu 1.0.1

steps to reproduce
* $ mu-editor
* select "circuitpython" in the dialog

the program immediately crashes when the selection is clicked in the dialog

error:
```
(base) me@ubuntu:~$ mu-editor
Logging to /home/me/snap/mu-editor/common/.cache/mu/log/mu.log
Gtk-Message: Failed to load module "gail"
Gtk-Message: Failed to load module "atk-bridge"
Gtk-Message: Failed to load module "canberra-gtk-module"
Traceback (most recent call last):
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/logic.py", line 1075, in select_mode
    self.change_mode(new_mode)
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/logic.py", line 1112, in change_mode
    self.modes[mode].workspace_dir()))
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/modes/adafruit.py", line 98, in workspace_dir
    mount_output = check_output(mount_command).splitlines()
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 693, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
PermissionError: [Errno 13] Permission denied

```

ajs256, 2021-01-23:
Looks like there're permission problems. Try:
* sudo-ing Mu
* using `ls -l` to see what group owns the port, and then adding yourself to that group.
The port should look something like `/dev/ttyACM0`.

devdanzin, 2021-01-23:
The crash is a bug, we should catch that error. I think the right thing to do is show an error message explaining that there was a permission error. If so, how detailed should it be? We can show simple text, or include port name and permissions, or propose a workaround.

steamfoundry, 2021-01-24:
I believe a simple error message including the port is a great start. 

I think documentation is a better place to put troubleshooting steps. Instructions can be fine tuned as users try them and new edge cases are found. 

It also means the instructions can be referred back to outside the program, bookmarked, quoted, links shared, and so on.  

Think of a classroom with a standardized PC image that runs into this issue for 30 students.  Will the instructor have an easier time emailing a link or a screenshot of the error message?

devdanzin, 2021-01-26:
In the end, the PermissionError is coming from a simple "mount" call with no arguments, so the port permissions shouldn't matter (but sudoing might fix it). PR #1246 attempts to fix this and display an error message as discussed, any review, suggestions or requests are most welcome.

@mkarikom: Thank you for this report, it helped to identify a class of crashes that could happen from CircuitPython or any mode that raised an exception while finding a workspace dir. The specific issue you're having seems to be triggered by one of the mount commands (either "mount" or "/sbin/mount", with no arguments) resulting in a Permission Denied error. Can you try those in a shell and see what happens? I've proposed a patch that should fix it, let me know if you'd like help to edit your Mu install to add the fix there.

@steamfoundry: Thank you for helping design this, it helped to focus on fixing first, now we should polish and then document the issue. If you can take a look at the proposed PR, or, later, at the proposed docs, any input will be greatly appreciated.

closed 2021-01-31 by ntoll

mkarikom, 2021-02-01:
Ok, thanks for the update, I'll give it a shot.

commit 90860904 (2021-01-31): Merge pull request #1246 from devdanzin/fix_permission_denied_crash

Avoid circuitpython permission denied crash, fixes #1237