mu-editor/mu #1239: Python3(Jupyter) REPL "Kernel died, restarting"

litodemesa, 2021-01-13:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

litodemesa, 2021-01-13:
Just try to run REPL
[log_file.txt](https://github.com/mu-editor/mu/files/5810126/log_file.txt)

devdanzin, 2021-01-13:
Thank you again! Can't test this right now, but from the log I see Mu is working with a `python3.6.zip` and the kernel is trying to start from a `python3.7` executable. I believe Mu shouldn't be invoking any external Python install.

devdanzin, 2021-01-13:
OK, I can reproduce this by passing a bad PYTHONPATH in the Python3 Environment tab (the one close to the log view). If your case is as simple as mine, just deleting a PYTHONPATH line from that tab will solve it.

Still not sure how exactly it gets recorded there, it somehow happens on its own when I repeatedly alternate between running one of two Mu versions (1.1.0-alpha.2 and current git) that were installed to different places, using different Python versions. Probably related to running the script or opening the REPL.

devdanzin, 2021-01-14:
It seems simply making Mu unable to receive PYTHONPATH from Editor.envars (both for REPL and Run actions) fixes the issue I can reproduce. 

We can also check whether the Python used to start the kernel is the same as `venv.interpreter` and use the latter if they differ (only requires a simple subclassing of `QtKernelManager`). Or just always use it.

Very similar to #944 (except in that case there was no Python3.7 installed) and #584 (version mismatch, but on MacOS).

dybber, 2021-01-26:
I will reopen this, at the fix is still not merged

dybber, 2021-02-01:
Fix merged. Closing

closed 2021-02-01 by dybber