mu-editor/mu #1242: Review pup-packaged venv interactions

tmontes, 2021-01-20:
Leaving this here, following up on last Monday's quick chat and recent experiments:

* Current (preliminary) status and `pup`-packaged Mu is not bundling venv-wheels.
* This leads to downloads on first launch.
* These downloads are downloaded into the `mu/wheels` package.
* They are not removed from disk on uninstall -- on Windows!

Something for us to track and, I figure, something that can easily be fixed between me and @tjguk.

tmontes, 2021-02-02:
THOUGHTS:

* Moving this to 1.1.0b2 -- 1.1.0b1 has sailed away, already.
* Partially addressed in #1264 -- partially because if ever Mu downloads wheels, it's downloading them to a non-safe place (can be read-only, for example, in system-wide installations).

carlosperate, 2021-11-04:
Is this already resolved @tmontes?

tmontes, 2021-11-04:
> Is this already resolved @tmontes?

Well... Yes and no. Let me be clear:

Yes, as long as we ensure that we always ship packaged Mu with the bundled wheel ZIP file.

If ever that is not the case, then there's the possibility that setting up Mu's venv might fail (that, however, might not be considered a pup vs venv interaction!). Case where it fails:

1. We ship Windows MSI *without* the bundled wheels ZIP.
   Not likely becase our packaging automation is downloading and creating the wheels ZIP beforehand.
2. Users install the MSI in "all users mode" -- needing elevated privileges.
   Not user friendly to do nowadays, but possible - see [this `pup` doc snippet](https://pup.readthedocs.io/en/latest/platform_notes/windows_msi.html#per-user-vs-per-machine-installs)
3. When lauching Mu for the first time, Mu will detect that it has no wheel ZIP [here](https://github.com/mu-editor/mu/blob/022de5872be0498b46b8c7267e2d5f99a68ac2f1/mu/virtual_environment.py#L857).
   It will then download the wheels into the ZIP, writing it to a non-user-writable directory [here](https://github.com/mu-editor/mu/blob/master/mu/wheels/__init__.py#L154).
   Thus, failing.

Future proofing such odd scenario should be simple, having Mu create the wheel ZIP file in a temp directory when downloading wheels from PyPI at venv setup time.

Thoughts, @tjguk ?

carlosperate, 2021-11-05:
> Future proofing such odd scenario should be simple, having Mu create the wheel ZIP file in a temp directory when downloading wheels from PyPI at venv setup time.

This is also useful for some issues we are seeing on Linux where users `pip install` in the global Python environment with `sudo` and then running Mu without `sudo` fails: https://github.com/mu-editor/mu/issues/1634