mu-editor/mu #1243: Support for Raspberry Pi Pico and other RP2040 powered devices

whaleygeek, 2021-01-21:
Is there any plan to support the MicroPython and CircuitPython that is on the various RP2040 boards that have been launched today?

Mu has a number of 'modes' supporting CircuitPython, MicroPython on micro:bit, Python 3 on Raspberry Pi and also Pygame, and it is pre-installed on the Raspberry Pi OS. 

It would be great if it knew about the API's on these other devices so that the code completion worked, and if the REPL detected the device, flashed it, and allowed access to the MicroPython filing system.

ZanderBrown, 2021-01-21:
> Is there any plan to support the MicroPython and CircuitPython that is on the various RP2040 boards

There is more than one? - and based on gitter chatter this morning: Yes!

whaleygeek, 2021-01-21:
Yes, quite a list of new devices already and it is only launch day today!!

Raspberry Pi Pico:  https://www.raspberrypi.org/blog/raspberry-pi-silicon-pico-now-on-sale/
Adafruit Feather RP2040: https://www.adafruit.com/product/4884
Pimoroni Pico System: https://shop.pimoroni.com/products/picosystem
Pimoroni Tiny 2040: https://shop.pimoroni.com/products/tiny-2040

ZanderBrown, 2021-01-21:
Ordered myself a pico earlier, now I want all of them! :-)

carlosperate, 2021-01-22:
Yeah, the goal is to support it, although the question right now would be if it would be supported inside an existing mode or if we should create a new mode specific for this chip.

The main reason to create a new mode would be to provide raspberry pi specific API to the autocompletion, looking at the docs it looks like most of it is the default MicroPython API, but there is one extra module (that I've seen so far, there might be more) `rp2` that contains the PIO functionality.

steamfoundry, 2021-01-26:
So does anyone know where to find the device IDs for new hardware?  Besides waiting for one to show up in the mail of course.
It would be nice to bundle them into a single pull.

whaleygeek, 2021-01-26:
@steamfoundry The last time I looked into this (when we were launching micro:bit V1 many years ago), if the device is not USB logo certified (as micro:bit isn't, and I don't think any of the Pi's are either), then the only place you can get that is from the manufacturer, or by enumerating the physical device.

This is not to say anything about the certification of the device, just that if you don't pay the licence fee and pass the USB certification tests required to put the official USB logo on the box of the product, then the central USB authority list of VID and PID is not updated with these devices (but self managed by the manufacturer).

The manufacturer (in this case, Raspberry Pi, Adafruit, BBC) may maintain their own list of VID and PID, but there is no obligation to make this public.

Short answer - ask a friend to enumerate it and get the VID and PID from them :-)

ZanderBrown, 2021-01-26:
Ah, thought I'd posted this here on the 22nd:

```py
        (0x2E8A, 0x0005, None, "Raspberry Pi Pico"),
```
Matches the official micropython build

carlosperate, 2021-01-31:
Implemented in https://github.com/mu-editor/mu/pull/1261 🎉

closed 2021-01-31 by carlosperate