mu-editor/mu #1253: Display information during Mu initialisation

devdanzin, 2021-01-29:
Sometimes Mu takes a long time to start and/or hangs, and so far we don't have any reporting about what's going on. There's some consensus that adding [some sort of splash screen](https://gitter.im/mu-editor/dev?at=6012ff3d84e66b7f7eb7efc9)[0] that shows progress is the way to go. However, details like where the information comes from (usual logging? added informational messages?) as well as how to display it (single line? special text widget? should it auto-close on app start?) don't seem to be completely fleshed out.

So let's use this issue to figure out what this information display should be able to do.

PR #1252 goes with just forwarding the usual log messages into a barebones splash screen, then positioning this "infoscreen" below the classic splash screen. The infoscreen doesn't close on a timer like the splash screen does, but currently will also close on mouse click. 

It looks like this (n.b.: logging was artificially slowed otherwise it's too fast):
[image]

[0] - I've read this in a couple of places, but can't find more references ATM

dybber, 2021-01-29:
I don't know if the log is useful to users, it could perhaps also be a progress bar in the bottom of the splash screen. Just an idea...

devdanzin, 2021-01-30:
Drawing a progress bar can be done as text, polygons or images, either in a separate splash screen or somewhere in the current one. The issue I have is how to calculate progress: unless we set arbitrary points in the code to update the progress bar, we don't know how much there is to be done (e.g. the venv might be already set up or yet to be installed).

But setting arbitrary points in the code, maybe with useful text messages to display in a single line, can be a nice way to do it.

ntoll, 2021-01-31:
@devdanzin hey hey... this is going in a good direction. I'd like to get this into the release ASAP, so will have a play around.

ntoll, 2021-01-31:
Orthogonal (an animated splash screen that doesn't [yet] display textual output about Mu initialization): #1260

carlosperate, 2021-02-11:
Can this be considered done? Or is there more work required?

carlosperate, 2021-02-12:
Recommendation from @tjguk in https://github.com/mu-editor/mu/issues/1295#issuecomment-778007163:

> Not sure whether to make this comment here or elsewhere, but in the light of the issues we're seeing with startup, I really think we'll need to go to some kind of text output mode such as PoC-ed by @devdanzin. As it stands there are too many things which can silently fail as part of the venv startup. Obviously we're trying to make that more robust so it doesn't fail and/or recovers well. But with no output to the user whatsoever it seems like a recipe for poor UX.

ntoll, 2021-02-12:
This definitely needs more work and Steve is on the design assets we'll need. I've looked into adding dynamic textual content, and I think it's definitely do-able. I hope to have something bodged together over the weekend for folks to poke with a stick. :+1:

I heartily agree with @tjguk - given our beginner focus, it would be troublesome if they can't see things are going on, on first start, or if there's an error.

carlosperate, 2021-02-12:
Yeah, I agree that having more info shown in the splash screen will be useful to.

One thing to take in consideration is that some of the issues we've found during startup will throw a useful error once, and then fail differently. For example, the venv could have an issue installing something and throw that error, but in the next Mu run the problem presents itself as different dependency not being installer (since pip did not install everything).

Having a splash screen showing the different steps is definitely better, but as far as debugging user reports go, unless we get the full log we will still struggle. Specially since the splash will likely close as soon as an error is thrown and the user won't even have a chance to see/capture what the last message in the splash screen was.

ntoll, 2021-02-14:
So I want to get our "first start" story lined up (something for tomorrow's Mu-moot perhaps?).

On my mind:

* Splash screen design (currently being ruminated on by the genius of the ever generous @hawkz).
* Displaying the stdout of the venv setup on first start.
* What to do in the event of a failure during any of the above.
* Migrating from older versions of Mu (i.e. we shouldn't paint ourselves into a corner, plan ahead and make it easy for us to update to 1.2 or 2.0 without killing folks work/settings/REPL etc).
* Could some of the first start steps actually be moved into the installers created by `pup` @tmontes ..?

I want to make first steps in displaying `stdout` this afternoon once I've cleared my email backlog.

dybber, 2021-02-19:
I just want to link this issue with #498, as when we've finalized the new splash screen, we'll have to check whether the issues in #498 are still resolved.

carlosperate, 2021-03-16:
Can we consider this ticket done?

carlosperate, 2021-04-07:
I'll close this as implemented, but please feel free to reopen, or open a new issue, if there is still more work to do (outside of bug fixes, which should go as new bug reports).

closed 2021-04-07 by carlosperate