mu-editor/mu #1256: Cannot install Mu on Mac - Error message that Apple can't check it for malicious software

rgjoeys, 2021-01-30:
If you are reporting a bug, we would like to know:

* What you were trying to do: Install Mu on my MacBook Air with Catalina macOS
* What steps you took to make this happen: Successfully downloaded the software from codewith.mu then tried to open it, but received an error message
* What you expected to happen: Open the app
* What actually happened: Error message received: 
[image]

* Why this difference is problematic (it may not be a bug!): cannot access the software
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running: Used the Mac OSX Installer from Mu download page (tried several times ranging from 27 - 30th Jan, 2021). Using MacBook Air with Catalina macOS

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

ajs256, 2021-01-31:
That's because the small team of devs can't pay the US$99/yr fee to get the app "notarized" by Apple. Right-clicking the app and choosing "Open", rather than double-clicking the app, should work.

dybber, 2021-02-01:
This is a known problem, which will be fixed in the next beta release (coming soon!). @tmontes had to develop our own packaging tool to package Mu, to allow us to package and notarize Mu. The main issue is #940, which I will refer to and then close this issue.

closed 2021-02-01 by dybber

rgjoeys, 2021-02-01:
Many thanks for the prompt feedback. I have successfully been able to open Mu :)