mu-editor/mu #1265: Should mock the logger in all tests

carlosperate, 2021-01-31:
Or reconfigure it to not log into the Mu log file, otherwise we get the logs from the tests (with a lot of errors printed) in the Mu log.

tjguk, 2021-02-21:
Just adding to this, as I've gone around mocking out a few venv instances for #1316. I think the right thing here is to have a test-global mechanism (probably via some imported fixture?) which creates a temp DATA_DIR for a test run. This would, by extension, include log files, settings files and the venv. (Although the latter already uses a tempfile setup).

The only caveat I'd give is that it can be useful to view the log outputs from a test run, which sits akwardly with an self-destructing temp area.

tjguk, 2021-02-21:
... ahh, such as we already have for the QtApp instance via the conventional `conftest.py` file. I might try to add something there to try it out.

tjguk, 2021-02-21:
I've included a global session fixture over in #1316 to redirect the data dir to a py.test temporary folder. It seems to work except that there's a random venv.ini file showing up in the standard location!

carlosperate, 2021-11-04:
This has been resolved at some point, the test logs are not capture in the normal Mu log any more.
Thanks Tim!

closed 2021-11-04 by carlosperate