mu-editor/mu #1270: Mu-editor can't get correct VID:PID on the Linux on the Chromebook

gamygin, 2021-02-02:
Hello,

I'm trying to use Mu-editor to work with [Adafruit Feather Bluefruit Sense](https://learn.adafruit.com/adafruit-feather-sense) running Circuitpython.
I have a Chromebook with Linux (Debian 10 (buster)) enabled on it ([here are details](https://chromium.googlesource.com/chromiumos/docs/+/HEAD/containers_and_vms.md)).
The issue is that Mu-editor doesn't see the board connected to the laptop ([mu_log.txt](https://github.com/mu-editor/mu/files/5914512/mu_log.txt)).
At the same time, I can connect to the board with "screen /dev/ttyACM0" and use REPL

It looks like Mu-editor can't read the "VID:PID".

1. Here is the outcome of the script which uses QSerialPortInfo ([QSerialPortInfo_Script.py.txt](https://github.com/mu-editor/mu/files/5914612/QSerialPortInfo_Script.py.txt))
```
~$ python3 ./Circuitpython/QSerialPortInfo_Script.py 
ttyACM0: VID:PID 0000:0000 ()
```

2. Here is the the outcome of the pySerial:
```
~$ python3 -m serial.tools.list_ports -v
/dev/ttyACM0        
    desc: Feather Bluefruit Sense - CircuitPython CDC control
    hwid: USB VID:PID=239A:8088 SER=5F7F72B6AB5B3CDC LOCATION=1-1:1.0
1 ports found
```

3. Here is the outcome of "lsusb" command:
```
~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 017: ID 239a:8088  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
PS
It was originally discussed here: https://gitter.im/mu-editor/general?at=6019ba1f9238c531ad1d9606

dybber, 2021-02-02:
I tagged it 3rd party, as your output from the `QSerialPortInfo_Script.py` indicates that this is a problem with Qt. We'll have to reach out to them about the problem.

dybber, 2021-02-03:
Before we can report it, we also need the Qt version you tested with, when running the `QSerialPortInfo_Script.py`. Could you run `pip list` and get the version of `PyQt5` that is listed. For example on my installation: 

```PyQt5                         5.15.2```

gamygin, 2021-02-03:
Hm, interesting, it does not pop up in the output:
```
~$ pip list
Package                 Version
----------------------- ----------
Adafruit-Blinka         5.13.1
Adafruit-PlatformDetect 2.28.0
Adafruit-PureIO         1.1.8
adafruit-python-shell   1.2.3
anytree                 2.8.0
appdirs                 1.4.3
args                    0.1.0
astroid                 2.4.2
beautifulsoup4          4.9.0
bs4                     0.0.1
certifi                 2019.11.28
chardet                 3.0.4
clint                   0.5.1
distlib                 0.3.0
filelock                3.0.12
future                  0.18.2
idna                    2.9
iso8601                 0.1.13
isort                   5.7.0
lazy-object-proxy       1.4.3
mccabe                  0.6.1
pgzero                  1.1
pip                     21.0.1
pipenv                  2018.11.26
pyftdi                  0.52.9
pygame                  2.0.1
pylint                  2.6.0
pyserial                3.5
pyusb                   1.1.1
PyYAML                  5.3.1
requests                2.23.0
serial                  0.0.97
setuptools              52.0.0
six                     1.14.0
soupsieve               2.0
timeloop                1.0.2
toml                    0.10.2
urllib3                 1.25.8
virtualenv              20.0.13
virtualenv-clone        0.5.3
wrapt                   1.12.1
```

gamygin, 2021-02-03:
So I installed it manually (pip install PyQt5).
Tried after all the steps above (launch Mu-editor, pySerial, QSerialPortInfo) with the same outcomes as in original bug.

Here is the new outcome of "pip list":
```
~$ pip list
Package                 Version
----------------------- ----------
Adafruit-Blinka         5.13.1
Adafruit-PlatformDetect 2.28.0
Adafruit-PureIO         1.1.8
adafruit-python-shell   1.2.3
anytree                 2.8.0
appdirs                 1.4.3
args                    0.1.0
astroid                 2.4.2
beautifulsoup4          4.9.0
bs4                     0.0.1
certifi                 2019.11.28
chardet                 3.0.4
clint                   0.5.1
distlib                 0.3.0
filelock                3.0.12
future                  0.18.2
idna                    2.9
iso8601                 0.1.13
isort                   5.7.0
lazy-object-proxy       1.4.3
mccabe                  0.6.1
pgzero                  1.1
pip                     21.0.1
pipenv                  2018.11.26
pyftdi                  0.52.9
pygame                  2.0.1
pylint                  2.6.0
PyQt5                   5.15.2
PyQt5-sip               12.8.1
pyserial                3.5
pyusb                   1.1.1
PyYAML                  5.3.1
requests                2.23.0
serial                  0.0.97
setuptools              52.0.0
six                     1.14.0
soupsieve               2.0
timeloop                1.0.2
toml                    0.10.2
urllib3                 1.25.8
virtualenv              20.0.13
virtualenv-clone        0.5.3
wrapt                   1.12.1
```

dybber, 2021-02-03:
Okay, so that's the newest version of Qt. So it's not something they have already fixed if we upgrade Qt (Mu doesn't currently use the newest Qt, to be a bit backward compatible).

In the meantime I've figured that when we report it to Qt, we probably have to report it with a C++ version of the example program.

Here's the same program as before, written in C++. I hope I can get you to try and build and run it.

**qserialportinfo_issue.cpp**
```
#include <QDebug>
#include <QtSerialPort/QSerialPortInfo>

int main()
{
    qDebug() << "Qt version: " << QT_VERSION_STR;
    Q_FOREACH(QSerialPortInfo port, QSerialPortInfo::availablePorts()) {
        QString vid = "0x" + QString::number(port.vendorIdentifier(), 16);
        QString pid = "0x" + QString::number(port.productIdentifier(), 16);

        QString line = QString("%1 VID:PID=%2:%3 (%4)").arg(port.portName(),
                                                            vid,
                                                            pid,
                                                            port.manufacturer());
        qDebug() << qPrintable(line);
    }
}
```

And a project file for qmake: 

**mu_issue1270.pro**
```
TEMPLATE = app
TARGET = mu_issue1270
INCLUDEPATH += .
QT += core serialport

# Input
SOURCES += qserialportinfo_issue.cpp
```

Then you have to run:
```
$ qmake mu_issue1270.pro
```

This generates a `Makefile` in the current directory and then you'll have to run `make` to compile the program:

```
$ make
```

The output will then be a "mu_issue1270" executable, which is a little different from platform to platform (e.g. .exe on Windows, .app on Mac), so I hope you can figure out to run it. (I'm on a Mac)

dybber, 2021-02-03:
This is just to make sure it's a Qt issue and not a PyQt issue

gamygin, 2021-02-03:
OK, I did the steps ()

Here is the outcome of the (`mu_issue1270`) execution:
```
~$ ./mu_issue1270
Qt version:  5.11.3
ttyACM0 VID:PID=0x0:0x0 ()
```

dybber, 2021-02-03:
Okay, that's an older version of Qt, but let's just try and post it to the Qt developers and let's see what they say.

dybber, 2021-02-03:
Ok, I've created an issue on their bugtracker: https://bugreports.qt.io/browse/QTBUG-90877

Thanks for the help so far, my guess is they will probably ask us you to try and upgrade Qt to a more recent version, but I'm not sure which version they want us to try with, so let's wait and see what they respond.

gamygin, 2021-02-03:
OK, Thank you for your help!

carlosperate, 2021-02-03:
>Thanks for the help so far, my guess is they will probably ask us you to try and upgrade Qt to a more recent version, but I'm not sure which version they want us to try with, so let's wait and see what they respond.

The PyQt5 version used for the Python test programme was 5.15.2, which is the most recent version available, so it looks like the issue is present there as well.

gamygin, 2021-02-03:
We'll see. I'm afraid this issue could be common for Chromebook users.
Hope we'll be able to crack it

dybber, 2021-02-22:
Hi all

I just had the thought that an alternative solution, which has been discussed previously in other issues, is to show all attached devices, even though some of them are not recognized by Mu. In this way the user will be able to select the device and use it, even though we can't directly recognize it. I will create a discussion, in the new Github discussions area, to discuss whether that's something we want to do or not.

Btw. looking at the Qt-bugtracker status, my guess is that what they really want is to figure out is whether it's a Qt-bug or a bug in some underlying system that Qt relies on. I'm not sure if I can help move it forward though, as I don't understand the ChromOS/Linux details that they are asking about