mu-editor/mu #1274: GitHub Action macOS 11 Big Sur is having problems

carlosperate, 2021-02-03:
We had a few big sub jobs complete successfully, but the majority hang.

Looks like GH is aware of this issue and have remove this environment into a "private preview": https://github.com/actions/virtual-environments/issues/2486

>_What does “private preview” mean?_
This limitation will live on organization level (personal accounts are considered as organization too), so if you are using macos-11.0 somewhere in your organization — you will still be able to use it. But this pool will be inaccessible for new organizations that never used it before.

So they will still run within the mu-editor org, but right now it's mostly broken.
We might want to consider removing them from our CI until the environment is a bit more stable.

tjguk, 2021-02-03:
Glad we're not the only ones having problems with Big Sur!

Happy for us to disable those builds/tests for now.

tjguk, 2021-02-03:
Since I've got an active PR over at #1273, do you want me to disable the Mac OS 11 build there?

carlosperate, 2021-02-03:
Yeah, sounds good 👍

carlosperate, 2022-05-17:
The CI now uses macOS 10.15 and the latest available (11.x right now), so this is no longer an issue.

closed 2022-05-17 by carlosperate