mu-editor/mu #1276: Ability to upgrade Third-party packages

dybber, 2021-02-04:
Currently, if you try to reinstall a third party package (by removing the line, and adding it back again), it will instead just install the old version from cache and not pull new versions from PyPi.

tjguk, 2021-02-04:
It's easy enough to specify `--upgrade` on all `pip install`s. Do we think that's the right thing to do?

ntoll, 2021-02-04:
Yes :+1:

dybber, 2021-02-04:
Yeah, that would be great I think.

carlosperate, 2021-02-04:
We might want to be careful using the new pip resolver, as upgrading everything is bound to end up with incompatible version clashes at some point, specially since that list is flat and doesn't differentiate between modules we care about and sub-dependencies.

For example, with the original pygamezero (before our fork), if Mu upgraded pygame it can detect a conflict but it will only print a warning out, install the new version, and exit 0.
Maybe there is a pip command to check before installing, but unless it's able to do a single check with a list of all the packages we want to upgrade (which might be possible, I don't know), then we are more likelyto break some environments than if we are conservatives about the updates.

dybber, 2021-02-17:
> We might want to be careful using the new pip resolver, as upgrading everything is bound to end up with incompatible version clashes at some point, specially since that list is flat and doesn't differentiate between modules we care about and sub-dependencies.

It would be really nice to have "Upgrade all packages"-button though. I lack that right now while teaching remotely. I've made a package, which some high school students and teachers use. Now when I'm fixing bugs in it, it's hard for them to upgrade and get the version without those bugs.

aroberge, 2021-03-05:
The behaviour I observe on Windows is that it does upgrade to the last version. I have done this many times in working to test Friendly-traceback and improve its compatibility with Mu.

4020jd, 2021-03-23:
My apologies for being so far behind the curve here. Brand new to the Mu Editor and I can not even get to the point of having 

> Currently, if you try to reinstall a third party package (by removing the line, and adding it back again), it will instead just install the old version from cache and not pull new versions from PyPi.

My apologies for being so far behind the curve here. Brand new to the Mu Editor and I can not even get to the point of knowing how to add a third party package. Please help. I have read other threads and they all say to go to the gear icon. I see three tabs when this is done, but no clue beyond that.

Thank you for your patience in advance. Please be easy with me on the terminology lol

carlosperate, 2021-03-24:
Hi @4020jd, make sure you are in the Python mode before clicking the cog to open the settings. More info can be found in: https://github.com/mu-editor/mu-editor.github.io/blob/6e6a2c78d1187ebe8960693d0c33137801cc628b/en/tutorials/1.1/pypi.md

4020jd, 2021-03-24:
Hi carlosperate, thank you so very much! Yup I was indeed in Python mode...and was clicking on the cog...was even going to the Third Party tabs and entered my desired package (pandas), but for some reason when the "Third Party Package Status" popped up, it was completely empty for minutes at a time.

This made me wonder if there were commands to type in the dialog box...such as pip install pandas or what have you.

The provided directions worked like a charm! Thank you so very much, my debt of gratitude to you. I am not sure why I didn't find those phenomenal instruction on my own...I wish I could have saved you the time. Many thanks!

By the way, pandas is working great, no errors have been thrown in my code. Have a great day!

carlosperate, 2021-03-24:
@4020jd glad you got it working, that was likely this issue, which has been resolved and will be fixed in the next release: https://github.com/mu-editor/mu/issues/1358

I'll hide these comments are they divert from the original topic discussed in this thread.