mu-editor/mu #1279: ESP Firmware flasher: add button to install esptool dependency

dybber, 2021-02-06:
Currently the esptool is not included with Mu, but is a dependency if users wants to flash firmware onto ESP-devices.

Thus users are required to install the package through the "Third Party Packages" tab:
[image]

It would be nice if we instead had a button for "Install esptool", especially if #1278 is merged and the Third Party Packages tab is hidden when in ESP-mode.

tjguk, 2021-02-06:
TBH is there any harm in simply including it in the list of wheels we ship with the installer? (Or the runtime deps in `setup.py`?). I think that was what I was planning to do when I saw you'd added the functionality, only I clearly forgot!

ntoll, 2021-02-14:
@dybber @tjguk so the solution is simply to bundle the `esptool` wheel in the installers..? :shipit: :+1:

dybber, 2021-02-14:
That's the simplest solution, yes, but it will also make Mu bigger for everyone. If we add an install-button, it will only be bigger for the very few installations that requires this feature (teachers using ESP32)

tmontes, 2021-02-14:
> That's the simplest solution, yes, but it will also make Mu bigger for everyone. If we add an install-button, it will only be bigger for the very few installations that requires this feature (teachers using ESP32)

..the installers are rougly 100 MB and AFAICT, from [`esptool` at PyPI](https://pypi.org/project/esptool/#files) the additional cost is under 200 kB -- even though I have no experience with ESP32 devices (but curious about them!), the benefit of not needing to have "internet connectivity" clearly outweights the minimal additional payload, I say.

My 2c.

ntoll, 2021-02-14:
What @tmontes says. :+1:

dybber, 2021-02-15:
Okay great, I will make the change.

dybber, 2021-02-17:
> ..the installers are rougly 100 MB and AFAICT, from [`esptool` at PyPI](https://pypi.org/project/esptool/#files) the additional cost is under 200 kB -- even though I have no experience with ESP32 [...]

Did those 200 kB include dependencies? It looks like it brings in several other packages:
```
    install_requires=[
        'bitstring>=3.1.6',
        'cryptography>=2.1.4',
        'ecdsa>=0.16.0',
        'pyserial>=3.0',
        'reedsolo>=1.5.3,<=1.5.4',
    ],
```

None of them are probably very big, I just don't like introducing so many dependencies, but we can perhaps wait until we find a way to make modes more pluggable to look at this.

carlosperate, 2021-02-19:
The cryptography module has moved to rust and could be a bit annoying about having wheels in all platforms, although I think PyWheels might have been updated recently to support it, so maybe we are good now.

carlosperate, 2021-02-19:
Okay, looking at the Arm Debian CI it installs `python3-cryptography` from the debian packages as a dependency of something else (not clear what exactly as it needs to install a few different python packages via apt, including PyQt5), and since the venv uses `--system-site-packages` that satisfies the requirement.

dybber, 2021-02-19:
I think I forgot to link the PR adressing this #1301.

> Okay, looking at the Arm Debian CI it installs `python3-cryptography` from the debian packages as a dependency of something else (not clear what exactly as it needs to install a few different python [...]

Are we ready to merge the PR then, or do you have any further concerns? I still think in the future we should make modes behave like plugins, and wait until a mode is installed before we install it's dependencies, so for me this is a temporary fix.

carlosperate, 2021-02-19:
Added a note to the PR, sorry it's taken so many rounds!

commit f634e70a (2021-02-19): Merge pull request #1301 from dybber/add_esptool_dependency

Add esptool as dependency to simplify user experience in ESP-mode (fixes #1279)
closed 2021-02-19 by dybber