mu-editor/mu #1281: Mac .app won't run on big sur. Only the terminal version works

ImolaS3, 2021-02-06:
I wanted to learn Python on the Adafruit M4. I downloaded the Mu package but it will not run. 

It simply hangs before any screens are shown, just an icon appears on the task bar

I found a report on here that suggested the issue was fixed, but that is not the case as making it work via a terminal download and then running a bunch of terminal commands I need to find, copy, paste, and run,  is not the same as a working app

Please can you fix the problem with the actual app itself (mu-editor.app) as i have no desire to have to run the terminal and put ina a bunch of commands every time i want to use a program. 

many thanks

DrNebin, 2021-02-08:
I am having the same issue.

ntoll, 2021-02-14:
We hope to have an OSX installer for a new version of Mu coming out very soon. I'll add a link to this issue for the new installer when it's been released.

carlosperate, 2021-03-04:
I'll close this issue assuming the issues listed here can be resolved with the steps listed in https://github.com/mu-editor/mu/discussions/1349

If you are still having problems not resolved with those steps please do reopen this issue and we can continue debugging it.

closed 2021-03-04 by carlosperate