mu-editor/mu #1282: Windows users can't find the application launcher after installation

dybber, 2021-02-07:
This is a problem I've seen several times, and experienced again this friday when I ran a workshop.

Users - both kids and adults - on Windows, often seems to have trouble figuring out how to launch the Mu editor after installation. I'm not sure why, as I'm not a Windows user myself, but it seems that after the installer has run, users should be somehow instructed in "Go to the start menu and search for Mu". On Mac users know where to find applications, but apparantly that's not the case on Windows anymore.

In old days you always got a shortcut on the desktop, but I guess you don't do that on Windows no more? What is the normal practice? Is this just because users are not accustomed to even installing programs, or what can be the cause of this issue?

carlosperate, 2021-02-10:
Have you been able to confirm if the shortcut is definitely not installed in the start menu?

From the gitter chat it sounds like this problem already existed with the previous installer? It might  be an issue with Wix in that case, as both Briefcase and Pup uses it.

dybber, 2021-02-10:
This is two different issues: this one is about users not able to find shortcuts (discoverability), the other is that the shortcut is sometimes never created (a bug in the old briefcase installers - don't know if that problem persist in pup builds).

carlosperate, 2021-02-10:
Thanks for clarifying. If in future workshops you could confirm shortcut are not created we can open a new GH issue for that.

As far a discoverability, I think creating desktop shortcuts is no longer the "modern" way, although I there is probably nothing stopping us from still doing that, and some applications (specially old ones) still do that.

This is a personal view rather than fact, but since Windows 7 I have always perceived searching for an app in the start menu (either by scrolling through the entries or typing the name) as the default way to find applications.
Apps have always been there since Windows 95, but it is true that desktop shortcuts were a lot more common.
This menu did chang a bit with Windows 8 and their tablet-like metro start menu, but then came around to the same pattern when a hybrid of both menus was introduced in Windows 8.1.

Apart from that, for a quick app access, I guess you are meant to "pin" the app to the taskbar. Although that might be a more advanced feature (that being said my parents have been able to do that in their PC without really knowing how they did it).

Also, if an application has been recently installed, their shortcut should appear at the top of the start menu, so that should help discoverability. But that doesn't help much if users are not really opening the start menu to begin with.

dybber, 2021-02-10:
I haven't used Windows since 2005-2006 where I switched to Ubuntu, so I'm not sure what's the norms, I just noticed that it's hard for users to find the application after installation, and are even questioning whether it succeeded. So somekind of "Do you want to launch Mu now?" or instructions for launching it in the last screen of the installer might be useful.

carlosperate, 2021-02-10:
Some installers have the option to open the app on exit, that might be worth looking into as well. Did you see in any of the Wix options @tmontes ?

tmontes, 2021-02-10:
> Some installers have the option to open the app on exit, that might be worth looking into as well. Did you see in any of the Wix options @tmontes ?

...WiX can definitely implement that. Now how to get that somehow cleanly into the current state of `pup` without making a huge hack out of it, is another issue! :)

PS: Amazed by @dybber experience... I also ran a 4-session remote workshop with 10 year olds these past two weeks and once I told them how to launch Mu (via the Start Menu) they were set -- maybe they didn't know from the start, but they certainly "learned that" once and for all. :)

PPS: Logged https://github.com/mu-editor/pup/issues/145

dybber, 2021-02-11:
My workshop was with highschool teachers! Perhaps I just need to add it to part of my presentation, that you need to go to the Start Menu to find it, but I think others will have this discoverability issue as well.

ntoll, 2021-02-14:
Hmm... can we perhaps add a flag to the installer so a launcher icon/thingamabob is added to the user's desktop..? I've seen that as a pattern in Windows installers.

carlosperate, 2021-04-16:
Was the final conclusion that we should by default put a shortcut in the desktop, **and also** add an option (ticked by default) in the last step of the installer to open Mu?

carlosperate, 2022-05-17:
✅ The installer already has an option (ticked by default) to open Mu.

Should we also add a shortcut to the desktop?
From my point of view this is still useful and is easy for users to delete it.

Some GitHub issues related to not having a shortcut:
- https://github.com/mu-editor/mu/issues/1498
- https://github.com/mu-editor/mu/issues/1148