mu-editor/mu #1283: Big Sur issues

fhassancq, 2021-02-08:
Have tried installing the current build (hangs, doesn't even load) and the alpha on Big Sur. The latter doesn't flash the hex file to the BBC micro:bit.

DrNebin, 2021-02-08:
This big sur issues does not seem to be fixed. I am on OS Big Sur and downloaded and tried to run Mu today. The program hangs when it attempts to load and never actually starts. I will try attaching two screenshots if I can.... also note there seems to be no way for me to bypass the security denial (right clicking does not bypass it) and I have to go to settings and get to the "open anyway" button to have it make an attempt to load.... then it hangs and the program doesn't open.

[image]
[image]
[image]

fmorton, 2021-02-11:
I know this seems unlikely, but I started playing with the code to zero in on the problem. I got an error message complaining about the virtualenv, so I did:

pip install virtualenv

and now everything is working fine. This is with the recent 11.2.1 Big Sur version.

DrNebin, 2021-02-11:
Thanks for the tip. Unfortunately, this did not work for me.

DrNebin, 2021-02-12:
i tried to upgrade my python... then failed at that... then tried to downgrade it, then upgrade it again, then installed tried to reinstall mu with brew, which failed, then used the OS X installer a second time. Now mu-editor opens. I'm not sure where in my dabbling things clicked in place... but it is working now on Big Sur...

spinninghamster1127, 2021-02-19:
I finally achieved success by installing the alpha version of the next release. Of course, I may create a lot of other trouble for myself, but, for now, it's working again.

ntoll, 2021-02-19:
Hurrah...! We hope to get a new beta release out very soon.

Nekxi, 2021-03-01:
Anyone have a solution for syntax highlight color lacking when using alpha 1.1.0 alpha.2 on mac OS Big Sur 11.2.2? A couple of my classmates and me included cannot seem to get any syntax highlights at all.

Phipanjo, 2021-03-02:
Color text problem same like when upgrading to Catalina. Black text everywhere. Hack of #936 doesn't work anymore. Hope this will fixed soon; programming at night with the day theme is a nightmare...

carlosperate, 2021-03-03:
These issues have been fixed in the Mu source code, and will be available in the next beta version, which we are working towards being able to release soon.

carlosperate, 2021-03-04:
I'll close this issue assuming all the issues here can be resolved with the steps listed in https://github.com/mu-editor/mu/discussions/1349

If you are still having problems not resolved with those steps please do reopen this issue and we can continue debugging it.

closed 2021-03-04 by carlosperate

fhassancq, 2021-03-05:
How can this be closed when the issue hasn't been resolved? When do you expect to release the next beta version?

carlosperate, 2021-03-16:
This should be fixed in the latest release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 🎉 

It'd be great if you could give that a spin, and if you are still experiencing problems we can reopen this issue or open a new one.

Thanks again for all your patience!

fhassancq, 2021-03-17:
I installed the latest version and it still isn't working. Installed on a machine that did not have the previous versions. The software does not flash to the bbc micro:bit.

carlosperate, 2021-03-17:
Thanks for getting back to us @fhassancq, in that case you would mind opening a new ticket with your Mu log?
https://codewith.mu/en/howto/1.1/read_logs

I suggest a new GH issue, instead of continue here, as the majority of the discussion in this issue is about permissions that are meant to be working on beta.2, and that we know were not working in older releases.
If you are having problems flashing the micro:bit with 1.1.0-beta.2 then it might be a different problem, or if it's related to permissions, it would require a different type of fix. So a new discussion can help us keep things focused.