mu-editor/mu #1284: make check test/test_app.py never finishes

fmorton, 2021-02-10:
I was working to catch a pull request up-to-date and discovered when running "make check" that teset/test_app.py never finished (after letting it run for more than an hour). This is on a mac with MacOS 11.1 (Big Sur). So, I cloned the current version and discovered that the same issue is present on the current main branch.

Does anyone know about this? Here is the output from "make check" (also showing the clone):

bash-3.2$ git clone https://github.com/mu-editor/mu.git
Cloning into 'mu'...
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 11211 (delta 0), reused 2 (delta 0), pack-reused 11208
Receiving objects: 100% (11211/11211), 19.28 MiB | 30.42 MiB/s, done.
Resolving deltas: 100% (7858/7858), done.
bash-3.2$ cd mu
bash-3.2$ make check
rm -rf build
rm -rf dist
rm -rf .coverage
rm -rf .eggs
rm -rf docs/_build
rm -rf .pytest_cache
rm -rf lib
rm -rf *.mp4
rm -rf .git/avatar/*
find . \( -name '*.py[co]' -o -name dropin.cache \) -delete
find . \( -name '*.bak' -o -name dropin.cache \) -delete
find . \( -name '*.tgz' -o -name dropin.cache \) -delete
find . | grep -E "(__pycache__)" | xargs rm -rf
python make.py black

black
black, version 20.8b1
All done! ✨ 🍰 ✨
1 file would be left unchanged.
All done! ✨ 🍰 ✨
1 file would be left unchanged.
All done! ✨ 🍰 ✨
49 files would be left unchanged.
All done! ✨ 🍰 ✨
2 files would be left unchanged.
All done! ✨ 🍰 ✨
36 files would be left unchanged.
All done! ✨ 🍰 ✨
6 files would be left unchanged.
flake8
export LANG=en_GB.utf8
pytest --random-order --cov-config .coveragerc --cov-report term-missing --cov=mu tests/
================================================= test session starts ==================================================
platform darwin -- Python 3.7.6, pytest-6.0.1, py-1.9.0, pluggy-0.13.1
Using --random-order-bucket=module
Using --random-order-seed=846489

rootdir: /Users/fmorton/tmp/tmp/mu, configfile: pytest.ini
plugins: random-order-1.0.4, cov-2.10.1
collected 1008 items                                                                                                   

tests/virtual_environment/test_process.py ......                                                                 [  0%]
tests/debugger/test_runner.py ...................................................                                [  5%]
tests/modes/test_esp.py ........................                                                                 [  8%]
tests/virtual_environment/test_pip.py ..................                                                         [  9%]
tests/interface/test_editor.py ................................................                                  [ 14%]
tests/interface/test_main.py ................................................................................... [ 22%]
................                                                                                                 [ 24%]
tests/virtual_environment/test_virtual_environment.py ....F.............                                         [ 26%]
tests/modes/test_pico.py .                                                                                       [ 26%]
tests/interface/test_panes.py .................................................................................. [ 34%]
...................s.................................................................                            [ 42%]
tests/debugger/test_client.py ...............................................                                    [ 47%]
tests/interface/test_themes.py ....                                                                              [ 47%]
tests/modes/test_base.py ...........................s..................................                          [ 54%]
tests/test_logic.py .......F................................................s................................... [ 63%]
.......................................................s.......................................                  [ 72%]
tests/interface/test_dialogs.py s.....ss..s.s...ss.sss.....s..s.s                                                [ 75%]
tests/modes/test_microbit.py ....................................................                                [ 81%]
tests/modes/test_web.py ....................                                                                     [ 83%]
tests/debugger/test_utils.py ......                                                                              [ 83%]
tests/modes/test_pygamezero.py ..................                                                                [ 85%]
tests/test_resources.py ......                                                                                   [ 86%]
tests/modes/test_circuitpython.py ...........                                                                    [ 87%]
tests/modes/test_python3.py ...........................                                                          [ 89%]
tests/modes/test_pyboard.py .........                                                                            [ 90%]
tests/modes/test_lego.py .                                                                                       [ 90%]
tests/interface/test_widgets.py .........                                                                        [ 91%]
tests/test_app.py .

Never finished

carlosperate, 2021-02-10:
thanks for the report @fmorton, I think this is the same problem we are seeing sporadically in the CI: https://github.com/mu-editor/mu/issues/1266
Is the issue constant in your environment, or do the tests complete sometimes?

fmorton, 2021-02-11:
Thanks @carlosperate, I also think it is the same issue. I posted on the other thread that I simply did:

pip install virtualenv

and both of these problems went away.

My PR now passes 'make check' without any issues, so I will close this one since it is the same issue as the other one.

fmorton, 2021-02-11:
I reopened this. When committing the changes to catch up with 1.1.0.beta.1, the travis-ci continuous integration check failed with tests/test_app.py never finishing, timing out after 10 minutes:

https://travis-ci.org/github/mu-editor/mu/jobs/758588188

ntoll, 2021-02-14:
I think this may be related to the `virtualenv` or `venv` (choose one) problem discussed elsewhere (#1291?)

carlosperate, 2021-03-17:
@fmorton are you still having this problem from the latest version of this repo?

fmorton, 2021-03-20:
@carlosperate, looks like there is still some sort of issue. When updating PR #485 23 of the tests work just fine and all finish very quickly (ie, less than a minute). But, right now there are two tests that seem to be stuck. They have been running for over 4 hours while other similar tests finish in less than one minute. Since they are still running, I do not see a way to look at the logs to see how far it got. Hopefully, it will timeout eventually and we will be able to learn something from the log.

The two tests that are not finishing are:

Run tests / Test Py 3.7 - ubuntu-16.04 (pull_request)
Run tests / Test Py 3.7 - ubuntu-latest (pull_request)

Thanks for all of your diligent help.

fmorton, 2021-03-20:
The test did finally timeout. Here is the resulting log:

Run xvfb-run make check
  xvfb-run make check
  shell: /bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.7.10/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.7.10/x64/lib
rm -rf build
rm -rf dist
rm -rf .coverage
rm -rf .eggs
rm -rf *.egg-info
rm -rf docs/_build
rm -rf .pytest_cache
rm -rf lib
rm -rf *.mp4
rm -rf .git/avatar/*
find . \( -name '*.py[co]' -o -name dropin.cache \) -delete
find . \( -name '*.bak' -o -name dropin.cache \) -delete
find . \( -name '*.tgz' -o -name dropin.cache \) -delete
find . | grep -E "(__pycache__)" | xargs rm -rf
python make.py black
black, version 20.8b1
All done! ✨ 🍰 ✨
1 file would be left unchanged.
All done! ✨ 🍰 ✨
1 file would be left unchanged.
All done! ✨ 🍰 ✨
49 files would be left unchanged.
All done! ✨ 🍰 ✨
2 files would be left unchanged.
All done! ✨ 🍰 ✨
36 files would be left unchanged.
All done! ✨ 🍰 ✨
6 files would be left unchanged.

black
flake8
export LANG=en_GB.utf8
pytest --random-order --cov-config .coveragerc --cov-report term-missing --cov=mu tests/
============================= test session starts ==============================
platform linux -- Python 3.7.10, pytest-6.2.2, py-1.10.0, pluggy-0.13.1
Using --random-order-bucket=module
Using --random-order-seed=54696

rootdir: /home/runner/work/mu/mu, configfile: pytest.ini
plugins: random-order-1.0.4, cov-2.11.1
collected 1023 items

tests/debugger/test_utils.py ......                                      [  0%]
tests/modes/test_esp.py ........................                         [  2%]
tests/test_settings.py ..............s............s.......               [  6%]
tests/virtual_environment/test_process.py ......                         [  6%]
tests/modes/test_python3.py ...........................                  [  9%]
tests/interface/test_dialogs.py .....sss.ss..s...ssss....s.ss......      [ 13%]
tests/modes/test_base.py ..........................s.................... [ 17%]
...............                                                          [ 19%]
tests/virtual_environment/test_virtual_environment.py .................. [ 20%]
..s.....                                                                 [ 21%]
tests/modes/test_pygamezero.py ..................                        [ 23%]
tests/modes/test_circuitpython.py ...........                            [ 24%]
tests/modes/test_lego.py .                                               [ 24%]
tests/modes/test_debug.py ........................................       [ 28%]
tests/modes/test_web.py ....................                             [ 30%]
tests/modes/test_pico.py .                                               [ 30%]
tests/modes/test_microbit.py ........................................... [ 34%]
.........                                                                [ 35%]
tests/interface/test_widgets.py .........                                [ 36%]
tests/modes/test_pyboard.py .........                                    [ 37%]
Error: The operation was canceled.

Post job cleanup log:
Post job cleanup.
/usr/bin/git version
git version 2.31.0
/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
http.https://github.com/.extraheader
/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :

Complete job cleanup:
Cleaning up orphan processes
Terminate orphan process: pid (4578) (xvfb-run)
Terminate orphan process: pid (4589) (Xvfb)
Terminate orphan process: pid (4594) (make)
Terminate orphan process: pid (4675) (pytest)

carlosperate, 2021-03-22:
Thanks for checking @fmorton. @tjguk has been looking into this and the latest updates should have improve this considerably, could you try again with the latest commits?

fmorton, 2021-03-22:
Those errors look to be fixed, but there is a new one with the travis CI. All tests passed except this one.

[image]

Here is the giant raw log. Glad to do any other tests you need.

travis_fold : start : worker_info
 [0K [33;1mWorker information [0m
hostname: e97c327d-6e96-44cf-a316-5aad34c997ab@927672.lxd-arm64-03-org
version: ? ?
instance: travis-job-mu-editor-mu-763944043 03e95732d355bb6061689d41d23ad21c759bbde32a2ce0d53b0ee90b1424e2c5 (via amqp)
startup: 3.984970699s
travis_fold : end : worker_info
 [0Ktravis_time : start : 0f7f6282
 [0Ktravis_time : end : 0f7f6282: start=1616422839130792829,finish=1616422839352313782,duration=221520953,event=no_world_writable_dirs
 [0Ktravis_time : start : 06f9f690
 [0Ktravis_time : end : 06f9f690: start=1616422839358190699,finish=1616422839372215910,duration=14025211,event=agent
 [0Ktravis_time : start : 00a6493a
 [0Ktravis_time : end : 00a6493a: start=1616422839377513482,finish=1616422839381584742,duration=4071260,event=check_unsupported
 [0Ktravis_time : start : 255d6b00
 [0Ktravis_fold : start : system_info
 [0K [33;1mBuild system information [0m
Build language: python
Build dist: xenial
Build id: 763944039
Job id: 763944043
Runtime kernel version: 5.8.0-45-generic
travis-build version: 091d532a
 [34m [1mBuild image provisioning date and time [0m
Fri Jul 10 13:20:51 UTC 2020
 [34m [1mOperating System Details [0m
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
 [34m [1mLinux Version [0m
5.0.0-23-generic
 [34m [1mGit version [0m
git version 2.7.4
 [34m [1mbash version [0m
GNU bash, version 4.3.48(1)-release (aarch64-unknown-linux-gnu)
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
 [34m [1mGCC version [0m
gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

 [34m [1mDocker version [0m
Client:
 Version:           18.09.7
 API version:       1.39
 Go version:        go1.10.4
 Git commit:        2d0083d
 Built:             Fri Aug 16 14:20:40 2019
 OS/Arch:           linux/arm64
 Experimental:      false
 [34m [1mLLVM version [0m
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: aarch64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
 [34m [1mPre-installed Ruby versions [0m
ruby-2.6.3
ruby-2.6.5
ruby-2.7.0
[... 2221 lines cut ...]
mu/modes/microbit.py           313     17    95%   78-90, 335-344, 373-377, 390-397
mu/modes/pico.py                14      0   100%
mu/modes/pyboard.py             58      0   100%
mu/modes/pygamezero.py          65      0   100%
mu/modes/python3.py            180     10    94%   42-58
mu/modes/web.py                107      0   100%
mu/resources/__init__.py        17      0   100%
mu/settings.py                 122      0   100%
mu/virtual_environment.py      283     35    88%   160, 163, 166, 212-217, 339, 395, 398, 414, 422, 433-435, 458, 467-471, 489, 502, 535, 548-560, 567
----------------------------------------------------------
TOTAL                         5722    260    95%

=========================== short test summary info ============================
FAILED tests/virtual_environment/test_virtual_environment.py::test_create_virtual_environment_on_disk
================== 1 failed, 979 passed, 43 skipped in 34.00s ==================
Makefile:56: recipe for target 'coverage' failed
make: *** [coverage] Error 1
travis_time : end : 27af3c40: start=1616423157748239574,finish=1616423195584479740,duration=37836240166,event=script
 [0K [31;1mThe command "if [ "$TRAVIS_PYTHON_VERSION" = "3.5" ]; then make coverage; fi" exited with 2. [0m
travis_time : start : 05c1cfe6
 [0K$ if [ "$TRAVIS_PYTHON_VERSION" != "3.5" ]; then make check; fi
travis_time : end : 05c1cfe6: start=1616423195590905102,finish=1616423195595346458,duration=4441356,event=script
 [0K [32;1mThe command "if [ "$TRAVIS_PYTHON_VERSION" != "3.5" ]; then make check; fi" exited with 0. [0m

Done. Your build exited with 1.

tjguk, 2021-03-22:
Thanks @fmorton  -- yes, @carlosperate pointed that one out to me earlier. I have no idea why it fails in Travis but succeeds in all the GH builds. I'll get to it later to see if I can work out what's happening

tjguk, 2021-03-22:
I've reverted the commit which -- for reasons I can't fathom -- was causing Travis to fail. It was a tentative solution to a problem I can't reproduce, so I can come at it some other way later on.

fmorton, 2021-03-23:
The Travis CI test now succeeds, but now Test Py 3.7 - macos-10.15 fails with a timeout in the install mu dependencies section:

Run pip install .[dev]
Processing /Users/runner/work/mu/mu
Collecting qtconsole==4.7.4
  Downloading qtconsole-4.7.4-py2.py3-none-any.whl (118 kB)
Collecting pyserial==3.4
  Downloading pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting flake8>=3.8.3
  Downloading flake8-3.9.0-py2.py3-none-any.whl (73 kB)
Collecting appdirs>=1.4.3
  Downloading appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting semver>=2.8.0
  Downloading semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting virtualenv
  Downloading virtualenv-20.4.3-py2.py3-none-any.whl (7.2 MB)
Collecting wheel
  Using cached wheel-0.36.2-py2.py3-none-any.whl (35 kB)
Collecting PyQt5==5.13.2
  Downloading PyQt5-5.13.2-5.13.2-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl (39.8 MB)
Collecting QScintilla==2.11.3
  Downloading QScintilla-2.11.3-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl (2.3 MB)
Collecting PyQtChart==5.13.1
  Downloading PyQtChart-5.13.1-5.13.2-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl (868 kB)
Collecting black>=19.10b0
  Downloading black-20.8b1.tar.gz (1.1 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Collecting pytest>=4.6
  Downloading pytest-6.2.2-py3-none-any.whl (280 kB)
Collecting pytest-cov
  Downloading pytest_cov-2.11.1-py2.py3-none-any.whl (20 kB)
Collecting pytest-random-order>=1.0.0
  Downloading pytest_random_order-1.0.4-py3-none-any.whl (10 kB)
Collecting pytest-faulthandler
  Downloading pytest_faulthandler-2.0.1-py2.py3-none-any.whl (2.7 kB)
Collecting pytest-timeout
  Downloading pytest_timeout-1.4.2-py2.py3-none-any.whl (10 kB)
Collecting coverage
  Downloading coverage-5.5-cp37-cp37m-macosx_10_9_x86_64.whl (207 kB)
Collecting nudatus
  Downloading nudatus-0.0.5-py3-none-any.whl (5.3 kB)
Collecting sphinx
  Downloading Sphinx-3.5.3-py3-none-any.whl (2.8 MB)
Collecting twine
  Downloading twine-3.4.1-py3-none-any.whl (34 kB)
Collecting PyQt5_sip<13,>=4.19.19
  Downloading PyQt5_sip-12.8.1-cp37-cp37m-macosx_10_9_x86_64.whl (63 kB)
Collecting ipykernel>=4.1
  Downloading ipykernel-5.5.0-py3-none-any.whl (120 kB)
Collecting traitlets
  Downloading traitlets-5.0.5-py3-none-any.whl (100 kB)
Collecting ipython-genutils
  Downloading ipython_genutils-0.2.0-py2.py3-none-any.whl (26 kB)
Collecting qtpy
  Downloading QtPy-1.9.0-py2.py3-none-any.whl (54 kB)
Collecting pyzmq>=17.1
  Downloading pyzmq-22.0.3-cp37-cp37m-macosx_10_9_x86_64.whl (1.0 MB)
Collecting pygments
  Downloading Pygments-2.8.1-py3-none-any.whl (983 kB)
Collecting jupyter-core
  Downloading jupyter_core-4.7.1-py3-none-any.whl (82 kB)
Collecting jupyter-client>=4.1
  Downloading jupyter_client-6.1.12-py3-none-any.whl (112 kB)
Collecting pathspec<1,>=0.6
  Using cached pathspec-0.8.1-py2.py3-none-any.whl (28 kB)
[... 107 lines cut ...]
Collecting rfc3986>=1.4.0
  Downloading rfc3986-1.4.0-py2.py3-none-any.whl (31 kB)
Collecting tqdm>=4.14
  Downloading tqdm-4.59.0-py2.py3-none-any.whl (74 kB)
Collecting pkginfo>=1.4.2
  Downloading pkginfo-1.7.0-py2.py3-none-any.whl (25 kB)
Collecting readme-renderer>=21.0
  Downloading readme_renderer-29.0-py2.py3-none-any.whl (15 kB)
Collecting bleach>=2.1.0
  Downloading bleach-3.3.0-py2.py3-none-any.whl (283 kB)
Collecting webencodings
  Downloading webencodings-0.5.1-py2.py3-none-any.whl (11 kB)
Collecting distlib<1,>=0.3.1
  Downloading distlib-0.3.1-py2.py3-none-any.whl (335 kB)
Collecting filelock<4,>=3.0.0
  Downloading filelock-3.0.12-py3-none-any.whl (7.6 kB)
Using legacy 'setup.py install' for mu-editor, since package 'wheel' is not installed.
Building wheels for collected packages: black
  Building wheel for black (PEP 517): started
  Building wheel for black (PEP 517): finished with status 'done'
  Created wheel for black: filename=black-20.8b1-py3-none-any.whl size=124184 sha256=86f358054afd539c7c004eeed5a73cb40435df6641481ea8a2b9a46b9b24aba0
  Stored in directory: /Users/runner/Library/Caches/pip/wheels/c5/85/79/f3af8daaf8037c0bf14beb3b7a1511a39b6e6902ca2aaf494e
Successfully built black
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, ptyprocess, parso, zipp, typing-extensions, tornado, pyzmq, python-dateutil, pyparsing, pygments, prompt-toolkit, pickleshare, pexpect, jupyter-core, jedi, decorator, backcall, appnope, webencodings, urllib3, PyQt5-sip, packaging, jupyter-client, ipython, importlib-metadata, idna, chardet, certifi, typed-ast, toml, requests, regex, qtpy, pytz, PyQt5, pyflakes, pycodestyle, py, pluggy, pathspec, mypy-extensions, mccabe, MarkupSafe, ipykernel, iniconfig, filelock, docutils, distlib, click, bleach, attrs, appdirs, wheel, virtualenv, tqdm, sphinxcontrib-serializinghtml, sphinxcontrib-qthelp, sphinxcontrib-jsmath, sphinxcontrib-htmlhelp, sphinxcontrib-devhelp, sphinxcontrib-applehelp, snowballstemmer, semver, rfc3986, requests-toolbelt, readme-renderer, qtconsole, QScintilla, pytest, pyserial, PyQtChart, pkginfo, keyring, Jinja2, imagesize, flake8, coverage, colorama, black, babel, alabaster, twine, sphinx, pytest-timeout, pytest-random-order, pytest-faulthandler, pytest-cov, nudatus, mu-editor
Error: The action has timed out.

carlosperate, 2021-03-23:
Thanks @fmorton, I've created PR https://github.com/mu-editor/mu/pull/1398 to increase the timeout.

fmorton, 2021-03-23:
Thanks @carlosperate. I merged the #1398 and all the tests were successful!

[image]

However, there is a strange thing you may understand. I received an email at the time of those tests stating that one of them failed even though it appears all the tests were successful:

[image]

Here is the log (and link to the run tests) for the test that says it was successful in one place and that it failed in another. Does this make sense somehow?

https://github.com/fmorton/mu/actions/runs/679649003

Run xvfb-run make check
rm -rf build
rm -rf dist
rm -rf .coverage
rm -rf .eggs
rm -rf *.egg-info
rm -rf docs/_build
rm -rf .pytest_cache
rm -rf lib
rm -rf *.mp4
rm -rf .git/avatar/*
find . \( -name '*.py[co]' -o -name dropin.cache \) -delete
find . \( -name '*.bak' -o -name dropin.cache \) -delete
find . \( -name '*.tgz' -o -name dropin.cache \) -delete
find . | grep -E "(__pycache__)" | xargs rm -rf
python make.py black

black
Black checks are not available in Python 3.5.
flake8
export LANG=en_GB.utf8
pytest -v --random-order --cov-config .coveragerc --cov-report term-missing --cov=mu tests/
============================= test session starts ==============================
platform linux -- Python 3.5.10, pytest-6.1.2, py-1.10.0, pluggy-0.13.1 -- /opt/hostedtoolcache/Python/3.5.10/x64/bin/python
cachedir: .pytest_cache
Using --random-order-bucket=module
Using --random-order-seed=25763

rootdir: /home/runner/work/mu/mu, configfile: pytest.ini
plugins: random-order-1.0.4, timeout-1.4.2, cov-2.11.1
collecting ... collected 1023 items

tests/modes/test_pyboard.py::test_workspace_dir_unknown_os PASSED        [  0%]
tests/modes/test_pyboard.py::test_api PASSED                             [  0%]
tests/modes/test_pyboard.py::test_pyboard_mode_no_charts PASSED          [  0%]
tests/modes/test_pyboard.py::test_workspace_dir_posix_exists PASSED      [  0%]
tests/modes/test_pyboard.py::test_workspace_dir_nt_exists PASSED         [  0%]
tests/modes/test_pyboard.py::test_workspace_dir_posix_missing PASSED     [  0%]
tests/modes/test_pyboard.py::test_workspace_dir_nt_missing PASSED        [  0%]
tests/modes/test_pyboard.py::test_workspace_dir_posix_no_mount_command PASSED [  0%]
tests/modes/test_pyboard.py::test_pyboard_mode PASSED                    [  0%]
tests/test_resources.py::test_load_icon PASSED                           [  0%]
tests/test_resources.py::test_load_pixmap PASSED                         [  1%]
tests/test_resources.py::test_path PASSED                                [  1%]
tests/test_resources.py::test_load_font_data PASSED                      [  1%]
tests/test_resources.py::test_stylesheet PASSED                          [  1%]
tests/test_resources.py::test_load_movie PASSED                          [  1%]
tests/modes/test_esp.py::test_add_fs_no_device PASSED                    [  1%]
tests/modes/test_esp.py::test_toggle_repl_fail PASSED                    [  1%]
tests/modes/test_esp.py::test_toggle_repl_with_fs PASSED                 [  1%]
tests/modes/test_esp.py::test_ESPMode_actions_no_charts PASSED           [  1%]
tests/modes/test_esp.py::test_add_fs PASSED                              [  1%]
tests/modes/test_esp.py::test_run_no_device PASSED                       [  2%]
tests/modes/test_esp.py::test_api PASSED                                 [  2%]
tests/modes/test_esp.py::test_toggle_repl_off PASSED                     [  2%]
tests/modes/test_esp.py::test_toggle_files_with_repl PASSED              [  2%]
tests/modes/test_esp.py::test_run_no_editor PASSED                       [  2%]
tests/modes/test_esp.py::test_run PASSED                                 [  2%]
[... 149 lines cut ...]
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/runner.py", line 256 in call_runtest_hook
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/runner.py", line 216 in call_and_report
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/runner.py", line 127 in runtestprotocol
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/runner.py", line 110 in pytest_runtest_protocol
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/main.py", line 338 in pytest_runtestloop
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/main.py", line 313 in _main
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/main.py", line 257 in wrap_session
[... cut ...]
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/config/__init__.py", line 165 in main
  File "/opt/hostedtoolcache/Python/3.5.10/x64/lib/python3.5/site-packages/_pytest/config/__init__.py", line 187 in console_main
  File "/opt/hostedtoolcache/Python/3.5.10/x64/bin/pytest", line 8 in <module>
make: *** [Makefile:57: coverage] Segmentation fault (core dumped)
Error: Process completed with exit code 2.

carlosperate, 2021-03-31:
Great, thanks for all the testing @fmorton !

closed 2021-03-31 by carlosperate