mu-editor/mu #1285: esp32 wroom32d

rdaguerre, 2021-02-10:
Hi, i use ubuntu 20.10 64bits and Mu 1.0.3
i cant use REPL not found my esp32 wroom32d

2021-02-10 13:46:13,736 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2021-02-10 13:46:17,692 - mu.modes.base:231(find_device) WARNING: Could not find device.
2021-02-10 13:46:17,693 - mu.modes.base:232(find_device) DEBUG: Available ports:
2021-02-10 13:46:17,693 - mu.modes.base:233(find_device) DEBUG: ['PID:0xea60 VID:0x10c4 PORT:ttyUSB0', 'PID:0x2303 VID:0x067b PORT:ttyUSB1', 'PID:0x0000 VID:0x0000 PORT:ttyS0']
2021-02-10 13:46:17,693 - mu.interface.main:723(show_message) DEBUG: No se pudo encontrar un dispositivo conectado.
2021-02-10 13:46:17,693 - mu.interface.main:724(show_message) DEBUG: Asegúrate de que el dispositivo esta conectado al ordenador.

Debe tener una versión de MicroPython (o CircuitPython) programada antes de que el REPL funcione.

Finalmente, presiona el botón de reinicio del dispositivo y espera unos segundos antes intentarlo otra vez.
2021-02-10 13:46:36,519 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2021-02-10 13:46:51,062 - mu.logic:1075(show_admin) INFO: Showing logs from /home/ronnie/.cache/mu/log/mu.log

excuse me for my english
thanks

ntoll, 2021-02-14:
Hi @rdaguerre,

The upcoming 1.1 release of Mu has an ESP mode which should with supported boards.

dybber, 2021-02-15:
Hi @rdaguerre, it seems that you had two different USB-devices connected when you captured the log-file:

- PID:0xea60 VID:0x10c4 on port `ttyUSB0`
- PID:0x2303 VID:0x067b on port `ttyUSB1`

The first device should be recognized by Mu already, the second is unknown to me. Can you confirm what VID/PID your ESP32­-WROOM­-32D is using?

ESP32-WROOM-32D is just the microcontroller, it doesn't come with USB-Serial, so some other USB-Serial interface is used, which I have to be sure of before adding it to Mu. If it's the former, it's already there, if it's the latter I'll add it.

rdaguerre, 2021-02-17:
Hi, excuse me my english.
the first is a Silicon Labs CP210x UART Bridge (esp32 wroom32d) connected in usb so
the second is a Prolific Technology, Inc. PL2303 Serial Port (not used in the project. connected in a pc without use now)

dybber, 2021-02-17:
Okay, great. That means it should already be supported. In a few days we'll probably have a new beta version of Mu out where it's included.

closed 2021-02-19 by dybber