mu-editor/mu #1290: Pygame libportmidi.dylib cannot be singned in macOS

carlosperate, 2021-02-10:
Trying to sign the macOS app encounters the following issue with `libportmidi.dylib` from the `pygame-2.0.1-cp37-cp37m-macosx_10_9_intel.whl` file:

```
The binary uses an SDK older than the 10.9 SDK
```

Hopefully newer versions of PyGame will have a different version of the lib built with a newer SDK: https://github.com/pygame/pygame/issues/2472 

In the meantime we can try making our own hacked wheel to ship the next release.

## Hacking a new wheel

To workaround this we can inject a different version of the lib into the wheel.

I've done this steps to create the following wheel:
- Downloaded `pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl` (as the next release will likely use Python 3.8 instead of 3.7)
- Got a version of the lib from brew
    - `$ brew install portmidi`
    - https://formulae.brew.sh/formula/portmidi#default
    - https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/portmidi.rb
    - Will end up in `/usr/local/Cellar/portmidi/217_2/lib`
- Need the SHA to update the wheel `RECORD` file (adapted from [this Stack Overflow answer](https://stackoverflow.com/questions/55904261/how-the-hash-in-record-file-of-a-wheel-package-is-constructed)):
    ```
    $ python3 -c "\
    import hashlib, base64, os;\
    f='libportmidi.dylib'
    print('sha256={},{}'.format(\
     base64.urlsafe_b64encode(\
     hashlib.sha256(open(f, 'rb').read()).digest()).decode('latin1').rstrip('='),\
     os.stat(f).st_size))"
     ```
- Replaced the provided SHA from the wheel `RECORD` file and replaced `libportmidi.dylib`
- Output wheel: [pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl.zip](https://github.com/mu-editor/mu/files/5961918/pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl.zip)

## Using this wheel

Assuming this is a temporary workaround I think we have these options:

- Add the wheel to the repo
    - We would need to update `mu.wheels` to use this wheel only in macOS with Python 3.8, and use a PyPI wheel everywhere else
- Upload the wheel somewhere (a GitHub release asset is a good option) and update `mu.wheels` to pip install our wheel from a URL
    - We would need to update `mu.wheels` to use this wheel only in macOS with Python 3.8, and use a PyPI wheel everywhere else
- Any other options?

@tjguk do you have a preference?

ntoll, 2021-02-14:
I've filed this in my internal "why on earth did I ever choose a career in programming?" file. :slightly_smiling_face: :facepalm: 

I can confirm that the package notarizes with the fix described by @carlosperate. I personally think a pragmatic approach would be easiest for us, and if we need to host our own PyGame wheel and keep tabs on how Rene (@illume ) is doing over in upstream then we can ship the next version of Mu with the notarized installer created via @tmontes wonderful `pup` work.

@illume just a FYI - should this bug go away, `pup` provides a *really simple* way to package and distribute PyGame projects for OSX.

illume, 2021-02-14:
¡Hola desde Buenos Aires!

Sorry this is blocking you/Mu, and that this made your `journal/programming-life-regrets.txt` file.

I expect we won't be able to do another pygame release with a fix for some weeks/months. (The fix should be relatively easy as far as hacking on exotic build systems go... just applying the same patch homerew portmidi does to the pygame build of portmidi. I tried to provide pointers to all the relative bits/docs in the pygame/pygame#2472 issue, but won't be able to do it myself this weekend... maybe next one)

ntoll, 2021-02-14:
¡Hola!

You're a :star: - we'll work our way around this for the time being..!

carlosperate, 2021-02-17:
Hola René!
Thanks for looking into the library patch and the update! It'll be great to move to the official pygame wheels again with the next release.

In the meantime, to unblock the next Mu release I've created this PR https://github.com/mu-editor/mu/pull/1302, which on Python 3.8 for macOS (the version used for building the Mu installers) it uses a custom `pygame` and `pgzero` wheel hosted in GH releases from forks in the mu-editor GH org:
- https://github.com/mu-editor/pygame/releases/tag/2.0.1
- https://github.com/mu-editor/pgzero/releases/tag/mu-wheel

carlosperate, 2021-03-02:
We can consider this done for the beta.2 release, I'll move this issue to the next milestone to track the pygame/pygamezero upstream releases.

TongDaDa, 2021-04-15:
@carlosperate Does this wheel is compatible with all platform of macOs including the Bigusr that based on Intel and M1, and previous versions of macOS.

carlosperate, 2021-04-15:
No M1 support at the moment in pygame, as it requires rebuilding everything for Arm, but it should work with all macOS versions in x64.

In M1 can you install it with x86 emulation? I assume that should work.

TongDaDa, 2021-04-15:
> No M1 support at the moment in pygame, as it requires rebuilding everything for Arm, but it should work with all macOS versions in x64.
> 
> In M1 can you install it with x86 emulation? I assume that should work.

I was used Electron framework, but also caused by the `libportmidi.dylib`, the process of notarization failed with the older SDK warning. I wish to find a way solving the problem. Can you give me some suggests?

tmontes, 2021-11-03:
Hey @carlosperate,

https://github.com/pygame/pygame/issues/2472 is fixed with the release of pygame 2.0.2. Subsequently, 2.0.3 has been released.

I installed the 2.0.3 wheel on my system and it looks like `libportmidi.dylib` is in good shape. I just currently don't have a developer certificate to try out the signing+notarization process.

tmontes, 2021-11-03:
PS: Also, PyPI `pgzero` 1.2.1 brings in `pygame` 2.0.3 so maybe the `if darwin` code block can be removed from `mu/wheels/__init__.py`.

tmontes, 2021-11-04:
Update,

I now have a valid developer certificate and confirmed that the updated `pygame` and `pgzero` from PyPI properly pass the Apple signing+notarization and that the resulting `pup`-packaged Application Bundle runs Pygame Zero mode scripts successfuly.

PR on the way.

PS: For some misterious reason, on my dev system, with Mu installed in editable mode, Pygame Zero fails to run scripts: both current `master` and the branch I'm about to PR. (?!)

carlosperate, 2021-11-04:
That's great, thanks @tmontes and the pygame team!
We can close this as resolved 🎉

closed 2021-11-04 by carlosperate