mu-editor/mu #1291: Missing python interpreter in venv causes

carlosperate, 2021-02-11:
I'm not 100% clear how my venv had a missing python executable.
I haven't been able to replicate yet but was able to retrieve the error from the log (tried to run twice before manually removing the venv):

```
-----------------

Starting Mu 1.1.0.beta.1
2021-02-11 08:24:50,516 - root:173(run) INFO: uname_result(system='Darwin', node='Carlos-MBP-8.local', release='18.7.0', version='Darwin Kernel Version 18.7.0: Tue Jan 12 22:04:47 PST 2021; root:xnu-4903.278.56~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-02-11 08:24:50,517 - root:174(run) INFO: Python path: ['/', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python37.zip', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/lib-dynload', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/setuptools-49.6.0-py3.7.egg', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/pip-20.2.2-py3.7.egg', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/IPython/extensions']
2021-02-11 08:24:50,517 - root:175(run) INFO: Language code: en_GB
2021-02-11 08:24:50,776 - mu.virtual_environment:371(ensure) DEBUG: Found existing virtual environment at /Users/microbit-carlos/Library/Application Support/mu/mu_venv
2021-02-11 08:24:50,776 - mu.virtual_environment:383(ensure_interpreter) ERROR: Interpreter not found where expected at /Users/microbit-carlos/Library/Application Support/mu/mu_venv/bin/python
2021-02-11 08:24:50,776 - root:106(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/mu/app.py", line 98, in run
    venv.ensure()
  File "/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/mu/virtual_environment.py", line 373, in ensure
    self.ensure_interpreter()
  File "/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/mu/virtual_environment.py", line 384, in ensure_interpreter
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: Interpreter not found where expected at /Users/microbit-carlos/Library/Application Support/mu/mu_venv/bin/python
2021-02-11 08:24:50,777 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/session.json
2021-02-11 08:24:50,778 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/venv.json
2021-02-11 08:25:06,987 - root:172(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.1
2021-02-11 08:25:06,987 - root:173(run) INFO: uname_result(system='Darwin', node='Carlos-MBP-8.local', release='18.7.0', version='Darwin Kernel Version 18.7.0: Tue Jan 12 22:04:47 PST 2021; root:xnu-4903.278.56~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-02-11 08:25:06,987 - root:174(run) INFO: Python path: ['/', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python37.zip', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/lib-dynload', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/setuptools-49.6.0-py3.7.egg', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/pip-20.2.2-py3.7.egg', '/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/IPython/extensions']
2021-02-11 08:25:06,987 - root:175(run) INFO: Language code: en_GB
2021-02-11 08:25:07,184 - mu.virtual_environment:371(ensure) DEBUG: Found existing virtual environment at /Users/microbit-carlos/Library/Application Support/mu/mu_venv
2021-02-11 08:25:07,184 - mu.virtual_environment:383(ensure_interpreter) ERROR: Interpreter not found where expected at /Users/microbit-carlos/Library/Application Support/mu/mu_venv/bin/python
2021-02-11 08:25:07,184 - root:106(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/mu/app.py", line 98, in run
    venv.ensure()
  File "/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/mu/virtual_environment.py", line 373, in ensure
    self.ensure_interpreter()
  File "/Users/microbit-carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.7/site-packages/mu/virtual_environment.py", line 384, in ensure_interpreter
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: Interpreter not found where expected at /Users/microbit-carlos/Library/Application Support/mu/mu_venv/bin/python
2021-02-11 08:25:07,185 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/session.json
2021-02-11 08:25:07,186 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/venv.json
2021-02-11 08:25:15,345 - root:172(run) INFO: 

-----------------
```

I believe @dybber has encountered the same issue. Did you check that the python file was indeed missing?

My last run of Mu was the night before, from a different app bundle, and it worked fine. Looking at the log from that session the python exec was there before? Not sure how it got "lost":
```
2021-02-10 22:12:58,088 - mu.virtual_environment:371(ensure) DEBUG: Found existing virtual environment at /Users/microbit-carlos/Library/Application Support/mu/mu_venv
2021-02-10 22:12:58,088 - mu.virtual_environment:378(ensure_interpreter) INFO: Interpreter found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv/bin/python
2021-02-10 22:12:58,088 - mu.virtual_environment:388(ensure_pip) INFO: Pip found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv/bin/pip
```

tjguk, 2021-02-12:
I'm obviously not sure what's going on here, but given that it clearly can happen, I propose the following general approach:

* Name the venv folder with a timestamp -- the user generally doesn't need to look here, and if they do for diagnostic purposes, we should be able to identify it fairly readily.
* If we fail to create the venv -- or determine some other fault on startup -- we could simply create a fresh venv with a corresponding new timestamp. The worst that can happen is that venv with some user-installed packages gets corrupted and has to be replaced. Then the user might have to re-install the packages.
* The current fairly naive check to see whether the venv exists should extend into checking *each time we run* that we still have a valid venv and, if not, recreate as above
* The ensure/create could run inside some kind of retry loop to try to alleviate the problem in *this* issue; although I'd obviously be happier if I understood what was going wrong here...

carlosperate, 2021-02-12:
We will also need to check that the venv present "works" for the current Mu installation. 

Specially when updating the Mu app and we might have updated the Python versions. If the venv symlinks point to a python interpreter in the Mu installation directory that is still there (but being a different Py version), then they might still work but the dependencies in the venv might not be compatible.

tjguk, 2021-02-12:
Thanks @carlosperate  -- yes, this was exactly in my mind (I've even dropped a comment in the code to remind myself how to check the major/minor version from `sys.version_info`).

Basically I plan to refactor so that "ensure" runs every time and checks things like:

* Valid venv at all -- config file in place; valid interpreter; expected directories
* Same Python version
* Perhaps import key modules
* (Anything else which seems to have tripped us up)

Any failure of these will cause a new venv to be created, as, eg, `mu-venv.38.20210212100900` and the venv.ini will be updated with that name. A new Jupyter Kernel will also be created with the same unambiguous name, hopefully avoiding the issues in #1294

PR #1306 (2021-02-20): WIP: Make venv setup more robust
(fixes #1291)

Initial PoC PR to ensure greater robustness around the venv on startup. Key points:

* First checks whether a venv exists and is a good state
* If not, creates a new venv and checks again
* Doesn't loop any further atm: if the second check fails, Mu will bomb out
* Checks include:
    * Does the venv directory exist? Is it a venv?
    * Is there an interpreter and pip exe where expected?
    * Does the interpreter version (x.y) match the Python x.y Mu is running under?
    * Can we import key modules (pgzero, flask, serial)?
closed 2021-02-20 by tjguk