mu-editor/mu #1294: Issue with iPython REPL with Windows installer using Python 3.8

carlosperate, 2021-02-11:
With the current master branch executable builds we have the Python 3 (3.7) REPL working correctly.

Moving the CI to build the installers with Python 3.8 and we encounter the `Kernel died, restarting` issue:
[image]

```
-----------------

Starting Mu 1.1.0.beta.1
2021-02-11 16:37:41,193 - root:173(run) INFO: uname_result(system='Windows', node='DESKTOP-V8VNTI4', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-02-11 16:37:41,193 - root:174(run) INFO: Python path: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-02-11 16:37:41,193 - root:175(run) INFO: Language code: en_GB
2021-02-11 16:37:41,193 - mu.settings:220(load) WARNING: No settings file found at C:\Users\microbit-carlos\AppData\Local\python\mu\settings.json; skipping
2021-02-11 16:37:41,263 - mu.virtual_environment:371(ensure) DEBUG: Found existing virtual environment at C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv
2021-02-11 16:37:41,263 - mu.virtual_environment:378(ensure_interpreter) INFO: Interpreter found at C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\python.exe
2021-02-11 16:37:41,263 - mu.virtual_environment:388(ensure_pip) INFO: Pip found at C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe
2021-02-11 16:37:42,425 - mu.logic:758(__init__) INFO: Setting up editor.
2021-02-11 16:37:42,425 - mu.logic:778(__init__) INFO: Log directory: C:\Users\microbit-carlos\AppData\Local\python\mu\Logs
2021-02-11 16:37:42,425 - mu.logic:779(__init__) INFO: Data directory: C:\Users\microbit-carlos\AppData\Local\python\mu
2021-02-11 16:37:42,425 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-02-11 16:37:42,425 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 16:37:42,862 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 16:37:42,862 - mu.logic:1519(change_mode) INFO: Workspace directory: C:\Users\microbit-carlos\mu_code
2021-02-11 16:37:42,862 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\microbit-carlos\AppData\Local\python\mu\session.json>
2021-02-11 16:37:42,878 - mu.logic:889(restore_session) INFO: Loaded files.
2021-02-11 16:37:42,878 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-02-11 16:37:42,878 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-02-11 16:37:42,931 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 16:37:42,931 - mu.logic:1519(change_mode) INFO: Workspace directory: C:\Users\microbit-carlos\mu_code
2021-02-11 16:37:43,269 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-02-11 16:37:44,452 - mu.modes.python3:280(toggle_repl) INFO: Toggle REPL on.
2021-02-11 16:37:44,452 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 16:37:44,452 - mu.modes.python3:78(__init__) DEBUG: About to create KernelRunner for mu_venv, C:\Users\microbit-carlos\mu_code, [] 
2021-02-11 16:37:44,452 - mu.modes.python3:94(start_kernel) DEBUG: About to start kernel
2021-02-11 16:37:44,452 - mu.modes.python3:95(start_kernel) INFO: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-02-11 16:37:44,452 - mu.modes.python3:99(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2021-02-11 16:37:44,577 - traitlets:56(start_kernel) DEBUG: Starting kernel: ['C:\\Users\\microbit-carlos\\AppData\\Local\\python\\mu\\mu_venv\\scripts\\python.exe', '-m', 'ipykernel_launcher', '-f', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Temp\\tmpzhbmrxg3.json']
2021-02-11 16:37:44,577 - traitlets:544(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:63285
2021-02-11 16:37:44,592 - asyncio:623(__init__) DEBUG: Using proactor: IocpProactor
2021-02-11 16:37:44,592 - asyncio:623(__init__) DEBUG: Using proactor: IocpProactor
2021-02-11 16:37:44,592 - traitlets:149(shell_channel) DEBUG: connecting shell channel to tcp://127.0.0.1:63282
2021-02-11 16:37:44,592 - traitlets:544(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:63282
2021-02-11 16:37:44,592 - traitlets:161(iopub_channel) DEBUG: connecting iopub channel to tcp://127.0.0.1:63283
2021-02-11 16:37:44,592 - traitlets:544(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:63283
2021-02-11 16:37:44,592 - traitlets:173(stdin_channel) DEBUG: connecting stdin channel to tcp://127.0.0.1:63284
2021-02-11 16:37:44,592 - traitlets:544(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:63284
2021-02-11 16:37:44,592 - traitlets:185(hb_channel) DEBUG: connecting heartbeat channel to tcp://127.0.0.1:63286
2021-02-11 16:37:44,592 - traitlets:196(control_channel) DEBUG: connecting control channel to tcp://127.0.0.1:63285
2021-02-11 16:37:44,592 - traitlets:544(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:63285
2021-02-11 16:37:47,595 - traitlets:107(poll) INFO: KernelRestarter: restarting kernel (1/5), new random ports
2021-02-11 16:37:47,595 - traitlets:479(_handle_kernel_died) WARNING: kernel died: 3.002912759780884
2021-02-11 16:37:47,595 - traitlets:491(_handle_kernel_restarted) WARNING: kernel restarted
2021-02-11 16:37:47,610 - traitlets:56(start_kernel) DEBUG: Starting kernel: ['C:\\Users\\microbit-carlos\\AppData\\Local\\python\\mu\\mu_venv\\scripts\\python.exe', '-m', 'ipykernel_launcher', '-f', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Temp\\tmpzhbmrxg3.json']
2021-02-11 16:37:47,610 - traitlets:544(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:63335
2021-02-11 16:37:50,595 - traitlets:479(_handle_kernel_died) WARNING: kernel died: 3.000040054321289
2021-02-11 16:37:50,612 - traitlets:107(poll) INFO: KernelRestarter: restarting kernel (2/5), new random ports
2021-02-11 16:37:50,612 - traitlets:491(_handle_kernel_restarted) WARNING: kernel restarted
[... cut ...]
2021-02-11 16:37:52,195 - mu.virtual_environment:503(installed_packages) INFO: Discovering installed third party modules in venv.
2021-02-11 16:37:52,195 - mu.virtual_environment:163(run) DEBUG: About to run blocking: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe, ['list', '--disable-pip-version-check'], 30.0
2021-02-11 16:37:52,704 - mu.virtual_environment:523(installed_packages) INFO: []
```

@ntoll this explains why your builds had this issue, but the CI installers were working.

carlosperate, 2021-02-11:
Okay, this is actually a problem installing the dependencies in the venv, as qtconsole is one of them:
https://github.com/mu-editor/mu/blob/26abb94b309a48aa550524de832010a07b09ace8/mu/wheels/__init__.py#L20

```
-----------------

Starting Mu 1.1.0.beta.1
2021-02-11 16:40:41,061 - root:173(run) INFO: uname_result(system='Windows', node='DESKTOP-V8VNTI4', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-02-11 16:40:41,061 - root:174(run) INFO: Python path: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-02-11 16:40:41,061 - root:175(run) INFO: Language code: en_GB
2021-02-11 16:40:41,061 - mu.settings:220(load) WARNING: No settings file found at C:\Users\microbit-carlos\AppData\Local\python\mu\settings.json; skipping
2021-02-11 16:40:41,108 - mu.virtual_environment:360(ensure) DEBUG: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv does not exist; creating
2021-02-11 16:40:41,108 - mu.virtual_environment:400(create) INFO: Creating virtualenv: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv
2021-02-11 16:40:41,108 - mu.virtual_environment:401(create) INFO: Virtualenv name: mu_venv
2021-02-11 16:40:45,373 - mu.virtual_environment:447(install_baseline_packages) INFO: Installing baseline packages
2021-02-11 16:40:45,373 - mu.virtual_environment:448(install_baseline_packages) INFO: C:\Users\microbit-carlos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-02-11 16:40:45,389 - mu.virtual_environment:193(install) DEBUG: About to pip install: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\appdirs-1.4.4-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\backcall-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\black-20.8b1-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\click-7.1.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\colorama-0.4.4-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\decorator-4.4.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Flask-1.1.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipykernel-5.4.3-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython-7.19.0-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython-7.20.0-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython_genutils-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\itsdangerous-1.1.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jedi-0.18.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Jinja2-2.11.3-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jupyter_client-6.1.11-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jupyter_core-4.7.1-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\MarkupSafe-1.1.1-cp37-cp37m-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\MarkupSafe-1.1.1-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\mypy_extensions-0.4.3-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Ed [... cut ...]
2021-02-11 16:41:07,159 - mu.virtual_environment:503(installed_packages) INFO: Discovering installed third party modules in venv.
2021-02-11 16:41:07,159 - mu.virtual_environment:163(run) DEBUG: About to run blocking: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe, ['list', '--disable-pip-version-check'], 30.0
2021-02-11 16:41:07,615 - mu.virtual_environment:523(installed_packages) INFO: []
```

Somehow `MarkupSafe-1.1.1-cp37-cp37m-win_amd64.whl` makes it into the wheels folder:

[image]

-----

And the error stops the entire wheel installation process and nothing gets installed:

[image]

tjguk, 2021-02-12:
Although I've not really looked into how the Jupyter Kernel stuff works, it seems v. likely that if we give the kernel a unique name (and, frankly, it could just be a GUID) when we install the venv then we won't see interference with any previous one. I'll have to check the code later to see how we know which Kernel to pick when we're running the REPL, but hopefully we can identify the one we've created.

tjguk, 2021-02-12:
Ok; quickly noting for my sake -- the REPL is selected by the call to create the KernelRunner at https://github.com/mu-editor/mu/blob/master/mu/modes/python3.py#L300 where it uses the venv.name. So we could easily switch to the venv holding a specific kernel name and hopefully avoiding version ambiguity from there.

tjguk, 2021-02-12:
Thinking a little further... if we go the route I propose in https://github.com/mu-editor/mu/issues/1291#issuecomment-778005178 then the venv itself will have a unique name and the kernel will inherit that by virtue of the code linked above. So this might become moot.

carlosperate, 2021-02-12:
Sorry, I wrote all this yesterday and then forgot to click send 😅

Looks like the problem is related to the venv stuff in general rather than iPython in particular.
In my first report the issue was because somehow my `mu_installation_path/site_packages/mu/wheels` contained wheels for both Python 3.8 and 3.7 (in a 3.8 installation). I unsuccessfully tried to replicate it, and I still cannot figure out exactly how that happened.

Nevertheless having a venv for one Python version and then installing Mu with a different Python version also causes issues.

Below this was the original message I didn't send yesterday.

-----

Okay, so if I uninstall Mu Editor, delete the user data, and then install again the wheel is not present in the installation `mu/wheels` directory.

[image]

The new venv installs everything and all works fine.

-----

If I then uninstall Mu Editor, but leave the Python 3.8 venv and install a version of Mu with Python 3.7 ([this one](https://github.com/mu-editor/mu/actions/runs/557225345)), then I get errors launching the iPython REPL. But it's not the same problem where it tries to install the wrong wheel again.

```
-----------------

Starting Mu 1.1.0.beta.1
2021-02-11 22:08:56,829 - root:173(run) INFO: uname_result(system='Windows', node='DESKTOP-V8VNTI4', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-02-11 22:08:56,829 - root:174(run) INFO: Python path: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\python37.zip', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\microbit-carlos\\AppData\\Roaming\\Python\\Python37\\site-packages', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.7.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.7.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-02-11 22:08:56,829 - root:175(run) INFO: Language code: en_GB
2021-02-11 22:08:56,829 - mu.settings:220(load) WARNING: No settings file found at C:\Users\microbit-carlos\AppData\Local\python\mu\settings.json; skipping
2021-02-11 22:08:57,111 - mu.virtual_environment:371(ensure) DEBUG: Found existing virtual environment at C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv
2021-02-11 22:08:57,111 - mu.virtual_environment:378(ensure_interpreter) INFO: Interpreter found at C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\python.exe
2021-02-11 22:08:57,111 - mu.virtual_environment:388(ensure_pip) INFO: Pip found at C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe
2021-02-11 22:08:58,263 - mu.logic:758(__init__) INFO: Setting up editor.
2021-02-11 22:08:58,263 - mu.logic:778(__init__) INFO: Log directory: C:\Users\microbit-carlos\AppData\Local\python\mu\Logs
2021-02-11 22:08:58,263 - mu.logic:779(__init__) INFO: Data directory: C:\Users\microbit-carlos\AppData\Local\python\mu
2021-02-11 22:08:58,263 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-02-11 22:08:58,263 - mu.modes.base:63(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 22:08:58,782 - mu.modes.base:63(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 22:08:58,782 - mu.logic:1519(change_mode) INFO: Workspace directory: C:\Users\microbit-carlos\mu_code
2021-02-11 22:08:58,798 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\microbit-carlos\AppData\Local\python\mu\session.json>
2021-02-11 22:08:58,798 - mu.logic:889(restore_session) INFO: Loaded files.
2021-02-11 22:08:58,798 - mu.logic:893(restore_session) INFO: User defined environment variables: []
2021-02-11 22:08:58,798 - mu.logic:898(restore_session) INFO: Minify scripts on micro:bit? False
2021-02-11 22:08:58,860 - mu.modes.base:63(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 22:08:58,860 - mu.logic:1519(change_mode) INFO: Workspace directory: C:\Users\microbit-carlos\mu_code
2021-02-11 22:08:59,251 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-02-11 22:09:01,460 - mu.modes.python3:280(toggle_repl) INFO: Toggle REPL on.
2021-02-11 22:09:01,460 - mu.modes.base:63(get_default_workspace) INFO: Using workspace C:\Users\microbit-carlos\mu_code from settings file
2021-02-11 22:09:01,460 - mu.modes.python3:82(__init__) DEBUG: About to create KernelRunner for mu_venv, C:\Users\microbit-carlos\mu_code, [] 
[... cut ...]
2021-02-11 22:24:53,527 - mu.virtual_environment:503(installed_packages) INFO: Discovering installed third party modules in venv.
2021-02-11 22:24:53,527 - mu.virtual_environment:163(run) DEBUG: About to run blocking: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe, ['list', '--disable-pip-version-check'], 30.0
2021-02-11 22:24:55,106 - mu.virtual_environment:523(installed_packages) INFO: ['File', 'ImportError:']
```

-----

I still don't know how I got the original issue, as the installed Mu Editor (Py3.8) `site-packages/mu/wheels/` directory had wheels for both Python 3.7 and 3.8.

tjguk, 2021-02-12:
I had already thought of adding a "clear-out" at the start of the `mu.wheels` download run as I'd been bitten by a variant of this problem.

tjguk, 2021-02-12:
Just merged #1294 which clears any existing wheels before downloading. Should prevent "rogue" wheels making their way into the installer

carlosperate, 2021-02-12:
Great, thanks Tim!
The other issue mentioned here (venvs from a different python version) should be covered by the changes listed in https://github.com/mu-editor/mu/issues/1291.

I'll close this ticket as that work is already captured over there.

closed 2021-02-12 by carlosperate

ntoll, 2021-02-14:
"Rogue Wheels" - sounds like a new StarWars movie. :-)