mu-editor/mu #1295: Clicking the splash screen on first run closes it while Mu continues to initialise

carlosperate, 2021-02-11:
This can be a bit disconcerting as it looks like nothing is happening.

Eventually Mu will open, but if the user tries to open Mu again while the previous run is still initialising, which is very **likely to happen**, it could cause unexpected issues as the processes interfeer with each other.

```
-----------------

Starting Mu 1.1.0.beta.1
2021-02-11 22:37:55,874 - root:173(run) INFO: uname_result(system='Windows', node='DESKTOP-V8VNTI4', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-02-11 22:37:55,874 - root:174(run) INFO: Python path: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-02-11 22:37:55,874 - root:175(run) INFO: Language code: en_GB
2021-02-11 22:37:55,874 - mu.settings:220(load) WARNING: No settings file found at C:\Users\microbit-carlos\AppData\Local\python\mu\settings.json; skipping
2021-02-11 22:37:55,882 - mu.settings:220(load) WARNING: No settings file found at C:\Users\microbit-carlos\AppData\Local\python\mu\session.json; skipping
2021-02-11 22:37:56,059 - mu.virtual_environment:360(ensure) DEBUG: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv does not exist; creating
2021-02-11 22:37:56,059 - mu.virtual_environment:400(create) INFO: Creating virtualenv: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv
2021-02-11 22:37:56,059 - mu.virtual_environment:401(create) INFO: Virtualenv name: mu_venv
2021-02-11 22:38:00,270 - mu.virtual_environment:447(install_baseline_packages) INFO: Installing baseline packages
2021-02-11 22:38:00,270 - mu.virtual_environment:448(install_baseline_packages) INFO: C:\Users\microbit-carlos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-02-11 22:38:00,270 - mu.virtual_environment:193(install) DEBUG: About to pip install: ['C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\backcall-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\click-7.1.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\colorama-0.4.4-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\decorator-4.4.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Flask-1.1.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipykernel-5.4.3-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython-7.20.0-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython_genutils-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\itsdangerous-1.1.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jedi-0.18.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Jinja2-2.11.3-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jupyter_client-6.1.11-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jupyter_core-4.7.1-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\MarkupSafe-1.1.1-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\nudatus-0.0.5-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\numpy-1.20.1-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\parso-0.8.1-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\pgzero-1.2-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\pickleshare-0.7.5-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\prompt_toolkit-3.0.16-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\pygame-2.0.1-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Pygments-2.7.4-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\pyserial-3.4-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\python_dateutil-2.8.1-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\pywin32-300-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\pyzmq-22.0.2-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\qtconsole-4.7.4-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\QtPy-1.9.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\setuptools-53.0.0-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\six-1.15.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\tornado-6.1-cp38-cp38-win_amd64.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\traitlets-5.0.5-py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\wcwidth-0.2.5-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Werkzeug-1.0.1-py2.py3-none-any.whl']
2021-02-11 22:38:00,270 - mu.virtual_environment:163(run) DEBUG: About to run blocking: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe, ['install', '--disable-pip-version-check', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\backcall-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\click-7.1.2-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\colorama-0.4.4-py2.py3-none-any.whl', 'C:\\Users\\microbit-carlos\\AppData\\Local [... cut ...]
2021-02-11 22:38:24,230 - mu.virtual_environment:163(run) DEBUG: About to run blocking: C:\Users\microbit-carlos\AppData\Local\python\mu\mu_venv\scripts\pip.exe, ['list', '--disable-pip-version-check'], 30.0
2021-02-11 22:38:25,065 - mu.virtual_environment:523(installed_packages) INFO: ['backcall', 'colorama', 'decorator', 'ipython', 'ipython-genutils', 'jedi', 'jupyter-client', 'jupyter-core', 'MarkupSafe', 'parso', 'pickleshare', 'pip', 'prompt-toolkit', 'pygame', 'Pygments', 'python-dateutil', 'pywin32', 'pyzmq', 'QtPy', 'setuptools', 'six', 'tornado', 'traitlets', 'wcwidth', 'Werkzeug', 'wheel']

```

tjguk, 2021-02-12:
Not sure whether to make this comment here or elsewhere, but in the light of the issues we're seeing with startup, I really think we'll need to go to some kind of text output mode such as PoC-ed by @devdanzin. As it stands there are too many things which can silently fail as part of the venv startup. Obviously we're trying to make that more robust so it *doesn't* fail and/or recovers well. But with no output to the user whatsoever it seems like a recipe for poor UX.

carlosperate, 2021-02-12:
I agree that being able to provide the users with a bit more info during startup will be useful. I'll continue this thought in https://github.com/mu-editor/mu/issues/1294.

As far as splash dismissal behaviour goes, I'm not sure what would be the most elegant way to implement a change in this area, but thinking just about the UX I'd suggest something like this:
- The splash screen should not be closable during initial startup
- In subsequent starts (when not creating all the venv stuff) it should maintain the same behaviour as now, where it can be dismissed with a click
    - I believe this has been quite useful since, at least in my laptop, Mu fully starts up quicker than the splash screen automatically disappears

ntoll, 2021-02-14:
In yet another "Nicholas is an idiot" moment, I've inadvertently removed some code introduced to ensure our splash screen never disappears when clicked. I'll re-add with some upcoming work on adding textual output to the splash screen.

carlosperate, 2021-03-17:
This has been fixed before the beta.2 release.

closed 2021-03-17 by carlosperate