mu-editor/mu #1298: Serial pane is difficult to resize

hwiguna, 2021-02-14:
Issue:
Hot spot for resizing the serial pane is too small and sometimes I cannot resize it at all because mouse cursor remains as arrow rather than showing the resize cursor.

I'm running Mu 1.0.3 under windows 10.

[image]

dybber, 2021-02-15:
I have that problem as well on Mac OS X Mojave.

ntoll, 2021-02-15:
This has been a long term pain-in-the-butt for me too... I've just spent 15 minutes Googling what the thing is in the Qt world, and apparently we need to figure out how to properly configure the `QSplitter class (https://doc.qt.io/qt-5/qsplitter.html). I imagine there may be a way to pass in a default (i.e. larger) size for these things into the parent window object. But will require more work.

ntoll, 2021-02-15:
Scratch that... this appears to be the answer we need (I've just tried it, and it works): https://stackoverflow.com/questions/4751594/how-can-i-change-the-width-of-a-qmainwindow-separator-without-it-affecting-the-q

I'll fix the existing CSS so the separator bar is slightly bigger.

hwiguna, 2021-02-15:
> Scratch that... this appears to be the answer we need (I've just tried it, and it works): https://stackoverflow.com/questions/4751594/how-can-i-change-the-width-of-a-qmainwindow-separator-without-it-affecting-the-q
> 
> I'll fix the existing CSS so the separator bar is slightly bigger.

Thank you!

ntoll, 2021-02-15:
No problem... this has been bugging me so much for years..! ;-)

ZanderBrown, 2021-02-16:
I had no idea people were having problems, sorry everyone!

ntoll, 2021-02-16:
No need to apologise..! Here's where I was hoping to get us:

For each theme:

* Make the separator a bit bigger.
* When the mouse is hovered over the separator, it changes colour slightly so you can see it's there and have something to aim for.

That's it. All do-able in CSS AFAICT.

dybber, 2021-02-19:
A fix have just been merged! I'm closing this issue. The fix will be part of the next beta release of Mu.

closed 2021-02-19 by dybber