mu-editor/mu #1305: MU is not compatible with M5Stack running UIFlow

mhfalken, 2021-02-18:
When I try to use the M5StickC running UIFlow (Python mode) together with the MU editor it doesn't work.
When I press 'Files', I get the following error:
[image]

When I press 'Run', I get the following error code in REPL (and the code don't run):
[image]
Sometimes it reboot first, and then show the error.

If I press REPL it seems to work, if I press CTRL-C a few times.

I had hoped that I could use the M5Stack devices together with the MU editor.
/Michael

MU version: Mu 1.1.0.alpha.2
M5Stack UIFlow (M5StickC): version: 1.7.2,

Here is the full log. The tests are in the end of the log.
2021-02-18 13:33:42,953 - root:122(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2021-02-18 13:33:42,956 - root:123(run) INFO: uname_result(system='Windows', node='Michaels-Legion', release='10', version='10.0.19041', machine='AMD64', processor='AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD')
2021-02-18 13:33:42,956 - root:124(run) INFO: Python path: ['C:\\Users\\Michael\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Michael\\AppData\\Local\\Mu', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Michael\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2021-02-18 13:33:42,956 - root:125(run) INFO: Language code: en_DK
2021-02-18 13:33:43,136 - mu.logic:845(__init__) INFO: Setting up editor.
2021-02-18 13:33:43,137 - mu.logic:867(__init__) INFO: Settings path: C:\Users\Michael\AppData\Local\python\mu\settings.json
2021-02-18 13:33:43,137 - mu.logic:868(__init__) INFO: Session path: C:\Users\Michael\AppData\Local\python\mu\session.json
2021-02-18 13:33:43,137 - mu.logic:869(__init__) INFO: Log directory: C:\Users\Michael\AppData\Local\python\mu\Logs
2021-02-18 13:33:43,137 - mu.logic:870(__init__) INFO: Data directory: C:\Users\Michael\AppData\Local\python\mu
2021-02-18 13:33:43,153 - mu.logic:884(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2021-02-18 13:33:43,660 - mu.logic:1612(change_mode) INFO: Workspace directory: C:\Users\Michael\mu_code
2021-02-18 13:33:43,666 - mu.logic:969(restore_session) INFO: Restoring session from: C:\Users\Michael\AppData\Local\python\mu\session.json
2021-02-18 13:33:43,666 - mu.logic:970(restore_session) DEBUG: {'theme': 'night', 'mode': 'esp', 'paths': ['C:\\Users\\Michael\\Documents\\Projects\\m5stack\\axp-test.py', 'C:\\Users\\Michael\\Documents\\Projects\\m5stack\\sound-test.py', 'C:\\Users\\Michael\\Documents\\Projects\\m5stack\\test.py', 'C:\\Users\\Michael\\Documents\\Projects\\m5stack\\gyro-test.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': 87, 'y': 118, 'w': 1448, 'h': 1167}}
2021-02-18 13:33:43,673 - mu.logic:1073(_load) INFO: Loading script from: C:\Users\Michael\Documents\Projects\m5stack\axp-test.py
2021-02-18 13:33:43,679 - mu.logic:351(read_and_decode) DEBUG: Trying to decode with utf-8
2021-02-18 13:33:43,679 - mu.logic:354(read_and_decode) INFO: Decoded with utf-8
2021-02-18 13:33:43,680 - mu.logic:367(read_and_decode) DEBUG: Detected newline '\n'
2021-02-18 13:33:43,680 - mu.logic:1172(_load) DEBUG: from m5stack import lcd
from m5stack import axp
import time

while True:
    y=5
    dy = 15
    lcd.clear()
    lcd.orient(lcd.LANDSCAPE_FLIP)

    if axp.getChargeState():
        lcd.text(10, y, "Charging, %imA" %axp.getBatCurrent())
        y += dy

    lcd.text(10, y, "Batt= %3.1fV" %axp.getBatVoltage())
    y += dy
    #lcd.text(10, y, "In= %iV, %imA" %(axp.getVinVoltage(), axp.getVinCurrent()))
    y += dy
    time.sleep(1)

2021-02-18 13:33:44,173 - mu.logic:1073(_load) INFO: Loading script from: C:\Users\Michael\Documents\Projects\m5stack\sound-test.py
2021-02-18 13:33:44,179 - mu.logic:351(read_and_decode) DEBUG: Trying to decode with utf-8
2021-02-18 13:33:44,179 - mu.logic:354(read_and_decode) INFO: Decoded with utf-8
2021-02-18 13:33:44,179 - mu.logic:367(read_and_decode) DEBUG: Detected newline '\r\n'
2021-02-18 13:33:44,179 - mu.logic:1172(_load) DEBUG: from machine import I2S, SDCard, Pin
from m5stack import *
import os, uos
from wav import wave

#mount the sd card
sd = SDCard(slot = 2, sck = Pin(23), miso = Pin(33), mosi = Pin(19), freq = 10000000)

[... 883 lines cut ...]
2021-02-18 13:47:07,771 - mu.modes.base:173(execute) INFO: Sending command b'\x03'
2021-02-18 13:47:07,773 - mu.modes.base:173(execute) INFO: Sending command b'print("\\n");'
2021-02-18 13:47:07,776 - mu.modes.base:173(execute) INFO: Sending command b'from m5stack import lcd\r'
2021-02-18 13:47:07,779 - mu.modes.base:173(execute) INFO: Sending command b'import imu\r'
2021-02-18 13:47:07,781 - mu.modes.base:173(execute) INFO: Sending command b'import time\r'
2021-02-18 13:47:07,783 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-18 13:47:07,785 - mu.modes.base:173(execute) INFO: Sending command b'myIMU = imu.IMU()\r'
2021-02-18 13:47:07,788 - mu.modes.base:173(execute) INFO: Sending command b'lcd.font(lcd.FONT_DejaVu18)\r'
2021-02-18 13:47:07,790 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-18 13:47:07,793 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-18 13:47:07,795 - mu.modes.base:173(execute) INFO: Sending command b'while True:\r'
2021-02-18 13:47:07,797 - mu.modes.base:173(execute) INFO: Sending command b'\tlcd.clear()\r'
2021-02-18 13:47:07,800 - mu.modes.base:173(execute) INFO: Sending command b'\tlcd.orient(lcd.LANDSCAPE_FLIP)\r'
2021-02-18 13:47:07,802 - mu.modes.base:173(execute) INFO: Sending command b'\t#v = myIMU.acceleration\r'
2021-02-18 13:47:07,804 - mu.modes.base:173(execute) INFO: Sending command b'\tv = myIMU.gyro\r'
2021-02-18 13:47:07,806 - mu.modes.base:173(execute) INFO: Sending command b'\tlcd.text(10, 10, "X= %.2f" %v[0])\r'
2021-02-18 13:47:07,809 - mu.modes.base:173(execute) INFO: Sending command b'\tlcd.text(10, 30, "Y= %.2f" %v[1])\r'
2021-02-18 13:47:07,811 - mu.modes.base:173(execute) INFO: Sending command b'\tlcd.text(10, 50, "Z= %.2f" %v[2])\r'
2021-02-18 13:47:07,813 - mu.modes.base:173(execute) INFO: Sending command b'\ttime.sleep(0.5)\r'
2021-02-18 13:47:07,816 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-18 13:47:07,818 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-18 13:47:07,820 - mu.modes.base:173(execute) INFO: Sending command b'\x04'
2021-02-18 13:47:07,823 - mu.modes.base:173(execute) INFO: Sending command b'\x02'
2021-02-18 13:47:40,013 - mu.logic:1491(show_admin) INFO: Showing admin with logs from C:\Users\Michael\AppData\Local\python\mu\Logs\mu.log
2021-02-18 13:47:40,013 - mu.logic:201(installed_packages) INFO: Packages found: []

dybber, 2021-02-18:
Thanks for reporting this, is the `crc error2k` and the question marks the only thing printed in the REPL?

@mhfalken and I had a video call yesterday, and I think we can add that it seems this problem only occurs in Mu, not when using e.g. Visual Studio Code and their extensions for interacting with M5Stack devices. Perhaps there's something we could learn from reading their code: https://github.com/curdeveryday/vscode-m5stack-mpy/blob/master/src/serialport.js

Currently I can't replicate the `crc error2k`-issue with the REPL on Mac OS X, but I can replicate the issue about M5Stack devices and not being able to open the file system pane. I will try to investigate further...

mhfalken, 2021-02-18:
Yes, this is the only thing in the REPL when I press Run.

carlosperate, 2021-02-18:
Entering the MicroPython raw REPL can fail depending on the user code, were the devices running something?
We should definitely make it more robust, but to reduce factors that could be affecting this it'd be good to check first it still fails with no user code.

dybber, 2021-02-18:
The M5Stack firmware runs a start-up script that's rather long running, initializing WiFi. Probably even interactive until some point, waiting for user to select what to do, I think.

So it's not even user code, it's just built-in when they ship these devices. When we execute user code, we do abort the current program with keyboard interrupts before executing the program:

https://github.com/mu-editor/mu/blob/52cb622575f60290318892c42ebc5a6a8e0665b9/mu/modes/base.py#L168-L189

carlosperate, 2021-02-18:
> The M5Stack firmware runs a start-up script that's rather long running, initializing WiFi. Probably even interactive until some point, waiting for user to select what to do, I think.

Can the REPL be accessed from boot.py? I thought it can't until the user code starts executing.
How long does this take? Is it waiting for user interaction and therefore never letting the user code run?

> When we execute user code, we do abort the current program with keyboard interrupts before executing the program:

Yes, but that can take several attempts, and in serial heavy programmes Mu might lock up a bit and somehow affect sending the commands as weel. Apart from that, if Mu is reading serial data to determine if the commands worked, like when it needs to transfer files, since it has to confirm the RAW REPL has been activated, the serial pipeline can be full of old user data and also affect the detection of the `raw REPL; CTRL-B to exit` string.

So many things can affect the process, so to identify the underlying problem it'd be better if we could test with a little else going on first, and then start introducing more things until it stops working.

dybber, 2021-02-19:
> Can the REPL be accessed from boot.py? I thought it can't until the user code starts executing.
> How long does this take? Is it waiting for user interaction and therefore never letting the user code [...]

I'm not sure what you mean "Can the REPL be accessed from boot.py?", but e.g. if you do `print(something)` in boot.py it will be printed to serial and shown in the Mu REPL-pane. The MicroPython REPL where user can enter their commands will first be made available after boot.py and main.py is executed.

It can take 30-60+ seconds for firmware on M5Stack devices or Edublocks to discover that it couldn't connect to WiFi. On M5Stack it seems like it's just trying indefinitely... 

IMO we could perhaps wait 2 seconds for startup code to run, but not much more than that, and if it's not completed we should abort the start-up and go on with out business (list files or execute user code). 

In my opinion M5Stack, Edublocks, etc. are abusing boot.py to do what should be part of application code (e.g. connecting to Wifi), and shouldn't be part of the boot process. However, we should at least try to let the boot process complete, for those devices who actually are well-behaved.

dybber, 2021-02-19:
> So many things can affect the process, so to identify the underlying problem it'd be better if we could test with a little else going on first, and then start introducing more things until it stops working.

I agree. @mhfalken could we get you to try running the same experiment as before, but with only a simple `print("Hello world")` programme and grab the logs one more time? Just to make sure none of your own code is interfering in anyway.

carlosperate, 2021-02-19:
> I'm not sure what you mean "Can the REPL be accessed from boot.py?", but e.g. if you do `print(something)` in boot.py it will be printed to serial and shown in the Mu REPL-pane. The MicroPython REPL [...]

Yes, sorry, I meant if sending "ctrl+c" while boot.py was running would halt execution of boot.py and start the REPL. Sounds like "ctrl+c" to enter the REPL only works from main.py, right?

> It can take 30-60+ seconds for firmware on M5Stack devices or Edublocks to discover that it couldn't connect to WiFi. On M5Stack it seems like it's just trying indefinitely...

Edublocks meaning https://edublocks.org ?
Are they compatible with this device? Or are you referring to this problem with other devices as well?

> IMO we could perhaps wait 2 seconds for startup code to run, but not much more than that, and if it's not completed we should abort the start-up and go on with out business (list files or execute user code).

I think 2 seconds from pressing something like the "files" button and the panel opening is a bit too long. I guess we if we do that we would have to show some kind of spinner or "connecting" view.

Btw, what does work while boot.py runs? If "ctrl+c" does nothing, do other shorcuts, like the soft reboot, work?

dybber, 2021-02-19:
> Yes, sorry, I meant if sending "ctrl+c" while boot.py was running would halt execution of boot.py and start the REPL. Sounds like "ctrl+c" to enter the REPL only works from main.py, right?

That was also my assumption, and I think I errorneously came to that conclusion in a previous issue/PR here on Github as well. But no, sending CTRL-C does abort any running boot.py scripts.

> Edublocks meaning https://edublocks.org ?
> Are they compatible with this device? Or are you referring to this problem with other devices as well?

I only own M5Stack devices, not the Edublocks. The issue is however the same, as has been reported previously: https://github.com/mu-editor/mu/issues/985

> I think 2 seconds from pressing something like the "files" button and the panel opening is a bit too long. I guess we if we do that we would have to show some kind of spinner or "connecting" view.

I just attempted the following:
https://github.com/dybber/microfs/blob/3d51c22b5c2585c91b22ce4055a6a7c5f39ab14e/microfs.py#L104-L114

Where we:
 - first do soft-reboot
 - try to wait for entering the RAW repl
 - if a timeout occurs in serial.read_until, and we thus don't enter raw REPL, we then do two keyboard interrupts before we attempt to enter raw repl again. 
   
This seems to work, and the timeout for serial.read_until is 1 seconds:
https://github.com/dybber/microfs/blob/3d51c22b5c2585c91b22ce4055a6a7c5f39ab14e/microfs.py#L133
It does take a bit more than a second, but it will not introduce further delays for "well-behaved" devices.

It's still just a suggestion, it will need further scrutiny and checks on more devices.

mhfalken, 2021-02-19:
I have now tried with the print("Hello world") and I get the same error on REPL as before. 
[image]

The log is here:

2021-02-19 15:56:25,345 - root:122(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2021-02-19 15:56:25,347 - root:123(run) INFO: uname_result(system='Windows', node='Michaels-Legion', release='10', version='10.0.19041', machine='AMD64', processor='AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD')
2021-02-19 15:56:25,347 - root:124(run) INFO: Python path: ['C:\\Users\\Michael\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Michael\\AppData\\Local\\Mu', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Michael\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\Michael\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2021-02-19 15:56:25,347 - root:125(run) INFO: Language code: en_DK
2021-02-19 15:56:25,503 - mu.logic:845(__init__) INFO: Setting up editor.
2021-02-19 15:56:25,503 - mu.logic:867(__init__) INFO: Settings path: C:\Users\Michael\AppData\Local\python\mu\settings.json
2021-02-19 15:56:25,503 - mu.logic:868(__init__) INFO: Session path: C:\Users\Michael\AppData\Local\python\mu\session.json
2021-02-19 15:56:25,503 - mu.logic:869(__init__) INFO: Log directory: C:\Users\Michael\AppData\Local\python\mu\Logs
2021-02-19 15:56:25,503 - mu.logic:870(__init__) INFO: Data directory: C:\Users\Michael\AppData\Local\python\mu
2021-02-19 15:56:25,520 - mu.logic:884(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2021-02-19 15:56:25,913 - mu.logic:1612(change_mode) INFO: Workspace directory: C:\Users\Michael\mu_code
2021-02-19 15:56:25,918 - mu.logic:969(restore_session) INFO: Restoring session from: C:\Users\Michael\AppData\Local\python\mu\session.json
2021-02-19 15:56:25,918 - mu.logic:970(restore_session) DEBUG: {'theme': 'night', 'mode': 'esp', 'paths': ['C:\\Users\\Michael\\Documents\\Projects\\m5stack\\gyro-test.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': 87, 'y': 118, 'w': 1448, 'h': 1167}}
2021-02-19 15:56:25,924 - mu.logic:1073(_load) INFO: Loading script from: C:\Users\Michael\Documents\Projects\m5stack\gyro-test.py
2021-02-19 15:56:25,929 - mu.logic:351(read_and_decode) DEBUG: Trying to decode with utf-8
2021-02-19 15:56:25,929 - mu.logic:354(read_and_decode) INFO: Decoded with utf-8
2021-02-19 15:56:25,930 - mu.logic:367(read_and_decode) DEBUG: Detected newline '\n'
2021-02-19 15:56:25,930 - mu.logic:1172(_load) DEBUG: from m5stack import lcd
import imu
import time

myIMU = imu.IMU()
lcd.font(lcd.FONT_DejaVu18)

while True:
	lcd.clear()
	lcd.orient(lcd.LANDSCAPE_FLIP)
	#v = myIMU.acceleration
	v = myIMU.gyro
	lcd.text(10, 10, "X= %.2f" %v[0])
	lcd.text(10, 30, "Y= %.2f" %v[1])
	lcd.text(10, 50, "Z= %.2f" %v[2])
	time.sleep(0.5)

2021-02-19 15:56:26,317 - mu.logic:992(restore_session) INFO: Loaded files.
2021-02-19 15:56:26,317 - mu.logic:997(restore_session) INFO: User defined environment variables: []
2021-02-19 15:56:26,317 - mu.logic:1003(restore_session) INFO: Minify scripts on micro:bit? False
2021-02-19 15:56:26,355 - mu.logic:1612(change_mode) INFO: Workspace directory: C:\Users\Michael\mu_code
2021-02-19 15:56:26,518 - mu.logic:833(check_usb) INFO: esp device connected on port: COM3(VID: 0x0403, PID: 0x6001, manufacturer: 'FTDI')
2021-02-19 15:56:30,161 - mu.logic:1057(new) INFO: Added a new tab.
2021-02-19 15:56:46,094 - mu.modes.esp:172(run) INFO: Running script.
2021-02-19 15:56:46,094 - mu.modes.base:123(open) INFO: Connecting to REPL on port: COM3
2021-02-19 15:56:46,131 - mu.modes.base:141(open) INFO: Connected to REPL on port: COM3
2021-02-19 15:56:46,135 - mu.modes.base:507(add_repl) INFO: Started REPL on port: COM3
2021-02-19 15:56:46,135 - mu.modes.base:478(toggle_repl) INFO: Toggle REPL on.
2021-02-19 15:56:46,136 - mu.modes.base:199(send_commands) INFO: [b'\x03', b'\x03', b'\x01', b'\x04', b'\x03', b'\x03', b'print("\\n");', b'print("Hello world")\r', b'\r', b'\r', b'\x04', b'\x02']
2021-02-19 15:56:46,136 - mu.modes.base:173(execute) INFO: Sending command b'\x03'
2021-02-19 15:56:46,142 - mu.modes.base:173(execute) INFO: Sending command b'\x03'
2021-02-19 15:56:46,146 - mu.modes.base:173(execute) INFO: Sending command b'\x01'
2021-02-19 15:56:46,148 - mu.modes.base:173(execute) INFO: Sending command b'\x04'
2021-02-19 15:56:46,150 - mu.modes.base:173(execute) INFO: Sending command b'\x03'
2021-02-19 15:56:46,152 - mu.modes.base:173(execute) INFO: Sending command b'\x03'
2021-02-19 15:56:46,154 - mu.modes.base:173(execute) INFO: Sending command b'print("\\n");'
2021-02-19 15:56:46,156 - mu.modes.base:173(execute) INFO: Sending command b'print("Hello world")\r'
2021-02-19 15:56:46,158 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-19 15:56:46,161 - mu.modes.base:173(execute) INFO: Sending command b'\r'
2021-02-19 15:56:46,163 - mu.modes.base:173(execute) INFO: Sending command b'\x04'
2021-02-19 15:56:46,166 - mu.modes.base:173(execute) INFO: Sending command b'\x02'
2021-02-19 15:56:57,512 - mu.logic:1491(show_admin) INFO: Showing admin with logs from C:\Users\Michael\AppData\Local\python\mu\Logs\mu.log
2021-02-19 15:56:57,512 - mu.logic:201(installed_packages) INFO: Packages found: []

aivarannamaa, 2021-02-19:
Somebody has met similar error elsewhere, again related to `os.listdir`: https://community.m5stack.com/topic/2070/when-are-we-going-to-get-new-firmware/12

aivarannamaa, 2021-02-19:
While meeting similar problems in Thonny, the first thing I check is: does it go away when I reduce the write block size or increase the delay between blocks. Most boards/ports are rather fragile in this regard and it's easy to overfill the input buffer (sometimes because of bugs in the firmware, sometimes because something (eg. flushing a file) makes everything else stall for serveral seconds).

dybber, 2021-02-20:
Hi @aivarannamaa, thanks! I think you're right with experimenting with timing/delays, but it's a bit difficult when I can't replicate the issue myself. 

Right now our delays in Mu are 2 milliseconds, which we could try to increase. 

I think @mhfalken and I perhaps could try investigating it a bit over a video call, or wait until lockdown ends here in Denmark, and meet in person (we live around 20-25 km apart)

mhfalken, 2021-02-21:
@dybber I am open for a debug session either online or later in person.

dybber, 2021-02-22:
> @dybber I am open for a debug session either online or later in person.

Great, I'll just write to you outside Github to try and schedule it

dybber, 2021-02-23:
Okay, @mhfalken and I just had a longer debugging session, trying various things. It's very unpredictable and confusing, so we're not sure we actually learned anything from the debugging session other than we might point blame towards the M5Stack MicroPython firmware itself.

**Firmware**
The M5Stack firmware version 1.7.2 doesn't seem to work either on Mu or with Thonny, at least for running code in the REPL. 

It seems that the 1.7.2 version of the firmware has a boot-procedure that can't be disabled, unlike previous versions of their boot procedures which were in boot.py. We had better luck with 1.4.5.1 firmware from M5Stack, where we could modify boot.py to avoid it trying to log onto WiFi etc.

It also seems that they distribute different firmware depending on the model e.g. M5StickC vs M5 Core, even though they have the same name and version numbers. It's all very confusing.

**Timing**
We tried changing the delay between commands send in the MicroPython mode, that is in the line `QTimer.singleShot(2, remaining_task)` `in mu/modes/base.py`.
If @mhfalken changed it to 50 ms or 80 ms executing code through the "Run"-button still didn't work, if @mhfalken changed it to 100 ms or 120 ms it did work, however, if we increased it even further to 150 ms to 200 ms it stopped working again.

So we're a bit confused. Do we really need to delay it that much, 100 ms between every line send, seems like a lot - that's a second per 10 lines, or is something else at play? And why wouldn't it work if we increased the delay even further?

**File system access**
Accessing the file system did seem to work in Thonny, but not in Mu.

We didn't play around with microfs.py and file system access, it seemed to work sporadically with the 1.4.5.1 firmware and not at all with the 1.7.2 version of the firmware.

**Other**
@mhfalken have found an VSCode extension for interfacing with M5Stack's MicroPython devices, which seems to connect through their own protocol. That is, it doesn't communicate with the MicroPython REPL directly, instead with a MicroPython program running on the device. The code is here: https://github.com/curdeveryday/vscode-m5stack-mpy/blob/master/src/serialport.js