mu-editor/mu #1312: SVG-versions of Mu-buttons and icons

dybber, 2021-02-20:
Currently Mu's buttons and icons are .PNG images, and thus not very easy to change and customize. It has been requested in #970 that we make new versions of the buttons for the "dark mode" and the "high contrast" modes, but I believe the best way to create those buttons, would be to first make SVG-versions of the existing buttons, and then experiment with the colors to create the dark-mode buttons.

The current Mu-buttons that will need to be recreated are available at: https://github.com/mu-editor/mu/tree/master/mu/resources/images

When creating the icons it can be useful to use these vector icons as a starting point:
https://github.com/mu-editor/Mu-tate/tree/master/mu/icons

I think we want to go for an almost exact identical look and feel as the current Mu icons and buttons.

ntoll, 2021-02-20:
OK... so the markup/CSS in "Mu-tate" should help with working out how to display the buttons as SVG. I may take a look today if I find the time.

ntoll, 2021-02-20:
OK... Just made a start by changing the images to SVG. Here they are pre-styling... ;-)

[image]

ntoll, 2021-02-20:
So, this is FAR more complicated than I at first thought.

The buttons in the `QToolBar` area are defined as `QAction` instances (as per idiomatic Qt). However, `QAction` instances require a `QIcon` instance to define the image. It turns out `QIcon` instances cannot be styled by CSS - hence no progress further than the black SVGs.

Unless we completely re-write how we do the button bar and move away from `QToolBar` / `QAction` (which feels like a LOT of work), I don't think using SVG for buttons will be possible.

I'm going to close this as a "won't fix" due to the problems outlined above. :-/

Please feel free to re-open if you're able to think of a quick/simple fix.

dybber, 2021-02-20:
I wasn't thinking that we should do that. No CSS is necessary for what I propose. I was thinking that for every PNG file, we should get someone to create a matching SVG file.

For example, we have the Run button styled in this way: https://github.com/mu-editor/mu/blob/master/mu/resources/images/run.png
I'd like to have that very same image, including the white background and orange border in SVG-format. 

That should be doable. Then when we have those SVG icons/buttons, it will be easy to make other SVG-versions for other color schemes.

dybber, 2021-02-20:
And I don't think we should do it, but that it would be possible to find someone skilled in Inkscape/Illustrator to make the conversion from PNG to SVG

ntoll, 2021-02-20:
Aha.... OK. I see what you mean now. I imagined we'd use CSS file since that's how you usually do these things in the world of web development. I'm going to re-open since your way is far superior. :-)

dybber, 2021-02-20:
I think it would be nice to make a task like this available for someone with graphical skills, but in principle might not know anything about Qt/QSS. I don't think it's a good idea if we start doing too much graphical work (not layouting) with QSS/CSS.

ntoll, 2021-02-20:
Yes.... I agree completely. It's also a way to welcome folks with diverse skills into the our band of happy Mu-tants. ;-)

carlosperate, 2021-10-04:
This has been merged in https://github.com/mu-editor/mu/pull/1814, thanks to @Rybec & @keith-packard, you rock! 🤘

closed 2021-10-04 by carlosperate