mu-editor/mu #1325: Wrong link in splash screen

ajs256, 2021-02-23:
_bug report template removed because it doesn't matter for this issue_

The new splash screen added in https://github.com/mu-editor/mu/commit/75b167bc21a3bea98c9f025cc241f46e0811f68d shows [`codewithmu.com`](https://codewithmu.com) as the website. I believe [`codewith.mu`](https://codewith.mu) is the correct URL.

carlosperate, 2021-02-23:
Great catch, thanks @ajs256!

@ntoll assigning to you as maybe this is something we'll need to ask Steve?

ntoll, 2021-02-23:
Yeah... one for @hawkz

ntoll, 2021-02-23:
Also, this week's Eagle eyes of code award goes to @ajs256 :eagle: :eyes:  :1st_place_medal:

carlosperate, 2021-03-23:
This has been fixed in 4c14d239abec9593c8950e591f8f7ce48273d96a 🎉 
Thanks @hawkz!

closed 2021-03-23 by carlosperate