mu-editor/mu #1326: Issue creating venv

carlosperate, 2021-02-23:
Had a `mu_venv` already present and working from running Mu from source code.
Pulled from master (my last pull was last week, but before the last venv changes) and had the following error trying to run:

```
python run.py
qt.qpa.drawing: Layer-backing cannot be explicitly controlled on 10.14 when built against the 10.14 SDK
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Process(0x7fb8bcd04c40), parent's thread is QThread(0x7fb8be3169c0), current thread is QThread(0x7fb8bce43d90)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Process(0x7fb8bcd04c40), parent's thread is QThread(0x7fb8be3169c0), current thread is QThread(0x7fb8bce43d90)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Process(0x7fb8bcd04c40), parent's thread is QThread(0x7fb8be3169c0), current thread is QThread(0x7fb8bce43d90)
Traceback (most recent call last):
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/app.py", line 98, in run
    venv.ensure_and_create()
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/virtual_environment.py", line 390, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment
Assertion failed: (!needsMainQueue() || pthread_main_np() != 0), function invoke, file /AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/SkyLight/SkyLight-570.7/SkyLight/Services/PortStreams/CGSDatagramReadStream.cc, line 147.
[1]    53997 abort      python run.py
``` 

This is the log:

```
2021-02-23 11:46:30,368 - root:172(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.1
2021-02-23 11:46:30,369 - root:173(run) INFO: uname_result(system='Darwin', node='Carloss-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-02-23 11:46:30,369 - root:174(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python38.zip', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8/lib-dynload', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages', '/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages/IPython/extensions']
2021-02-23 11:46:30,369 - root:175(run) INFO: Language code: en_GB
2021-02-23 11:46:30,734 - mu.virtual_environment:379(ensure_and_create) DEBUG: Checking venv; attempt #1
2021-02-23 11:46:30,735 - mu.virtual_environment:416(ensure_path) INFO: Virtual Environment found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210223-094842
2021-02-23 11:46:30,736 - mu.virtual_environment:427(ensure_interpreter) INFO: Interpreter found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210223-094842/bin/python
2021-02-23 11:46:30,823 - mu.virtual_environment:456(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-02-23 11:46:30,825 - mu.virtual_environment:483(ensure_pip) INFO: Pip found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210223-094842/bin/pip
2021-02-23 11:46:30,825 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pygame
2021-02-23 11:46:37,848 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import numpy
2021-02-23 11:46:41,418 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pgzero
2021-02-23 11:46:41,456 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import flask
2021-02-23 11:46:41,809 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import serial
2021-02-23 11:46:41,844 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import qtconsole
2021-02-23 11:46:41,874 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import nudatus
2021-02-23 11:46:41,921 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import esptool
2021-02-23 11:46:41,951 - mu.virtual_environment:478(ensure_key_modules) ERROR: Failed to import esptool
2021-02-23 11:46:41,951 - mu.virtual_environment:382(ensure_and_create) DEBUG: Venv not present or correct
2021-02-23 11:46:41,951 - mu.virtual_environment:384(ensure_and_create) DEBUG: Creating new venv at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210223-114641
2021-02-23 11:46:41,951 - mu.virtual_environment:329(relocate) DEBUG: Virtual environment set up mu_venv-38-20210223-114641 at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210223-114641
2021-02-23 11:46:41,952 - mu.virtual_environment:495(create) INFO: Creating virtualenv: /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210223-114641
2021-02-23 11:46:41,952 - mu.virtual_environment:496(create) INFO: Virtualenv name: mu_venv-38-20210223-114641
2021-02-23 11:46:42,763 - mu.virtual_environment:543(install_baseline_packages) INFO: Installing baseline packages
2021-02-23 11:46:42,764 - mu.virtual_environment:544(install_baseline_packages) INFO: /Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels exists
2021-02-23 11:46:42,765 - mu.virtual_environment:190(install) DEBUG: About to pip install: ['/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/MarkupSafe-1.1.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/QtPy-1.9.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/prompt_toolkit-3.0.16-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/pyzmq-22.0.3-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/jedi-0.18.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/parso-0.8.1-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/pickleshare-0.7.5-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/traitlets-5.0.5-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/ptyprocess-0.7.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/pyserial-3.4-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/Flask-1.1.2-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/pexpect-4.8.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/setuptools-53.0.0-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/jupyter_core-4.7.1-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/pgzero-1.2-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/jupyter_client-6.1.11-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/Werkzeug-1.0.1-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/backcall-0.2.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/numpy-1.20.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/python_dateutil-2.8.1-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/click-7.1.2-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/tornado-6.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/ipykernel-5.4.3-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/qtconsole-4.7.4-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/Pygments-2.8.0-py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/appnope-0.1.2-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/mu-upstream/mu/wheels/six-1.15.0-py2.py3-none-any.whl', '/Users/microbit-carlos/workspace/mine/m [... cut ...]
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment
2021-02-23 11:47:55,797 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/session.json
2021-02-23 11:47:55,798 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/venv.json
```

I can see the 3 attemps at creating the mu_venvs in ~/Library/Application Support/mu.

@tjguk anything you'd like me to try or check?

tjguk, 2021-02-23:
Thanks for picking this up, @carlosperate. 

It's failing because it's failed to import the `esptool` module. But I can see no log of its being installed. I *think* what's happening here is that the logic in `virtual_environment.py:install_baseline_packages` will only download wheels if there are *no* wheels already in the `wheels` folder. In this case, you had wheels from a previous installation prior to the addition of `esptool` as a baseline package.

So it skips the download step (which would have removed any existing wheels / tarballs) and goes straight to the install. Which just installs whatever wheels are present, without reference to the baseline packages.

And so you're missing the `esptool` package and so the venv fails.

The good new is: we know that the venv mechanism correctly raises an exception afer three attempts. The bad news is: we have to come up with some logic which will detect and address this.

tjguk, 2021-02-23:
Do we have a clean way of detecting that we're running in "dev" mode? (Or, at least, not in packaged mode..) If so, then a simple option here is to _always_ download the wheels when creating the venv if we're in dev mode. Pip will have cached most of them so it won't take too long.

carlosperate, 2021-02-23:
>  I _think_ what's happening here is that the logic in `virtual_environment.py:install_baseline_packages` will only download wheels if there are _no_ wheels already in the `wheels` folder. In this case, you had wheels from a previous installation prior to the addition of `esptool` as a baseline package.
> 
> So it skips the download step (which would first have removed any existing wheels / tarballs) and goes straight to the install. Which just installs whatever wheels are present, without reference to the baseline packages.
>
> And so you're missing the esptool package and so the venv fails.

Great guess, sounds like this is what is happening. From the wheels directory:

```
total 60272
drwxr-xr-x  40 microbit-carlos  staff   1.3K 20 Feb 18:59 .
drwxr-xr-x  19 microbit-carlos  staff   608B 22 Feb 15:30 ..
-rw-r--r--@  1 microbit-carlos  staff   6.0K 16 Feb 23:16 .DS_Store
-rw-r--r--   1 microbit-carlos  staff    92K 16 Feb 23:23 Flask-1.1.2-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   123K 16 Feb 23:23 Jinja2-2.11.3-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    16K 16 Feb 23:23 MarkupSafe-1.1.1-cp38-cp38-macosx_10_9_x86_64.whl
-rw-r--r--   1 microbit-carlos  staff   960K 16 Feb 23:23 Pygments-2.8.0-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    53K 16 Feb 23:23 QtPy-1.9.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   292K 16 Feb 23:23 Werkzeug-1.0.1-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   3.6K 20 Feb 18:59 __init__.py
-rw-r--r--   1 microbit-carlos  staff   243B 19 Feb 18:40 __main__.py
-rw-r--r--   1 microbit-carlos  staff   4.2K 16 Feb 23:23 appnope-0.1.2-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    11K 16 Feb 23:23 backcall-0.2.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    81K 16 Feb 23:23 click-7.1.2-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   9.0K 16 Feb 23:23 decorator-4.4.2-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   117K 16 Feb 23:23 ipykernel-5.4.3-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   766K 16 Feb 23:23 ipython-7.20.0-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    26K 16 Feb 23:23 ipython_genutils-0.2.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    16K 16 Feb 23:23 itsdangerous-1.1.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   1.3M 16 Feb 23:23 jedi-0.18.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   106K 16 Feb 23:23 jupyter_client-6.1.11-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    81K 16 Feb 23:23 jupyter_core-4.7.1-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   5.2K 16 Feb 23:23 nudatus-0.0.5-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    15M 16 Feb 23:23 numpy-1.20.1-cp38-cp38-macosx_10_9_x86_64.whl
-rw-r--r--   1 microbit-carlos  staff    91K 16 Feb 23:23 parso-0.8.1-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    58K 16 Feb 23:23 pexpect-4.8.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    70K 16 Feb 23:23 pgzero-1.2-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   6.7K 16 Feb 23:23 pickleshare-0.7.5-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   358K 16 Feb 23:23 prompt_toolkit-3.0.16-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    14K 16 Feb 23:23 ptyprocess-0.7.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   6.6M 16 Feb 23:23 pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl
-rw-r--r--   1 microbit-carlos  staff   189K 16 Feb 23:23 pyserial-3.4-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   222K 16 Feb 23:23 python_dateutil-2.8.1-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   1.0M 16 Feb 23:23 pyzmq-22.0.3-cp38-cp38-macosx_10_9_x86_64.whl
-rw-r--r--   1 microbit-carlos  staff   116K 16 Feb 23:23 qtconsole-4.7.4-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   766K 16 Feb 23:23 setuptools-53.0.0-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    11K 16 Feb 23:23 six-1.15.0-py2.py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff   407K 16 Feb 23:23 tornado-6.1-cp38-cp38-macosx_10_9_x86_64.whl
-rw-r--r--   1 microbit-carlos  staff    98K 16 Feb 23:23 traitlets-5.0.5-py3-none-any.whl
-rw-r--r--   1 microbit-carlos  staff    30K 16 Feb 23:23 wcwidth-0.2.5-py2.py3-none-any.whl
```

carlosperate, 2021-02-23:
> Do we have a clean way of detecting that we're running in "dev" mode? (Or, at least, not in packaged mode..)

Not sure if we have a very clean way to do it. We could update pup to provide something to let us know if it's in packaged mode, but right now maybe the simplest way is to look at the source cod directory structure (does it have "repository" files or is it included in a site_packages inside a pup packed directory?

tjguk, 2021-02-23:
> Not sure if we have a very clean way to do it. We could update pup to provide something to let us know if it's in packaged mode, but right now maybe the simplest way is to look at the source cod [...]

Thinking about it, detecting that we're running inside a venv (eg via `os.environ['VIRTUAL_ENV']`) ought to be a good enough marker for a dev environment. I'll put together a PR to:

* Add a config flag that we're running in dev mode
* If in dev mode the venv will always run the `download` step when setting up a venv. 
 
That will slow things down but not by much, and will dodge this sort of bullet hopefully

carlosperate, 2021-02-23:
I guess if somebody pip installs mu not in a venv would be also considered "dev mode"?

tjguk, 2021-02-23:
> 
> 
> I guess if somebody pip installs mu not in a venv it would be also considered "dev mode"?
> Not sure how the deb package will be updated, but that will likely end up running not from a venv.

(Not 100% sure how to read your comment, but treating it as raising warning flags that certain dev-ish environments might not involve a venv)

I think the point here is that issues like this are really only affecting us (the developers of Mu). If someone's `pip install`-ing Mu or `apt install`-ing it then they're a user. Clearly, as @ntoll pointed out, we need to get our update story cleaned up. That's when the pip/apt users would need some help. 

Thinking aloud a little, if we were to rely on a version marker within Mu itself, could that be our standard release tag for Mu (thus capturing the installer/pip/apt audience) and the commit marker for dev mode (thus capturing us). Any change of marker could trigger a venv cleanup/refresh. Something like that?

carlosperate, 2021-02-23:
I guess I was just raising a question without a specific opinion about what it meant 😅

If a user pip installs mu from PyPI then maybe I'd consider them a bit more in the "advanced" area where downloading the wheels on every mu_venv installation is probably fine. Specially since they are not using an offline installer and are not getting the wheels with the `pip install mu` command, and will be downloading them on the first run anyway.

About the debian packaging, I honestly don't know how that will work. If I had to guess, the maintainers will likely patch mu to install mu dependencies and mu-modes dependencies via apt packages, so this might not actually apply in that case.

> I think the point here is that issues like this are really only affecting us (the developers of Mu).

I'd say in general we are going to encounter these issues a lot more often than end users, but these issues can still affect them. If a PyPI package user gets a bit tired of waiting on the first Mu run (which sounds quite plausible in very slow connection), and exits with ctrl+c while it's downloading the wheels, then their Mu installation will never run and fail like in this issue.

Of course there will be some problems that can only be trigger by running Mu as a developer, but I think the majority could still happen to "normal" users.

> Thinking aloud a little, if we were to rely on a version marker within Mu itself, could that be our standard release tag for Mu (thus capturing the installer/pip/apt audience) and the commit marker for dev mode (thus capturing the developer audience). Any change of marker could trigger a venv cleanup/refresh. Something like that?

Not sure I 100% follow on what the marker could be, but do you mean something like Mu is in "user mode" if it's running from a tagged commit or in "dev mode" from a non-tagged commit (or from a dirty working tree)?

mjackdk, 2021-02-23:
I'm using macOS CLI developer tools and I've updated pip to version 21.0.1.

I've cloned the mu repository, and used setup from [instructions](https://mu.readthedocs.io/en/latest/setup.html). No traces of previous mu or venv installations.

On first run I got the same errors as https://github.com/mu-editor/mu/issues/1326#issue-814394890.

I then used pip to install esptool in the venv, and I still get the same errors. This is what I see (~//Library/Application Support/mu bottom right):

https://user-images.githubusercontent.com/50518949/108895183-90291600-7613-11eb-9af4-8c4b1ed9846b.mov

Maybe I misunderstood https://github.com/mu-editor/mu/issues/1326#issuecomment-784213583, but installing epstool (this is after recent venv changes) is not enough to get Mu running on macOS Big Sur.

carlosperate, 2021-02-23:
Can you copy the log from `~/Library/Logs/mu` ?
There is another issue where pip throws an error due to different versions of pyserial being needed by different dependencies https://github.com/mu-editor/mu/pull/1316.

mjackdk, 2021-02-23:
From latest run, looks like #1316:
`-----------------

Starting Mu 1.1.0.beta.1
2021-02-23 20:52:01,969 - root:173(run) INFO: uname_result(system='Darwin', node='jackbook.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-02-23 20:52:01,969 - root:174(run) INFO: Python path: ['/Users/mjack/GitHub/mu', '/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python38.zip', '/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8', '/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/lib-dynload', '/Users/mjack/.venv/lib/python3.8/site-packages', '/Users/mjack/GitHub/mu', '/Users/mjack/.venv/lib/python3.8/site-packages/IPython/extensions']
2021-02-23 20:52:01,969 - root:175(run) INFO: Language code: en_DK
2021-02-23 20:52:02,315 - mu.virtual_environment:379(ensure_and_create) DEBUG: Checking venv; attempt #1
2021-02-23 20:52:02,316 - mu.virtual_environment:416(ensure_path) INFO: Virtual Environment found at /Users/mjack/Library/Application Support/mu/mu_venv-38-20210223-200606
2021-02-23 20:52:02,317 - mu.virtual_environment:427(ensure_interpreter) INFO: Interpreter found at /Users/mjack/Library/Application Support/mu/mu_venv-38-20210223-200606/bin/python
2021-02-23 20:52:02,430 - mu.virtual_environment:456(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-02-23 20:52:02,431 - mu.virtual_environment:483(ensure_pip) INFO: Pip found at /Users/mjack/Library/Application Support/mu/mu_venv-38-20210223-200606/bin/pip
2021-02-23 20:52:02,432 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pygame
2021-02-23 20:52:02,521 - mu.virtual_environment:478(ensure_key_modules) ERROR: Failed to import pygame
2021-02-23 20:52:02,521 - mu.virtual_environment:382(ensure_and_create) DEBUG: Venv not present or correct
2021-02-23 20:52:02,521 - mu.virtual_environment:384(ensure_and_create) DEBUG: Creating new venv at /Users/mjack/Library/Application Support/mu/mu_venv-38-20210223-205202
2021-02-23 20:52:02,522 - mu.virtual_environment:329(relocate) DEBUG: Virtual environment set up mu_venv-38-20210223-205202 at /Users/mjack/Library/Application Support/mu/mu_venv-38-20210223-205202
2021-02-23 20:52:02,522 - mu.virtual_environment:495(create) INFO: Creating virtualenv: /Users/mjack/Library/Application Support/mu/mu_venv-38-20210223-205202
2021-02-23 20:52:02,522 - mu.virtual_environment:496(create) INFO: Virtualenv name: mu_venv-38-20210223-205202
2021-02-23 20:52:03,905 - mu.virtual_environment:543(install_baseline_packages) INFO: Installing baseline packages
2021-02-23 20:52:03,906 - mu.virtual_environment:544(install_baseline_packages) INFO: /Users/mjack/GitHub/mu/mu/wheels exists
2021-02-23 20:52:03,907 - mu.virtual_environment:190(install) DEBUG: About to pip install: ['/Users/mjack/GitHub/mu/mu/wheels/esptool-3.0-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/MarkupSafe-1.1.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/mjack/GitHub/mu/mu/wheels/ecdsa-0.16.1-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/QtPy-1.9.0-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/prompt_toolkit-3.0.16-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pyzmq-22.0.3-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/mjack/GitHub/mu/mu/wheels/jedi-0.18.0-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/parso-0.8.1-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pickleshare-0.7.5-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/traitlets-5.0.5-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/ptyprocess-0.7.0-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pyserial-3.4-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl', '/Users/mjack/GitHub/mu/mu/wheels/Flask-1.1.2-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pycparser-2.20-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/bitstring-3.1.7-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pexpect-4.8.0-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/setuptools-53.0.0-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/jupyter_core-4.7.1-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/cffi-1.14.5-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/mjack/GitHub/mu/mu/wheels/pgzero-1.2-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/jupyter_client-6.1.11-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/Werkzeug-1.0.1-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/pyserial-3.5-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/backcall-0.2.0-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/numpy-1.20.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/mjack/GitHub/mu/mu/wheels/python_dateutil-2.8.1-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/click-7.1.2-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/tornado-6.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Users/mjack/GitHub/mu/mu/wheels/qtconsole-4.7.4-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/Pygments-2.8.0-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/wheels/ipykernel-5.5.0-py3-none-any.whl', '/Users/mjack/GitHub/mu/mu/whee [... cut ...]
  File "/Users/mjack/GitHub/mu/mu/virtual_environment.py", line 390, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment
2021-02-23 20:52:17,683 - mu.settings:169(save) DEBUG: Saving to /Users/mjack/Library/Application Support/mu/session.json
2021-02-23 20:52:17,692 - mu.settings:169(save) DEBUG: Saving to /Users/mjack/Library/Application Support/mu/venv.json
`

[mu.log](https://github.com/mu-editor/mu/files/6031609/mu.log)

jarl-dk, 2021-02-24:
OS: Ubuntu 20.04.2

I experience the exact same problem.
I also believe that https://mu.readthedocs.io/en/latest/setup.html
is missing important information, such as
1. What is the "correct" way to create a virtualenv for mu-editor
2. What OS developer packages are needed for developing mu-editor, only packages for raspberry pi are listed. I would like to see for Ubuntu.
I have opened a separate issue regarding this; issue #1328.

carlosperate, 2021-02-24:
The pyserial version conflict issue should now be fixed, so if you clone again from master it should work.

mjackdk, 2021-02-24:
> The pyserial version conflict issue should now be fixed, so if you clone again from master it should work.

I can confirm. Only one 'mu_venv-...' folder created during startup.

I've only tested Python 3 mode.

jarl-dk, 2021-02-24:
I can confirm as well after a `git pull` and `git clean -fdX` and starting over, the editor starts with `python run.py` (using python3)

carlosperate, 2021-02-24:
Awesome, thanks for confirming!

carlosperate, 2021-04-09:
Thanks everyone!
I'll close this in favour of https://github.com/mu-editor/mu/issues/1421, which should deal with the issue at hand of having some wheel (but maybe not the right wheels) can cause issues.

closed 2021-04-09 by carlosperate