mu-editor/mu #1327: pyperclip not found in mu-editor

vbadfrog, 2021-02-23:
Hi all,

2 days I've been looking for the solution, read hundreds of messages, watched videos... I can't find a clear step-by-step, spell-it-for-me-I-know-nohting, solution for that issue that a lot a people seem to have!

Following the automatetheboringstuff book, chp6, when it's time to install pyperclip.

I am using:
- Macbook Pro 2017
- macOS Mojave
- Python --version in CMD gives: Python 2.7.16
- mu-editor 1.1.0.alpha.2

I have done:
- installed successfully pyperclip with CMD, using pip3 cmd.
- Tried to install it in the 3rd party module of mu-editor, where the answer is:

pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting pyperclip
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/pyperclip/
  Could not fetch URL https://pypi.python.org/simple/pyperclip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.python.org', port=443): Max retries exceeded with url: /simple/pyperclip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping
  Could not find a version that satisfies the requirement pyperclip (from versions: )
No matching distribution found for pyperclip

FINISHED

- Current log:
2021-02-23 18:32:06,995 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/user/Library/Logs/mu/mu.log
2021-02-23 18:32:06,996 - mu.logic:163(installed_packages) INFO: Packages found: []
2021-02-23 18:32:15,446 - mu.logic:1203(sync_package_state) INFO: Synchronize package states...
2021-02-23 18:32:15,446 - mu.logic:1204(sync_package_state) INFO: Old: set()
2021-02-23 18:32:15,446 - mu.logic:1205(sync_package_state) INFO: New: {'pyperclip'}
2021-02-23 18:32:15,447 - mu.logic:1209(sync_package_state) INFO: To add: {'pyperclip'}
2021-02-23 18:32:15,447 - mu.logic:1210(sync_package_state) INFO: To remove: set()
2021-02-23 18:32:15,447 - mu.logic:1211(sync_package_state) INFO: Site packages: /Users/user/Library/Application Support/mu/site-packages
2021-02-23 18:32:15,448 - mu.interface.dialogs:456(run_pip) INFO: /Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3 -m pip install pyperclip --target /Users/user/Library/Application Support/mu/site-packages
2021-02-23 18:33:02,855 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/user/Library/Logs/mu/mu.log
2021-02-23 18:33:02,856 - mu.logic:163(installed_packages) INFO: Packages found: []

My total experience with coding is:
The first 5 chapters of the book!!

Thank you so much for your help!

carlosperate, 2021-02-23:
Sorry you are having issues installing 3rd party modules with the 1.1.0.alpha.2 version.
Unfortunately Mu in this version has an issue where it expects a library called OpenSSL to be installed on the computer.
If it cannot find this library, like I suspect might be in your case, it throws these errors and cannot download pyperclip from the internet.

The easiest way would be to use [homebrew](https://brew.sh/) to install OpenSSL library. If you have it, could you try running this command on a terminal and see if that fixes the issue?
```
brew install openssl@1.0
```

vbadfrog, 2021-02-23:
Hi @carlosperate, thanks for the answer.

Here the answer from terminal to this:

-bash: brew: command not found

carlosperate, 2021-02-23:
It's a very popular package manager for macOS, very useful, but like all software it can sometimes be a bit of a headache. I'd recommend to use it when you really need it and try not to install too much.

vbadfrog, 2021-02-23:
Ok, installed homebrew (successful), then ran the openssl line.

Here is the result:

==> Tapping homebrew/cask
Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask'...
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 551097 (delta 2), reused 0 (delta 0), pack-reused 551089
Receiving objects: 100% (551097/551097), 243.33 MiB | 228.00 KiB/s, done.
Resolving deltas: 100% (389163/389163), done.
Tapped 3832 casks (3,948 files, 261MB).
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
Error: No available formula or cask with the name "openssl@1.0".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.

carlosperate, 2021-02-23:
In that case an you try:

```
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/663e4792e1f355ef2571a06a9101606027cfedf9/Formula/openssl.rb
```

Hopefully this one should work.

vbadfrog, 2021-02-23:
Thank you so much for your help!

Unfortunately the result seems negative again:

==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
==> Updated Casks
Updated 2 casks.

Traceback (most recent call last):
	9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:122:in `<main>'
	8: from /usr/local/Homebrew/Library/Homebrew/cmd/install.rb:132:in `install'
	7: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:301:in `parse'
	6: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:644:in `formulae'
	5: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:644:in `map'
	4: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:648:in `block in formulae'
	3: from /usr/local/Homebrew/Library/Homebrew/formulary.rb:366:in `factory'
	2: from /usr/local/Homebrew/Library/Homebrew/formulary.rb:153:in `get_formula'
	1: from /usr/local/Homebrew/Library/Homebrew/formulary.rb:157:in `klass'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:242:in `load_file': Invalid usage: Installation of openssl from a GitHub commit URL is unsupported! `brew extract openssl` to a stable tap on GitHub instead. (UsageError)
	12: from /usr/local/Homebrew/Library/Homebrew/brew.rb:155:in `<main>'
	11: from /usr/local/Homebrew/Library/Homebrew/brew.rb:157:in `rescue in <main>'
	10: from /usr/local/Homebrew/Library/Homebrew/help.rb:64:in `help'
	 9: from /usr/local/Homebrew/Library/Homebrew/help.rb:83:in `command_help'
	 8: from /usr/local/Homebrew/Library/Homebrew/help.rb:103:in `parser_help'
	 7: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:301:in `parse'
	 6: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:644:in `formulae'
	 5: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:644:in `map'
	 4: from /usr/local/Homebrew/Library/Homebrew/cli/parser.rb:648:in `block in formulae'
	 3: from /usr/local/Homebrew/Library/Homebrew/formulary.rb:366:in `factory'
	 2: from /usr/local/Homebrew/Library/Homebrew/formulary.rb:153:in `get_formula'
	 1: from /usr/local/Homebrew/Library/Homebrew/formulary.rb:157:in `klass'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:242:in `load_file': Invalid usage: Installation of openssl from a GitHub commit URL is unsupported! `brew extract openssl` to a stable tap on GitHub instead. (UsageError)

carlosperate, 2021-02-23:
Looks like brew has disable this way of installing old versions of libraries, and that is very unfortunate, as that means we cannot really install the library Mu needs using the "normal" method.

Doing a quick search around I found this alternative way to install OpenSSL v1.0.2, can you try following these steps to see if it works?
https://smithua.medium.com/install-openssl-1-0-2t-version-on-macos-7084ac52fa8b

Btw, this issue is something we have resolved (thanks to the great work from @tmontes) and will not be a problem in the next Mu release, but we are still resolving some packaging issues before we can release it.

carlosperate, 2021-02-23:
Maybe this would work as well (from https://stackoverflow.com/a/61368160):

```
brew install rbenv/tap/openssl@1.0
```

vbadfrog, 2021-02-24:
The stackoverflow method seems to have done something.

Here is the result from CMD:

==> perl ./Configure --prefix=/usr/local/Cellar/openssl@1.0/1.0.2t --openssldir=/usr/local/etc/openssl no-ssl2 no-ssl3 no-zl
==> make depend
==> make
==> make test
==> make install MANDIR=/usr/local/Cellar/openssl@1.0/1.0.2t/share/man MANSUFFIX=ssl
==> Caveats
A CA file has been bootstrapped using certificates from the SystemRoots
keychain. To add additional certificates (e.g. the certificates added in
the System keychain), place .pem files in
  /usr/local/etc/openssl/certs

and run
  /usr/local/opt/openssl@1.0/bin/c_rehash

openssl@1.0 is keg-only, which means it was not symlinked into /usr/local,
because Apple has deprecated use of OpenSSL in favor of its own TLS and crypto libraries.

If you need to have openssl@1.0 first in your PATH, run:
  echo 'export PATH="/usr/local/opt/openssl@1.0/bin:$PATH"' >> /Users/user/.bash_profile

For compilers to find openssl@1.0 you may need to set:
  export LDFLAGS="-L/usr/local/opt/openssl@1.0/lib"
  export CPPFLAGS="-I/usr/local/opt/openssl@1.0/include"

==> Summary
🍺  /usr/local/Cellar/openssl@1.0/1.0.2t: 1,787 files, 12.0MB, built in 4 minutes 43 seconds

However, after that, I still cannot install pyperclip in mu (same error as before, no change). I have tried to run the line as specified above as well, but it didn't change anything.
echo 'export PATH="/usr/local/opt/openssl@1.0/bin:$PATH"' >> /Users/user/.bash_profile

vbadfrog, 2021-02-25:
@carlosperate, thank you so much for the time you spent helping me! It seems the time spent on this issue is not really worth the trouble if you guys are about to release a version where this will not be a problem anymore.
I'll use IDLE until the update.

Thanks again!

carlosperate, 2021-02-26:
Yes, hopefully we are getting very close to the next release. Sorry I couldn't be of more help and thank you for your perseverance testing multiple ways of installing OpenSSL!

closed 2021-02-26 by carlosperate