mu-editor/mu #1328: Missing information in developer setup

jarl-dk, 2021-02-24:
I am trying to build and run mu-editor from source
I try to follow the instructions on https://mu.readthedocs.io/en/latest/setup.html

First thing is
> Create a working development environment by installing all the dependencies into _your virtualenv_ with

The instructions are missing steps prior to this such as
1. What OS development packages are necessary to build from source. Only for raspberry pi such development packages are listed. Personally I am missing a list of packages for Ubuntu.
2. What is the "correct" way to create "your virutalenv", if it is 
```
python -mvenv .venv
source .venv/bin/activate
```
then please state so.

carlosperate, 2022-07-16:
Perhaps surprisingly, creating a virtual environment can be quite different dependending on the operating systems, the way Python was installed, and other factors.
For this issue I think that rather than include instructions in the dev docs that might not work for eveybody, we can find some article only with a good intro to virtual environments and steps for different operating systems, and link to that for users to find out more if they need to.

prcutler, 2022-07-16:
What about the Real Python guide: https://realpython.com/python-virtual-environments-a-primer/ 

Or is that too much information?

prcutler, 2022-07-26:
PR #2299 was merged - I think this issue can be closed.  Thanks!

carlosperate, 2022-07-27:
You are right, thanks @prcutler!

closed 2022-07-27 by carlosperate