mu-editor/mu #1334: Update auto-suggest and "intellisense" for CircuitPython mode

ntoll, 2021-02-24:
CircuitPython and related libraries have type information and comments as Python stubs (see [here](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/stubs/)). We should use those to update the "intellisense" (tool-tips and autocomplete in Scintilla's parlance). For this to happen we need a quick tool to replace the [scraper we made to do this from the Adafruit Sphinx docs](https://github.com/mu-editor/mu/blob/master/utils/adafruit_api.py).

carlosperate, 2021-02-25:
I think the stub file format is useful for autocompletion systems where we can just point to them and it does its magic.

Looking at https://github.com/mu-editor/mu/tree/master/mu/modes/api it sounds like QScintilla maybe would need the API to be in the same format stored in these files?
If it could just parse stub files, then fantastic, but if we need to create a script to convert stub files into this format, then we can just continue using the sphinx docs parser.
If the new (sphinx) CircuitPython docs are generated from the stub files, then we are getting the same data either way.

carlosperate, 2021-04-15:
This might also be a useful link: https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html