mu-editor/mu #1338: Help button links to home site error page

ErikVanp, 2021-02-28:
Mu 1.0.2 on kubuntu 20.04 language nl
Help button >> ¡Lo siento! La página que buscas no existe. :-(

cause: site supports only 'en' and 'es'
fixed by editing /usr/share/mu-editor/mu/logic.py

added:
if language_code != 'en' and language_code != 'es':
            language_code = 'en'
above:
     url = 'https://codewith.mu/{}/help/{}'.format(language_code,major_version)

dybber, 2021-03-01:
Good catch!

carlosperate, 2021-03-29:
@ntoll maybe we can add a bit of js to the 404 to:
- Check if the URL has a language flag
    - If it does, ping the same page with an `en` flag to check if it exists
        - If it exists, redirect to the english version

closed 2022-12-04 by ntoll