mu-editor/mu #1339: Using mu with birdbrain technologies hummingbird/finch robot

fmorton, 2021-03-01:
Birdbrain Technologies provides their library as a single file contain classes for their devices. The file is available at:

https://www.birdbraintechnologies.com/downloads/installers/BirdBrainPython.zip

To make this available in the Python 3 mode on a Mac, I copied the BirdBrain.py file into the site-packages directory buried under the Library/Application Support directory, specifically on my system:

~/Library/Application Support/mu/mu_venv/lib/python3.7/site-packages

Is this the best way to add a class library like this to sys.path?

Thanks.

carlosperate, 2021-03-02:
Is that a "desktop" Python module? Is it meant for a Raspberry Pi? Or how does it interface with the robot?

carlosperate, 2021-03-02:
If the code is in github it can probably be pip installed via URL, and that could probably be done via the Third Party packages window. Although I haven't tried that yet.

dybber, 2021-03-02:
Perhaps off-topic, but I'll add it anyway: if the library is a single file, I would just ask my pupils/students to copy it to the directory of their project. I think that kids should also learn that there's nothing "magical" about creating libraries and showing them that it's just another Python file is a nice illustration of that.

As this is not really an issue, I will convert this to a discussion in our discussion forum until we've figured out if there's something that needs to change in Mu.

closed 2021-03-02 by dybber